espressif/esp-idf #14582: `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` fails to compile: dhcp_network_changed not defined

xobs, 2024-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-3023-g3c99557eee

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Previously, `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` compiled just fine.

### What is the actual behavior?

As of tonight, `CONFIG_LWIP_DHCP_RESTORE_LAST_IP` fails to compile. This is because it invokes a macro defined as:

```
#ifdef CONFIG_LWIP_DHCP_RESTORE_LAST_IP
/*
 * Make the post-init hook check if we could restore the previously bound address
 * - if yes reset the state to bound and mark result as ERR_OK (which skips discovery state)
 * - if no, return false to continue normally to the discovery state
 */
#define LWIP_HOOK_DHCP_POST_INIT(netif, result) \
    (dhcp_ip_addr_restore(netif) ? ( dhcp_set_state(dhcp, DHCP_STATE_BOUND), \
                                     dhcp_network_changed(netif), \
                                     (result) = ERR_OK , \
        true ) : \
        false)
#else
#define LWIP_HOOK_DHCP_PRE_DISCOVERY(netif, result) (false)
#endif /* CONFIG_LWIP_DHCP_RESTORE_LAST_IP */
```

However, `dhcp_network_changed()` no longer exists.

### Steps to reproduce.

1. Enable `CONFIG_LWIP_DHCP_RESTORE_LAST_IP`

### Build or installation Logs.

```plain
In file included from C:/Users/Sean/esp/esp-idf/components/lwip/lwip/src/include/lwip/opt.h:51,
                 from C:/Users/Sean/esp/esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c:66:
C:/Users/Sean/esp/esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c: In function 'dhcp_start':
C:/Users/Sean/esp/esp-idf/components/lwip/port/include/lwipopts.h:365:38: error: implicit declaration of function 'dhcp_network_changed'; did you mean 'dhcp_network_changed_link_up'? [-Wimplicit-function-declaration]
  365 |                                      dhcp_network_changed(netif), \
      |                                      ^~~~~~~~~~~~~~~~~~~~
C:/Users/Sean/esp/esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c:945:7: note: in expansion of macro 'LWIP_HOOK_DHCP_POST_INIT'
  945 |   if (LWIP_HOOK_DHCP_POST_INIT(netif, result)) {
      |       ^~~~~~~~~~~~~~~~~~~~~~~~
```
```

### More Information.

This started failing this evening.

xobs, 2024-09-27:
It looks like `network_changed()` was split into `network_changed_link_down()` and `network_changed_link_up()`, and in fact `network_changed()` was simply renamed to `network_changed_link_up()`.

So this should be as simple as renaming the function.

xobs, 2024-09-28:
As a workaround while this is getting fixed, you can add this to your `CMakeLists.txt`:

```
idf_build_set_property(COMPILE_DEFINITIONS "dhcp_network_changed=dhcp_network_changed_link_up" APPEND)
```

expresspotato, 2024-11-12:
Still an issue here on 5.0.7 from what I can see

closed 2025-01-24 by espressif-bot