espressif/esp-idf #14587: can't set efuses as read protected after secure boot v2 is enabled

greenaddress, 2024-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32s3 tdisplay s3

### Power Supply used.

USB

### What is the expected behavior?

I expect to be able to set some efuses as read protected and have secure boot v2 as a feature.

However it appears that any efuse writing/protection ought to be done before or at the same time as secure boot v2 as after secure boot v2 is enabled efuses can no longer be set as read protected.

What is the suggested way of doing so?  

### What is the actual behavior?

Not clear what is the best procedure to configure some efuses as read protected and enable secure boot v2

### Steps to reproduce.

1. Enable secure boot v2
2. set some efuse as read protected fails

### More Information.

the problem is not in the code per se, it is not clear from the documentation what is the ideal procedure to set some efuses as read protected before secure boot v2.

options explored so far:
(1) flash bootloader and image without secure boot which sets up the efuses as appropriate, then flash secure boot v2 enabled firmware
(2) use support hooks/override of the bootloader to set things up before the efuses are finalized by secure boot v2 (but we do not have access to post initialization freertos facilities)
(3) fork the bootloader to do the above (hooks/override tend to not have access to private includes which may be needed to avoid reimplementing or copy and pasting things)

KonstantinKondrashov, 2024-09-16:
Hi @greenaddress!
It is correct behavior that after the bootloader enables the SBV2 it disables further read-protection changes. It is done to protect the secure boot key from being read-protected (it has to be readable).
There is [a Kconfig option - CONFIG_SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS.](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig.html#config-secure-boot-v2-allow-efuse-rd-dis) to ignore this if yu want to make some efuse setting in your app letter, (but do not forget to burn ESP_EFUSE_WR_DIS_RD_DIS using esp_efuse_write_field_bit() after all).
Doc - https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/secure-boot-v2.html?highlight=secure_boot_v2_allow_efuse_rd_dis#burning-read-protected-keys

Probably, we have to describe this case better in the doc, If this raises questions, or did you not find the doc?

greenaddress, 2024-09-16:
@KonstantinKondrashov Thank you for the quick answer, the documentation indeed gives some options but it felt that using CONFIG_SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS was highly discouraged and we excluded this option by mistake and perhaps it is the best way to setup protected efuses after secure boot v2 is enabled and later secure things by burning ESP_EFUSE_WR_DIS_RD_DIS.

In which case i am  satisfied with the answer, want me to close the issue? thanks again

AdityaHPatwardhan, 2024-09-17:
@greenaddress Thanks for the issue.
We will update the documentation internally to make it more clear and then close this issue.

closed 2024-10-07 by espressif-bot