espressif/esp-idf #14588: Low RAM available after updating from 5.0.4 to 5.1.4

yanmartins, 2024-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a custom board with external memory.

Before updating from version 5.0.4 to 5.1.4 I had around 8MB of free RAM. Now this value has dropped to approximately 300KB.

When I run the code, I add this section;

```c
void app_main(void)
{
    for(uint8_t i = 0; i < 5; i++)
    {
        const int heap_size = xPortGetFreeHeapSize();
        ESP_LOGI(TAG, "Heap size -> %i", heap_size);
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
   ...
}
```

And I got this for 5.0.4:

<details>

<summary> Logs </summary>

```
I (24) boot: ESP-IDF v5.0.4-dirty 2nd stage bootloader
I (25) boot: compile time 15:48:30
I (25) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length  
I (62) boot:  0 otadata          OTA data         01 00 0000e000 00002000
I (69) boot:  1 phy_init         RF data          01 01 00010000 00001000
I (77) boot:  2 ota_0            OTA app          00 10 00020000 00400000
I (84) boot:  3 ota_1            OTA app          00 11 00420000 00400000
I (91) boot:  4 nvs              WiFi data        01 02 00820000 00008000
I (99) boot:  5 nvs_app          WiFi data        01 02 00828000 00020000
I (106) boot:  6 nvs_prof         WiFi data        01 02 00848000 00200000
I (114) boot:  7 factory_conf     WiFi data        01 02 00a48000 00006000
I (122) boot:  8 nvs_keys         NVS keys         01 04 00a4e000 00001000
I (129) boot:  9 reserved         unknown          af af 00a4f000 00400000
I (137) boot: End of partition table
I (141) esp_image: segment 0: paddr=00020020 vaddr=3c1a0020 size=a73d0h (685008) map
I (272) esp_image: segment 1: paddr=000c73f8 vaddr=3fc9ed00 size=0731ch ( 29468) load
I (279) esp_image: segment 2: paddr=000ce71c vaddr=40374000 size=018fch (  6396) load
I (281) esp_image: segment 3: paddr=000d0020 vaddr=42000020 size=192bcch (1649612) map
I (583) esp_image: segment 4: paddr=00262bf4 vaddr=403758fc size=192f8h (103160) load
I (617) boot: Loaded app from partition at offset 0x20000
I (617) boot: Disabling RNG early entropy source...
I (629) cpu_start: Multicore app
I (629) octal_psram: vendor id    : 0x0d (AP)
I (629) octal_psram: dev id       : 0x02 (generation 3)
I (632) octal_psram: density      : 0x03 (64 Mbit)
I (638) octal_psram: good-die     : 0x01 (Pass)
I (643) octal_psram: Latency      : 0x01 (Fixed)
I (648) octal_psram: VCC          : 0x01 (3V)
I (653) octal_psram: SRF          : 0x01 (Fast Refresh)
[... 148 lines cut ...]
I (1517) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1523) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (1530) esp_psram: Adding pool of 8154K of PSRAM memory to heap allocator
I (1538) spi_flash: detected chip: gd
I (1542) spi_flash: flash io: dio
W (1546) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (1558) sleep: Configure to isolate all GPIO pins in sleep state
I (1564) sleep: Enable automatic switching of GPIO sleep configuration
I (1571) app_start: Starting scheduler on CPU0
I (1576) app_start: Starting scheduler on CPU1
I (1576) main_task: Started on CPU0
I (1586) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1596) main_task: Calling app_main()
I (1596) app_main: Heap size -> 288647
I (2606) app_main: Heap size -> 288647
I (3606) app_main: Heap size -> 288647
I (4606) app_main: Heap size -> 288647
I (5606) app_main: Heap size -> 288647
```

</details>

Are there any new settings that could have impacted this value so much?

Thanks!

yanmartins, 2024-09-16:
Here are the SDKs for both tests:

<details>

<summary> SDK 5.0.4 </summary>

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_LEDC_SUPPORT_APB_CLOCK=y
CONFIG_SOC_LEDC_SUPPORT_XTAL_CLOCK=y
CONFIG_SOC_LEDC_CHANNEL_NUM=8
CONFIG_SOC_LEDC_TIMER_BIT_WIDE_NUM=14
CONFIG_SOC_LEDC_SUPPORT_FADE_STOP=y
CONFIG_SOC_MPU_MIN_REGION_SIZE=0x20000000
CONFIG_SOC_MPU_REGIONS_MAX_NUM=8
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_LCDCAM_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_MEMPROT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_APPCPU_HAS_CLOCK_GATING_BUG=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_ARBITER_SUPPORTED=y
CONFIG_SOC_ADC_FILTER_SUPPORTED=y
CONFIG_SOC_ADC_MONITOR_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=24
[... 4254 lines cut ...]
# CONFIG_PPP_DEBUG_ON is not set
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S3_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S3_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```

</details>

SoucheSouche, 2024-09-17:
Hi @yanmartins,
It seems that `xPortGetFreeHeapSize` is calling `heap_caps_get_free_size(MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT)` which as you can see does no longer include the external memory available.
It is still available and can be used to allocate memory but this API does no longer include the external memory value into the calculation. If you call by yourself heap_caps_get_free_size(MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM) you should get the 8MB you are expecting.

yanmartins, 2024-09-17:
> Hi @yanmartins, It seems that `xPortGetFreeHeapSize` is calling `heap_caps_get_free_size(MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT)` which as you can see does no longer include the external memory available. It is still available and can be used to allocate memory but this API does no longer include the external memory value into the calculation. If you call by yourself `heap_caps_get_free_size(MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT | MALLOC_CAP_SPIRAM)` you should get the 8MB you are expecting. (available in `esp_heap_caps.h`).

Now I got Zero:

```
I (1596) main_task: Calling app_main()
I (1596) app_main: Heap size -> 0
I (2606) app_main: Heap size -> 0
I (3606) app_main: Heap size -> 0
I (4606) app_main: Heap size -> 0
I (5606) app_main: Heap size -> 0
```

yanmartins, 2024-09-17:
@SoucheSouche By using just the line below, it worked

`c
heap_caps_get_free_size(MALLOC_CAP_SPIRAM)
`

Thanks!

closed 2024-09-17 by yanmartins

yanmartins, 2024-09-17:
> MALLOC_CAP_8BIT

@SoucheSouche Yes! I did it:

```
I (1595) main_task: Calling app_main()
I (1595) app_main: Heap size: Ext -> 8346648 Int -> 288647
I (2605) app_main: Heap size: Ext -> 8346648 Int -> 288647
I (3605) app_main: Heap size: Ext -> 8346648 Int -> 288647
I (4605) app_main: Heap size: Ext -> 8346648 Int -> 288647
I (5605) app_main: Heap size: Ext -> 8346648 Int -> 288647
```

Thank you!