espressif/esp-idf #14592: W (1388) SPIFFS: Failed to update mtime (-10001)

allacmc, 2024-09-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Guys,

Why is this error occurring?

Would it be when the flash memory is full?

pacucha42, 2024-10-17:
Hi @allacmc,
if I decode your report well, you are getting error -10001 on fopen/fstat call, and the file modification time is not updated in case of fopen. The error says your file system is full and SPIFFS cannot process any further operations (SPIFFS_ERR_FULL). You can enable SPIFFS debugging output using SPIFFS_DBG Kconfig option to see more details. I assume that cleaning your file-system or extending your SPIFFS partition size should be enough, but it's just a guess.

If this doesn't help, please provide more information about your application context (configuration, code excerpt with the failing sequence) and ideally also verbose logs.

pacucha42, 2025-03-06:
Closing as expired. Please, feel free to reopen or file a new tracker. Thank you

closed 2025-03-06 by pacucha42