espressif/esp-idf #14595: examples/peripherals/usb/host/msc not working

diplfranzhoepfinger, 2024-09-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF Release v5.3.1

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected it to work.

### What is the actual behavior?

Instead it does print: I (326) example: Waiting for USB flash drive to be connected

### Steps to reproduce.

compile and flash the examples/peripherals/usb/host/msc

i did use this Cable: https://www.amazon.de/deleyCON-Adapter-Kabel-Nylon-Metallstecker/dp/B071K8KVKY

and connected it to a ESP32 S3 Devkitc

### Build or installation Logs.

```plain
I (23) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (24) boot: compile time Sep 17 2024 11:12:19
I (29) boot: Multicore bootloader
I (29) boot: chip revision: v0.2
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (35) boot.esp32s3: SPI Mode       : DIO
I (40) boot.esp32s3: SPI Flash Size : 2MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (76) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0dbc4h ( 56260) map
I (106) esp_image: segment 1: paddr=0001dbec vaddr=3fc92b00 size=0242ch (  9260) load
I (108) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=31d08h (204040) map
I (150) esp_image: segment 3: paddr=00051d30 vaddr=3fc94f2c size=006d8h (  1752) load
I (150) esp_image: segment 4: paddr=00052410 vaddr=40374000 size=0ea78h ( 60024) load
I (174) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (187) cpu_start: Multicore app
I (196) cpu_start: Pro cpu start user code
I (196) cpu_start: cpu freq: 160000000 Hz
I (196) app_init: Application information:
I (199) app_init: Project name:     usb-msc
I (204) app_init: App version:      1
I (208) app_init: Compile time:     Sep 17 2024 11:12:35
I (214) app_init: ELF file SHA256:  a5ee46827...
I (220) app_init: ESP-IDF:          v5.3.1
I (224) efuse_init: Min chip rev:     v0.0
I (229) efuse_init: Max chip rev:     v0.99 
I (234) efuse_init: Chip rev:         v0.2
I (239) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC95F28 len 000537E8 (333 KiB): RAM
I (252) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (258) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (265) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (272) spi_flash: detected chip: gd
I (275) spi_flash: flash io: dio
W (279) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (292) sleep: Configure to isolate all GPIO pins in sleep state
I (299) sleep: Enable automatic switching of GPIO sleep configuration
I (306) main_task: Started on CPU0
I (326) main_task: Calling app_main()
I (326) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (326) example: Waiting for USB flash drive to be connected
```

diplfranzhoepfinger, 2024-09-17:
[image]

roma-jam, 2024-09-17:
Hi @diplfranzhoepfinger , 

Unfortunately, ESP32-S3-DEVKITC supports only USB Device example out of the box. 
For USB Host examples, the board should supply the connected USB Device. 
To achieve that, you may short the Diode D7 to provide 5V on the USB-Port, 

Here is related Q&A: https://esp32.com/viewtopic.php?f=25&t=26377&p=126685&hilit=usb#p126685

Sorry for the inconvenience.

diplfranzhoepfinger, 2024-10-09:
i remove Diode D7, and add a solid wire instead!

diplfranzhoepfinger, 2024-10-09:
```
--- esp-idf-monitor 1.5.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
����x����������x�x��x�x����8480) example: Done
�ESP-ROM:esp32s3-20210327rned from app_main()
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2db0
entry 0x403c8914
I (27) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (27) boot: compile time Sep 17 2024 11:12:19
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0dbc4h ( 56260) map
I (109) esp_image: segment 1: paddr=0001dbec vaddr=3fc92b00 size=0242ch (  9260) load
I (112) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=31d08h (204040) map
I (153) esp_image: segment 3: paddr=00051d30 vaddr=3fc94f2c size=006d8h (  1752) load
I (153) esp_image: segment 4: paddr=00052410 vaddr=40374000 size=0ea78h ( 60024) load
I (178) boot: Loaded app from partition at offset 0x10000
I (178) boot: Disabling RNG early entropy source...
I (190) cpu_start: Multicore app
I (199) cpu_start: Pro cpu start user code
I (199) cpu_start: cpu freq: 160000000 Hz
I (200) app_init: Application information:
I (202) app_init: Project name:     usb-msc
I (207) app_init: App version:      1
I (212) app_init: Compile time:     Sep 17 2024 11:12:35
I (218) app_init: ELF file SHA256:  a5ee46827...
I (223) app_init: ESP-IDF:          v5.3.1
I (228) efuse_init: Min chip rev:     v0.0
I (232) efuse_init: Max chip rev:     v0.99 
I (237) efuse_init: Chip rev:         v0.2
I (242) heap_init: Initializing. RAM available for dynamic allocation:
I (249) heap_init: At 3FC95F28 len 000537E8 (333 KiB): RAM
I (255) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (262) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (268) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (275) spi_flash: detected chip: gd
I (278) spi_flash: flash io: dio
W (282) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (295) sleep: Configure to isolate all GPIO pins in sleep state
I (302) sleep: Enable automatic switching of GPIO sleep configuration
I (310) main_task: Started on CPU0
I (330) main_task: Calling app_main()
I (330) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (330) example: Waiting for USB flash drive to be connected
I (740) example: MSC device connected
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x0
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x781
idProduct 0x5567
bcdDevice 1.00
iManufacturer 1
iProduct 2
iSerialNumber 3
bNumConfigurations 1
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 32
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0x80
bMaxPower 200mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 2
        bInterfaceClass 0x8
        bInterfaceSubClass 0x6
        bInterfaceProtocol 0x50
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x2    EP 2 OUT
                bmAttributes 0x2        BULK
                wMaxPacketSize 64
                bInterval 0
Device info:
         Capacity: 15263 MB
         Sector size: 512
         Sector count: 31260671
         PID: 0x5567 
         VID: 0x0781 
         iProduct: Cruzer Blade 
         iManufacturer: SanDisk 
         iSerialNumber: 00017316102523132238 
I (940) example: ls command output:
SYSTEM~1
ESP
I (950) example: Reading file
I (950) example: Read from file '/usb/esp/test.txt': 'Hello World!'
I (970) example: Writing to file /usb/esp/dummy
I (2170) example: Write speed 0.87 MiB/s
I (2180) example: Reading from file /usb/esp/dummy
I (3200) example: Read speed 1.02 MiB/s
I (3210) example: Example finished, you can disconnect the USB flash drive

```

diplfranzhoepfinger, 2024-10-09:
WORKING !

diplfranzhoepfinger, 2024-10-09:
[image]

closed 2024-10-09 by diplfranzhoepfinger