espressif/esp-idf #14598: mbedtls_dhm_context does not have a len member

nopnop2002, 2024-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

Chip is ESP32D0WDQ6 (revision 1)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Development Board

### Power Supply used.

USB

### What is the expected behavior?

mbedtls_dhm_context  have a len member

### What is the actual behavior?

mbedtls_dhm_context does not have a len member

### Steps to reproduce.

This is the definition of mbed TLS v2.2.0.
```
typedef struct
{
    size_t len; 
    mbedtls_mpi P;      
    mbedtls_mpi G;      
    mbedtls_mpi X;      
    mbedtls_mpi GX;     
    mbedtls_mpi GY;     
    mbedtls_mpi K;      
    mbedtls_mpi RP;     
    mbedtls_mpi Vi;     
    mbedtls_mpi Vf;     
    mbedtls_mpi pX;     
}
mbedtls_dhm_context;
```

Here is the definition of mbedtls_dhm_context.
https://github.com/espressif/mbedtls/blob/5d112760293404ca0452bf244aa025079cf2e5af/include/mbedtls/dhm.h#L101

There is no len member in the EP-IDF definition.

```
/**
 * \brief          The DHM context structure.
 */
typedef struct mbedtls_dhm_context {
    mbedtls_mpi MBEDTLS_PRIVATE(P);      /*!<  The prime modulus. */
    mbedtls_mpi MBEDTLS_PRIVATE(G);      /*!<  The generator. */
    mbedtls_mpi MBEDTLS_PRIVATE(X);      /*!<  Our secret value. */
    mbedtls_mpi MBEDTLS_PRIVATE(GX);     /*!<  Our public key = \c G^X mod \c P. */
    mbedtls_mpi MBEDTLS_PRIVATE(GY);     /*!<  The public key of the peer = \c G^Y mod \c P. */
    mbedtls_mpi MBEDTLS_PRIVATE(K);      /*!<  The shared secret = \c G^(XY) mod \c P. */
    mbedtls_mpi MBEDTLS_PRIVATE(RP);     /*!<  The cached value = \c R^2 mod \c P. */
    mbedtls_mpi MBEDTLS_PRIVATE(Vi);     /*!<  The blinding value. */
    mbedtls_mpi MBEDTLS_PRIVATE(Vf);     /*!<  The unblinding value. */
    mbedtls_mpi MBEDTLS_PRIVATE(pX);     /*!<  The previous \c X. */
}
mbedtls_dhm_context;
```

nopnop2002, 2024-09-18:
It turns out that the len member is gone in mbed TLS 3.6.1.

closed 2024-09-18 by nopnop2002

eiSouthBoy, 2024-12-24:
Yes, I'm getting the same error as well. So, what is the reason for this?

```bash
error: 'mbedtls_dhm_context' has no member named 'P'
  140 |         printf("P(%d): \n", blufi_sec->dhm.P.n);
```