espressif/esp-idf #14599: Decreased total ram size after upgrading from IDF 4.4.7 to IDF 5.3.0

hitecSmartHome, 2024-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**I'm using an ESP32-Wrover-E with 16mb flash and 8mb PSRAM.**

### Ram usage after IDF 5.3.0
```json
{
    "external": {
        "total": 4178604,
        "free": 3853256,
        "largestFree": 3801088
    },
    "internal": {
        "total": 337667,
        "free": 126359,
        "largestFree": 90112
    },
}
````

### Ram usage on IDF 4.4.7
```json
{
    "external": {
        "total": 4179927,
        "free": 3769007,
        "largestFree": 3670016
    },
    "internal": {
        "total": 365375,
        "free": 157115,
        "largestFree": 77824
    },
}
````

This is how I get the ram values and storing it in a json document with `ArduinoJson V7`

```cpp
JsonObject extRam = memoryInfo["external"].to<JsonObject>();
JsonObject internalRam = memoryInfo["internal"].to<JsonObject>();

extRam["total"] = heap_caps_get_total_size(MALLOC_CAP_SPIRAM);
extRam["free"] = heap_caps_get_free_size(MALLOC_CAP_SPIRAM);
extRam["largestFree"] = heap_caps_get_largest_free_block(MALLOC_CAP_SPIRAM);

internalRam["total"] = heap_caps_get_total_size(MALLOC_CAP_INTERNAL);
internalRam["free"] = heap_caps_get_free_size(MALLOC_CAP_INTERNAL);
internalRam["largestFree"] = heap_caps_get_largest_free_block(MALLOC_CAP_INTERNAL);
```

As you can see, the reported internal ram is **365375 bytes** on 4.4.7 and **337667 bytes** on 5.3.0
Also the largest free block in internal is decreased significantly. Roughly with the same amount as the total ram decrease.
The largest free block is not representative in this case because it is two separate system but the chip is the same. The total ram should have not changed!

**Where did the ram go?**

hitecSmartHome, 2024-09-18:
I have six systems currently running with ESP32-Wrover-E.
Three of them using IDF 4.4.7 and the other three using IDF 5.3.0

**IDF 5.3.0**
[image]

**IDF 4.4.7**
[image]

Sorry for the language in the pictures, it's Hungarian. The important part is this

[image]

The first block is the internal ram, the second block is the external in this order.

- Total ram
- Used ram
- Free ram
- Largest free block ram

SoucheSouche, 2024-09-19:
Hi @hitecSmartHome,
This is quite unexpected drop.
I will have a look at it. In the mean time, could you provide the sdkconfig of your project?

hitecSmartHome, 2024-09-19:
Thank you very much for the response.

Meanwhile I have changed a lot of things in the menuconfig ( mainly wifi related stuffs ) since than and the total ram went up to **372kb**. I don't really understand how **the total ram can change** if I enable or disable something in the menuconfig...

SoucheSouche, 2024-09-19:
When calling `heap_caps_get_total_size` you get the total of internal memory available as heap, not the total internal memory. The total amount of internal memory is unchanged but the usage of this memory by statically allocated resources will influence the amount of memory left as heap. Certain options will require static memory and this will lead to a total of memory available as heap to decrease.

SoucheSouche, 2024-09-19:
You can use the size tool to compare the memory usage of your application on 4.4 and 5.3. You might notice that the amount of internal memory used on 5.3 is bigger than on 4.4. This can be due to the WIFI related options you updated.

hitecSmartHome, 2024-09-19:
Oh. I see.

So have can I get these memory informations at runtime?

**I want to see these values for both internal and external**

- Total ram
- Total allocated ( with static and dynamic )
- Total free
- Total block

Or it would be even better to see the statically allocated memory and the dinamically allocated memory separatelly for both internal and external

ESP-Marius, 2024-09-19:
@hitecSmartHome The total (heap) ram is based on how much memory is left over after everything that is statically allocated is placed in RAM.  So anything that increases the amount of code placed in IRAM, or statically allocated data in DRAM will decrease the total amound of RAM

hitecSmartHome, 2024-09-19:
**This would be ideal**
- Total ram
- Total statically allocated ( so not available )
- Total dinamically allocated ( so not available )
- Total free left
- Total max block left

SoucheSouche, 2024-09-19:
@hitecSmartHome, I am not sure it is a great idea to create an API to retrieve compile time information. The total RAM information is available in the datasheet of the concerned SoC and the total statically allocated memory can be retrieve using IDF size tool  or by subtracting `heap_caps_get_total_size(MALLOC_CAP_INTERNAL);`  to the total RAM size.

hitecSmartHome, 2024-09-19:
Okay... Thanks for the clarification. It is something