espressif/esp-idf #14601: ESP32-S3 issue on SPI pins

Nezaemmy, 2024-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have been using the 3 chained 74hc595 shift register with ESP32; the system has been working for more than one year.
After switching to ESP32-S3, the system works for 2 to 3 days. The 74HC595 shift register starts misbehaving (for example, if you want to switch on pin 5 of the second chained shift register, pin 3 of the first shift register, and pin 6 of the shift register ON things like that), and with replacing the damaged one, the system works after the same time the issue comes again. The ESP32 doesn't have this issue with 74hc595.

Nezaemmy, 2024-10-09:
I didn't connect to a scope. I don't why the esp32 is working very well but esp32s3 has this issue.
For example, when you swicth ON two output pins shift register 74hc595, after sometime on esp32s3 the output pins of shift register change the status randomly and back to the normal status after 3 seconds. it seems like esp32s3 send what I can call noise

wanckl, 2024-10-09:
@Nezaemmy Hi, can you try to decrease the issue time, then will be convenience to connect scop or debug, 2-3 day is too hard to debug

e.g. Do you use DMA? what the expect data.

Nezaemmy, 2024-10-09:
for more info I use arduino core for the esp32. this is the example code

#include <ShiftRegister74HC595.h>    //https://github.com/Simsso/ShiftRegister74HC595

// create a global shift register object
// parameters: <number of shift registers> (data pin, clock pin, latch pin)
ShiftRegister74HC595<1> sr(11, 10, 12);  //
 
void setup() { 
}

void loop() {

  // setting all pins at the same time to either HIGH or LOW
  sr.setAllHigh(); // set all pins HIGH
  delay(500);
  
  sr.setAllLow(); // set all pins LOW
  delay(500); 
  

  // setting single pins
  for (int i = 0; i < 8; i++) {
    
    sr.set(i, HIGH); // set single pin HIGH
    delay(250); 
  }
  
  
  // set all pins at once
  uint8_t pinValues[] = { B10101010 }; 
  sr.setAll(pinValues); 
  delay(1000);

  
  // read pin (zero based, i.e. 6th pin)
  uint8_t stateOfPin5 = sr.get(5);
  sr.set(6, stateOfPin5);

  // set pins without immediate update
  sr.setNoUpdate(0, HIGH);
  sr.setNoUpdate(1, LOW);
  // at this point of time, pin 0 and 1 did not change yet
  sr.updateRegisters(); // update the pins to the set values
}

Nezaemmy, 2024-10-09:
I tried to report issue on arduino esp32 channel but no response till now

closed 2024-11-02 by Nezaemmy