espressif/esp-idf #14602: WARNING: tool esp32ulp-elf version 2.38_20240113 is installed, but has reported version unknown

mjdswan, 2024-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Successfully run the build script.

### What is the actual behavior?

I have a problem where esp32ulp-elf always reports version unknown.  This began happening on ESP-IDF v5.2.2, but I could usually recover it by running `install.bat`.  It started to become a more persistent problem so I updated to ESP-IDF v5.3.1, however it is still happening.  The log below is only the relevant part of the install script run, it completes but then if I run `export.bat`, it reports the version unknown for esp32ulp-elf and prompts to run `install.bat` again.

`C:\Users\mswan\.espressif\Espressif\frameworks\esp-idf-v5.3.1>install`
Installing ESP-IDF tools
Selected targets are: esp32c2, esp32, esp32s3, esp32c6, esp32c5, esp32c61, esp32p4, esp32h2, esp32c3, esp32s2
Current system platform: win64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@14.2_20240403 (already installed)
Skipping riscv32-esp-elf-gdb@14.2_20240403 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20240530 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20240530 (already installed)
**WARNING: tool esp32ulp-elf version 2.38_20240113 is installed, but has reported version unknown**
Installing esp32ulp-elf@2.38_20240113
file esp32ulp-elf-2.38_20240113-win64.zip is already downloaded
WARNING: destination path already exists, removing
Extracting `C:\Users\mswan\.espressif\Espressif\dist\esp32ulp-elf-2.38_20240113-win64.zip` to `C:\Users\mswan\.espressif\Espressif\tools\esp32ulp-elf\2.38_20240113`
**WARNING: tool esp32ulp-elf version 2.38_20240113 is installed, but reporting version unknown**
Skipping cmake@3.24.0 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20240318 (already installed)
Skipping ninja@1.11.1 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.8 (already installed)
Skipping dfu-util@0.11 (already installed)
Skipping esp-rom-elfs@20240305 (already installed)
Setting up Python environment
Python 3.11.2

### Steps to reproduce.

1. Install ESP-IDF v5.3.1 using offline installer `esp-idf-tools-setup-offline-5.3.1.exe`
2. Open ESP-IDF CMD prompt.
3. Run `idf.py set-target esp32c6` (for example)

jakub-kocka, 2024-09-19:
Hello @mjdswan,
We are sorry for the inconvenience.
Unfortunately, I have not been able to reproduce your issue on the clean Windows installation with just ESP-IDF installed. So it will probably be an issue with your environment.

The installation was probably not fully correct. Can you try to reinstall the ESP-IDF? (https://dl.espressif.com/dl/esp-idf/)

And can you choose another destination for the installation, because the PATH part with the tools `.espressif\Espressif\` doesn't look correct, maybe this might be the issue.

Also please check the `User Environmental Variables` in the `IDF_TOOLS_PATH` - it should be the PATH to the directory where the `tools ` directory is located (the same as you chose during the installation process)

And if it is possible you can try the PowerShell terminal as well - if the behavior will be the same.

dobairoland, 2024-09-23:
> WARNING: tool esp32ulp-elf version 2.38_20240113 is installed, but has reported version unknown

The issue seems to be restricted to the ULP tool only. I could help to go to a clean terminal Window and running the tool manually with only `--version` argument. This usually prints some clue (e.g. missing DLL) which might help to solve the issue.

dobairoland, 2024-09-30:
I'm closing this for inactivity. Please let us know if this is still a valid issue and we'll reopen.

closed 2024-09-30 by dobairoland