espressif/esp-idf #14603: ESP32-S2: v5.3.1 and later: I2C legacy driver clock speed is 80 times slower than requested

dhalbert, 2024-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-3023-g3c99557eee

### Espressif SoC revision.

ESP32-S2 (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Adafruit Metro ESP32-S2 (uses ESP32-S2-WROVER)

### Power Supply used.

USB

### What is the expected behavior?

When using the legacy I2C driver in ESP-IDF v5.3.1 and the latest master (as above), the I2C clock frequency should be measurable as 100 kHz, if that frequency is requested.

### What is the actual behavior?

Instead the I2C clock frequency is 1.25 kHz (a factor of 80 less), as measured by a Saleae.
[image]

### Steps to reproduce.

To test, I used an OLD version of `examples/peripherals/i2c/i2c_tools/` that uses the legacy I2C driver copied from `esp-idf` checked out to v5.2.2: https://github.com/espressif/esp-idf/tree/v5.2.2/examples/peripherals/i2c/i2c_tools. To verify that it's using the legacy driver, look at `i2c_tools/main/cmd_i2ctools.c` and verify it's doing `#include "driver/i2c.h"` (the legacy driver), NOT `#include "driver/i2c_master.h"` (the new driver).

Compile this OLD version of `i2c_tools` (v5.2.2) with a newer version of the ESP-IDF. I used both `v5.3.1` and `v5.4-dev-3023-g3c99557eee`.

In both cases the Saleae trace showed 1.25 kHz, as shown in the screenshot above.

If I try the NEW version of `i2c_tools`, the trace shows the expected clock frequency of 100 kHz. And if It try the OLD version of `i2c_tools` on v5.2.2, it also shows 100 kHz.

I have tested on both ESP32-S2 and ESP32-S3. This problem occurs _only_ on ESP32-S2. ESP32-S3 is fine.

### Debug Logs.

```plain
There's no crash. Here's the test I used to while recording with the Saleae. The prompt is `i2c-tools-OLD` so I can confirm I'm running the right version:

i2c-tools-OLD> i2cconfig --sda=33 --scl=34
i2c-tools-OLD> i2cdetect
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
```
```

### More Information.

This problem arose due to an issue filed on on Adafruit CircuitPython: https://github.com/adafruit/circuitpython/issues/9561.

The issue didn't arise with CircuitPython 9.1.4, which uses ESP-IDF v5.2.2.
It did arise with CircuitPython 9.2.0-beta.0, which uses ESP-IDF v5.3.1.

I also tested requesting an I2C bus speed of 400 kHz, and it measured as 5 kHz, again, 80 times less.

ac0ra, 2025-02-03:
swoboda1337 found this; posting on their behalf.

It appears in master mode the source clock doesn't get set correctly, 
Other varients only use APB, but S2 can use APB_CLK OR REF_CLK, but appears to be default on REF_CLK.

in slave mode
`
        I2C_CLOCK_SRC_ATOMIC() {
            i2c_ll_set_source_clk(i2c_context[i2c_num].hal.dev, src_clk);
        }
`

And

`
#if SOC_I2C_SUPPORT_REF_TICK
        /* On ESP32-S2, APB clock shall always be used in slave mode as the
         * other one, I2C_CLK_SRC_REF_TICK, is too slow, even for sampling a
         * 100KHz SCL. */
        src_clk = I2C_CLK_SRC_APB;
`

From my understanding, this would suggest APB is where timings get configured,
In master mode it would then use REF as the source.
But in Slave mode it would be forced to use APB.

As REF is 1/80th of APB, this results in the much lower clock speeds when in master mode.

swoboda1337, 2025-02-03:
To clarify S2 is the only variant that supports REF_TICK. 

Looking at i2c_ll_set_source_clk it seems to indicate REF_TICK is the default clock as a 0 value would be REF_TICK:
```
static inline void i2c_ll_set_source_clk(i2c_dev_t *hw, i2c_clock_source_t src_clk)
{
    // src_clk : (1) for APB_CLK, (0) for REF_CLK
    hw->ctr.ref_always_on = (src_clk == I2C_CLK_SRC_REF_TICK) ? 0 : 1;
}
```

The function i2c_param_config will configure the clock timings for APB however i2c_param_config only calls i2c_ll_set_source_clk in slave mode, not in master mode. 

An update to i2c_param_config is needed to set the clock source in master mode. A workaround is to configure slave mode first to set the clock source then configure master mode.

closed 2025-02-19 by espressif-bot

swoboda1337, 2025-02-19:
That change wont actually fix it as we discovered later:

    The issue, i2c_ll_master_init resets the ctr reg which contains the clock config: https://github.com/espressif/esp-idf/blob/release/v5.1/components/hal/esp32s2/include/hal/i2c_ll.h#L653
    
    The value for APB clock is 1 and REF_CLK is 0: https://github.com/espressif/esp-idf/blob/release/v5.1/components/hal/esp32s2/include/hal/i2c_ll.h#L628
    
    The function i2c_ll_master_init always switches the clock back to REF_CLK