espressif/esp-idf #14604: 4G personal hotspot native OTA fail, blocked at esp_http_client_read()

zheyuejiang12, 2024-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.0.4

### Espressif SoC revision.

ESP32-D0WD (revision 1)

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

4G personal hotspot OTA successfully

### What is the actual behavior?

4G personal hotspot native OTA fail, blocked at esp_http_client_read()

### Steps to reproduce.

1. idf v4.0.4
2. make native ota example
3. connect 4G personal hotspot 
4. ota fail
...

### Debug Logs.

```plain
st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:12
load:0x3fff0028,len:7268
load:0x40078000,len:15744
ho 0 tail 12 room 4
load:0x40080400,len:3844
entry 0x400806b0
I (79) boot: Chip Revision: 1
I (80) boot_comm: chip revision: 1, min. bootloader chip revision: 0
W (132) rtc_clk_init: Potentially bogus XTAL frequency: 35 MHz, guessing 40 MHz
I (53) boot: ESP-IDF v4.0.4-dirty 2nd stage bootloader
I (53) boot: compile time 09:20:24
I (53) boot: Enabling RNG early entropy source...
[... 858 lines cut ...]
D (12796) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1097760
D (12806) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=512, ridx=0
D (12816) HTTP_CLIENT: http_on_body 512
D (12816) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1097760
D (12826) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=512, ridx=512
D (12826) HTTP_CLIENT: http_on_body 512
D (12836) native_ota_example: Written image length 7168
D (12836) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1097760
D (12846) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=512, ridx=0
D (12856) HTTP_CLIENT: http_on_body 512
D (12856) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1097760
D (12866) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=480, ridx=512
D (12876) HTTP_CLIENT: http_on_body 480
D (12876) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1097760
D (22886) HTTP_CLIENT: need_read=32, byte_to_read=32, rlen=0, ridx=992
D (22886) native_ota_example: Written image length 8160
D (22886) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1097760
D (32886) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=0, ridx=0
D (32886) HTTP_CLIENT: Data processed 8160 != Data specified in content length 1097760
D (32886) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=1097760
```

### More Information.

[sdkconfig.txt](https://github.com/user-attachments/files/17052270/sdkconfig.txt)

nileshkale123, 2024-09-26:
Hello @zheyuejiang12,

Could you please check the connection  status and speed? I also recommend upgrading to the latest IDF version, as support for IDF 4.0.4 has reached End-of-Life (EOL).

zheyuejiang12, 2024-09-28:
@nileshkale123 

status and speed is OK, I can definitely confirm this

Because the current project is being developed on version 4.0.4, there is currently no idea of switching branches, which may cause instability and a lot of workload. Does this issue exist on the current version 4.0.4, what is the reason for it, and is there a solution?

nileshkale123, 2024-09-30:
Hello @zheyuejiang12,

You can test it in the latest IDF using the [simple_ota_example](https://github.com/espressif/esp-idf/tree/master/examples/system/ota/simple_ota_example) with the appropriate configuration.

Feel free to reach out if you encounter any issues.

nileshkale123, 2024-10-15:
I am closing this issue as this support for IDF 4.0.4 is EOL.

closed 2024-10-15 by nileshkale123