espressif/esp-idf #14605: LCD SPI Bus quad data support.

kdschlosser, 2024-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The documentation for version 5.2.2 states that the esp_lcd component using SPI that quad data is not supported. Is the documentation correct when it states this? If it is what does the timeline look like for adding that feature? 

If the documentation is not correct (docs need updating) and it is supported is there anything else that needs to be done for it to work properly?

suda-morris, 2024-09-19:
Hi @kdschlosser The quad SPI LCD IO support was added in https://github.com/espressif/esp-idf/commit/20ebcf76fd17b357e97a8791ad7f272281b7b05f

I believe we need to update the docs to reflect that. Thanks for the report!

You can check out [SPD2010](https://components.espressif.com/components/espressif/esp_lcd_spd2010/versions/1.0.2)  as an example of how to initialize QSPI LCD.

kdschlosser, 2024-09-19:
Just out of question. is there a reason why this setting is not automatically set by checking the bus that was created to see if the data2 and data3 pins were set and if they are then set the quad marker?

suda-morris, 2024-09-20:
Good suggestion. @kdschlosser I think it's doable!

kdschlosser, 2024-09-20:
I think it should be. You should be able to pull the information using the supplied host. same goes for octal. That would remove the need for those 2 "flags". It is also a way to ensure the user has actually has set the needed pins in order for it to work,

kdschlosser, 2024-09-20:
Just out of question... There would only be one scenario where having that flag would serve a purpose. That scenario is if the display doesn't use quad but the user has some other device attached to the bus that does use quad. IDK if you can do that or not.

Can a quad SPI bus be created and there is a quad SPI LCD display attached it that bus and also a normal SPI touch panel? Can the bus be bot quad and standard at the same time??

It would be something that i would not see happening all that often where the display would be standard attached to a quad bus, but it could happen... That is if it is even supported by the ESP-IDF....

closed 2024-09-21 by espressif-bot