espressif/esp-idf #14607: Add compatibility between VScode and C++ code

Bonv1Arnaud, 2024-09-19:
### Is your feature request related to a problem?

Now we can only programme in python ESP from VS code with the espressif library. It will be a good input to add compatibility with C++ code.

### Describe the solution you'd like.

It will be a good input to add compatibility with C++ code.

### Describe alternatives you've considered.

Add reference to follow to allow C++ program with VScode and esspresif environnement

ermacv, 2024-09-20:
Hello!
Have you checked the C++ examples included in the IDF:
https://github.com/espressif/esp-idf/tree/master/examples%2Fcxx

o-marshmallow, 2024-09-25:
Hello @Bonv1Arnaud ,

ESP-IDF supports C++, please check the documentation to find all the details: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/cplusplus.html

sudeep-mohanty, 2024-10-21:
Closing the issue as no update is needed in ESP-IDF for it. Please feel free to open a new ticket for further support. Thanks.

closed 2024-10-21 by sudeep-mohanty