espressif/esp-idf #14609: result of 16bit unsigned arithmetics treated as signed

caspehre, 2024-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

It seems result of unsigned short arithmetics is treated as signed. Consider following code, the u32 example works, but not the u16 example.

volatile unsigned long u32integer1 = 10;
volatile unsigned long u32integer2 = 0xFFFFFFFF;

volatile unsigned short u16integer1 = 10;
volatile unsigned short u16integer2 = 0xFFFF;

void AppTask(void){
    if ((u32integer1 - u32integer2) > 10){
        printf("u32 works\n");
    }else{
        printf("u32 does not work\n");
    }
    if ((u16integer1 - u16integer2) > 10){
        printf("u16 works\n");
    }else{
        printf("u16 does not work\n");
    }
}

igrr, 2024-09-20:
That's the correct observation, but it's not specific to ESP-IDF.

- https://stackoverflow.com/q/57846966
- https://godbolt.org/z/8fMhvqbq5

caspehre, 2024-09-20:
Ok, then I learned something new, I thought it was reasonable to assume second example should work. I guess this issue is closed then?

caspehre, 2024-09-20:
Since this seems to be something not specific to xtensa, I close this issue

closed 2024-09-20 by caspehre

igrr, 2024-09-20:
Yeah, that's just one the many of C language's footguns, unfortunately.