espressif/esp-idf #14610: PyYAML 6.0.2 Invalid Wheel File during ./install.sh

SmartAppsML, 2024-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-rc2-1985-g8858a0e618

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

During ./install.sh the installation routine should download and install the PyYAML file from your server and continue with the next dependency.

### What is the actual behavior?

It fails as the wheel file is invalid:

[image]

We are using a custom fork of esp-idf 5.1 and the ./install.sh usually works. It is performed during our CI/CD pipeline. However, since today it does not work.

### Steps to reproduce.

1. Get the esp-idf version 5.1.
2. Run ./install.sh

fhrbata, 2024-09-20:
Hello, it appears the wheel is functioning properly. It might have been corrupted during the download or something. Is this a one-time issue or does it happen repeatedly?
```sh
$ pip install https://dl.espressif.com/pypi/pyyaml/PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl
Collecting PyYAML==6.0.2
  Downloading https://dl.espressif.com/pypi/pyyaml/PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (751 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 751.2/751.2 kB 11.1 MB/s eta 0:00:00
Installing collected packages: PyYAML
Successfully installed PyYAML-6.0.2
$ wget -O- https://dl.espressif.com/pypi/pyyaml/PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl | sha256sum -
[0] Downloading 'https://dl.espressif.com/pypi/pyyaml/PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl' ...
HTTP response 200  [https://dl.espressif.com/pypi/pyyaml/PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl]
ec031d5d2feb36d1d1a24380e4db6d43695f3748343d99434e6f5f9156aaa2ed  -

```

This doesn't appear to be an issue with esp-idf.

Thank you

SmartAppsML, 2024-09-20:
Hey, thank you for the Response. I tried it multiple times in clean setups
but always fail, even when trying it manually with pip. The other wheels
work somehow.

SmartAppsML, 2024-09-20:
Tried it again on another system and it does not work. What Python version
are you using?

fhrbata, 2024-09-20:
> Tried it again on another system and it does not work. What Python version are you using? 
```
$ python --version
Python 3.10.14
```
Do you get the same sha256sum `ec031d5d2feb36d1d1a24380e4db6d43695f3748343d99434e6f5f9156aaa2ed`  of the downloaded wheel? Also is it possible that there is some proxy? Maybe `curl -v -O https://dl.espressif.com/pypi/pyyaml/PyYAML-6.0.2-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl` may help.

SmartAppsML, 2024-09-23:
Today it worked for some reason. Thanks for your Support!

fhrbata, 2024-09-23:
Hello @SmartAppsML, glad to hear it's working for you and thank you for informing us. I'm closing this issue as it appears the problem has been resolved.