espressif/esp-idf #14615: esp32 module,phone read service  cmd,esp32 no response

luacerbb, 2024-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf_v4.4.4

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

read ble dev service success

### What is the actual behavior?

read ble dev service no response

### Steps to reproduce.

[image]

[image]

wireshark log:

[蓝牙通信异常0828_143459.zip](https://github.com/user-attachments/files/17095739/0828_143459.zip)

esp-zhp, 2024-09-29:
Does this issue occur every time? Could you provide a method to reproduce the problem?

luacerbb, 2024-09-30:
> Does this issue occur every time? Could you provide a method to reproduce the problem?

not occur every time,but once occur,esp32 dev need restart to restore normal.I just test connect and disconnect

rahult-github, 2024-10-03:
Hi @luacerbb , 

 I quickly tried the same at my end, but haven't seen the issue so far. May i request you to please share your sdkconfig also , so we can debug more ?

luacerbb, 2024-10-08:
[sdkconfig.zip](https://github.com/user-attachments/files/17285864/sdkconfig.zip)

rahult-github, 2024-12-18:
Hi @luacerbb ,  release/v4.4 is EOL.  Am closing this issue. If you observe it on any other existing branches, please feel free to reopen it.

closed 2024-12-18 by rahult-github