espressif/esp-idf #14616: Connection Range Issue with Wi-Fi Mesh in SDK 5.3 vs. 5.1.4

ismaia, 2024-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I encountered an issue while running the mesh lite example (no_router)[https://github.com/espressif/esp-mesh-lite/tree/master/examples/no_router]. When building and running both the root and node firmware using SDK 5.3, the node's connection range to the root significantly decreases compared to when the root and node are built using SDK 5.1.4. I performed multiple tests, and all of them consistently showed the same results: Wi-Fi Mesh with SDK 5.3 has a noticeably shorter range than with SDK 5.1.4. he only configuration change between builds was the ROOT/NODE selection in menuconfig.

tswen, 2024-09-24:
Hello, were the testing environments consistent when using SDK 5.1.4 and SDK 5.3? Environmental differences and other external interference may cause some impact. Also, were you testing the maximum distance at which a stable connection could be formed between the two devices?

ismaia, 2024-09-24:
> Hello, were the testing environments consistent when using SDK 5.1.4 and SDK 5.3? Environmental differences and other external interference may cause some impact. Also, were you testing the maximum distance at which a stable connection could be formed between the two devices?

I have separate SDK environments, each with its own tools folder, and I use different load scripts for each one.

I conducted the tests on different days and at least twice a day.
For all tests I erased the entire flash:
idf.py clean build erase-flash flash monitor

and checked SDK version on ESP32 init logs on every test:
ESP-IDF v5.3-dirty 2nd stage bootloader
ESP-IDF v5.1.4-dirty 2nd stage bootloader

The tests were as follows: one ESP32(root) was in a fixed location, and a second ESP32(node) was connected to my PC, with messages being displayed in the terminal as I moved. The path and points were the same.

The first test consisted of taking small steps very slowly until losing the connection (disconnect event). The second test involved stopping at fixed points spaced 1 meter apart and turning on the ESP32 to see if it could connect. In both cases, the results were the same: the SDK version 5.1.4 achieved much better communication range than SDK 5.3.

tswen, 2024-10-10:
Hello, are the 5.1.4 and 5.3 SDK firmware being used on different hardware, or are they on the same hardware? Could you rule out the possibility of hardware being a contributing factor?

ismaia, 2024-10-10:
Hi  @tswen , Yes I conducted all the tests on the exact same hardware: one ESP32 for the node and another ESP32 for the root. Both were flashed with the same firmware, with the only difference being the SDK version.

tswen, 2024-10-21:
In theory, this can also be reproduced using the station and softap demos under IDF, because mesh lite is just an application based on the standard 802.11. Have you tested these two different IDF demos? Do you observe the same issue with the station and softap demos in different versions of IDF

ismaia, 2024-10-22:
Hi @tswen 
I just conducted tests using the SDK 5.1.4 examples for both SoftAP and Station, as you suggested, where the Station connects to the SoftAP. 
I switched the SoftAP to channel 11 and also included the latest stable SDK 5.3.1 in the tests. 
The results were the same: firmware built with SDK 5.1.4 was able to connect from a greater distance (approximately 10 meters further) compared to firmware built with SDK 5.3 or SDK 5.3.1. 
For each build, I executed: idf.py erase-flash flash

zhangyanjiaoesp, 2024-11-04:
@ismaia 
Can you describe your testing method? Did you test it in an office environment? Or was it tested in an open environment?

Can you provide your test data? What is the difference of about 10m you mentioned? Is there a difference between 300m and 290m? Or is it the difference between 20m and 10m?

zhangyanjiaoesp, 2024-11-06:
@ismaia 
What if you put the phy lib (`idf/components/esp_phy/lib/esp32`)under v5.1.4 on v5.3? Will the connection range be the same?

ismaia, 2024-11-06:
Hi @zhangyanjiaoesp ,
I tested the SoftAP (example) indoors, within the office, while the Station was placed outdoors. I walked very slowly with the ESP32 Station connected to the USB port of my PC, observing the log messages until the connection was lost (retrying).

The scenario varied slightly at times, but the results were consistent. The test data consisted of the output from the SoftAP and Station examples. The output was the same in both cases, with the only difference being the distance at which the Station lost its connection.

I did not perform the test of phy lib (idf/components/esp_phy/lib/esp32) of v5.1.4 on v5.3. I'll try it tomorrow

AxelLin, 2024-11-07:
@zhangyanjiaoesp 
https://github.com/espressif/esp-idf/issues/14797 might be related?

ismaia, 2024-11-07:
@zhangyanjiaoesp , Yes I think #14797  is related , the behavior seems similar

zhangyanjiaoesp, 2024-11-08:
@ismaia thanks for the feedback, we will check the phy lib version

zhangyanjiaoesp, 2024-11-08:
@ismaia 
1. > What is the difference of about 10m you mentioned? Is there a difference between 300m and 290m? Or is it the difference between 20m and 10m?

    Could you please answer this question?
2. Were you testing v5.3 and v5.1.4 on the same channel? 
3. Did you use deep sleep?

Our colleagues mentioned that there haven't been any changes to the PHY library regarding RSSI in these two versions, so we need to confirm the above details with you.

ismaia, 2024-11-08:
@zhangyanjiaoesp 
1.When I mentioned a difference of around 10 meters, I meant that in SDK 5.1.4, the distance between the station and softAP extended approximately 10 meters further than in SDK 5.3 in the test environment (outdoor). In other words, with SDK 5.3, the station and softAP need to be 10 meters closer to establish a connection compared to SDK 5.1.4.
2. Yes. Channel 11 on all tests
3. No, I used the examples SoftAP and Station from SDK 5.3, these examples don't use deep sleep If I am not mistaken.

zhangyanjiaoesp, 2024-11-08:
@ismaia 
I understand your point now. You mentioned that v5.3 has a 10m shorter connection range compared to v5.1.4, but I believe you may not fully understand what I mean. I would like you to provide specific numbers. For example, if the connection range is 300m on v5.1.4 and 290m on v5.3, we would consider that 10m difference to be relatively minor. However, if the connection range is 20m on v5.1.4 and 10m on v5.3, then that 10m difference becomes significant. Does this clarify what I'm asking? Could you share the specific values you observed in your tests?

ismaia, 2024-11-08:
@zhangyanjiaoesp ,
I got the idea. The maximum range was around 30m because the softAP was placed indoors, and I walked with the Station + PC outdoors. I used two ESP32 DevKitC boards with PCB antennas. 
With SDK 5.3, the maximum range was around 20m, while with SDK 5.1, it was 30m.

AxelLin, 2024-11-19:
> @zhangyanjiaoesp , I got the idea. The maximum range was around 30m because the softAP was placed indoors, and I walked with the Station + PC outdoors. I used two ESP32 DevKitC boards with PCB antennas. With SDK 5.3, the maximum range was around 20m, while with SDK 5.1, it was 30m.

@ismaia Any chance to check SDK5.2 for comparison?

zhangyanjiaoesp, 2025-01-07:
@ismaia 
This test scenario does not seem to provide conclusive evidence, as the antenna angles during the two tests might differ. Additionally, factors such as people moving around or other environmental interference could affect the test results. Internally, we conducted tests in a shield box and did not observe any significant differences between v5.3 and v5.1.4.

zhangyanjiaoesp, 2025-02-13:
@ismaia Do you still have this problem? Can wen close it?

ismaia, 2025-02-13:
Hi @zhangyanjiaoesp,
I'm still using v5.1.4 to develop my firmware as v5.3 gets unstable when I'm using WiFi mesh. I'll wait new SDK versions to re-test my firmware.
Thanks

zhangyanjiaoesp, 2025-05-08:
@ismaia Have you retest it on v5.3.3?

Alvin1Zhang, 2025-07-01:
Thanks for reporting, will close due to short of updates, feel free to reopen if the issue still happens.

closed 2025-07-01 by Alvin1Zhang