espressif/esp-idf #14618: extend example HCI BLE controller on ESP32C3

jkorenj, 2024-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, 

I would like to extend ESP-IDF UART HCI Controller example.
I would like to extend the controller commands with custom vendor specific commands, but I can't find a way to do this. 

I will add a custom command on the host side (STM32), but I need to add it also on the ESP32C3 side, so I can make some action when parsing this commands. 

Thank you for your help.

esp-zhp, 2024-09-28:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/hci/controller_hci_uart_esp32c3_and_esp32s3

jkorenj, 2024-10-10:
Thank you. I already enabled that, but I still don't understand how can I make a custom response to a vendor opcode. 
When I send vendor specific code, stack automatically responds to it, and I don't know how to change the response. 

Thank you for your help.

esp-xzk, 2024-10-10:
>how can I make a custom response to a vendor opcode

Making a custom response would require altering Controller code, which we don't recommend.
>are there are vendor specific commands that are already implemented?

We do have some of those. May I ask what sort of commands do you need? What specific functions are you looking for? 
Thank you.

jkorenj, 2024-10-10:
Hello, 

I would like for example to read current firmware version on the module. And I would like to update the firmware through the same UART that it is used for HCI communication. 

That is why I wanted to implement a few more commands that I would call as normal HCI commands

Thank you.

esp-xzk, 2024-10-12:
Sorry for the late reply. I have checked with my colleagues and unfortunately neither of these options are available at this time.

jkorenj, 2024-10-16:
OK. Thank you.

closed 2024-12-24 by Alvin1Zhang