espressif/esp-idf #14623: GCOV not working

gjs-1, 2024-09-24:
hello,
i am working on implementing GCOV in ESP32,
and i got an example code of GCOV given in SDK and i am using ESP32-wroom-32E,
and i am using external j-link with OpenOCD version v0.12.0-esp32-20240318.

when i follow the steps given in example code readme file,
https://github.com/espressif/esp-idf/blob/master/examples/system/gcov/README.md 
so when i try send command **esp gcov dump** in telnet i got this error,

 
[image]

any solution for this.

kdaksh8, 2024-09-26:
Hello,
Same issue for ESP32C3 was faced by me
Issue can be solved by running OpenOCD with help of in built JTAG of ESP32C3, simply debug the Code in IDE and open the telnet connection in Different IDF terminal as you have done here, Parallelly observe logs and follow the steps in the README file. 
After running telnet session and executing "esp gcov dump" command you should get following output.
[image]

Once you are done with it, _.gcda_ file will be generated in build directory, now to generate Coverage Report run "cmake --build build/ --target gcovr-report", make sure you have installed _gcovr_ with command "pip install gcovr". These commands are to be run in IDF terminal.

erhankur, 2024-09-26:
@gjs-1 As I see from the screenshot, your first problem is related to OpenOCD connection.. Please check the wires to the Jlink. And your OpenOCD script must be something like 

```
openocd -f tcl/interface/jlink.cfg -f tcl/target/esp32.cfg
```

gjs-1, 2024-09-27:
thank you @kdaksh8, I have tried for my ESP kit and it works perfectly.

closed 2024-10-03 by gjs-1