espressif/esp-idf #14630: MQTT and Multiple Connections for Transmission Wifi

allacmc, 2024-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Teste

euripedesrocha, 2024-09-25:
Hi @allacmc could you share the LOG with the errors you are facing and idf version in use?

allacmc, 2024-09-26:
More details in POST this fórum:

https://forums.freertos.org/t/mqtt-and-multiple-connections-for-transmission-wifi/21515

euripedesrocha, 2025-02-05:
Closing this as looks related to application memory management. 
Feel free to reopen adding extra information that points to idf related issue/