espressif/esp-idf #14635: install.sh ignores target

higaski, 2024-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I'd expect the install script to only install the selected target (as documented).

### What is the actual behavior?

It seems that with a recent version of the install script, selecting a target has absolutely no impact on what actually get installed.

```sh
vinci@vinci-VirtualBox:~/esp/esp-idf$ ./install.sh esp32s3
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.3
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Selected targets are: esp32c2, esp32h2, esp32p4, esp32c6, esp32, esp32c61, esp32s3, esp32c5, esp32c3, esp32s2
```

```sh
vinci@vinci-VirtualBox:~/esp/esp-idf$ ./install.sh esp32
Detecting the Python interpreter
Checking "python3" ...
Python 3.12.3
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Selected targets are: esp32c61, esp32c5, esp32c2, esp32c3, esp32s3, esp32h2, esp32c6, esp32, esp32p4, esp32s2
```

### Steps to reproduce.

1. Follow the [installation guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/linux-macos-setup.html)
2. When getting to the point where the install scripts are run, try different targets

dobairoland, 2024-09-26:
Hi @higaski. I'm afraid that this is the intended behavior as the install script only installs and is not uninstalling anything (which you would expect to happen).

To explain the behavior, here is an example
```sh
./install.sh esp32 
./install.sh esp32s3 esp32c3   # it won't uninstall esp32
./install.sh    # keep updating esp32, esp32s3 and esp32c3
```

If you install tools for all chips then running `install.sh` consequently won't uninstall them. This is what you observe.

You can clean up "targets" by manually editing `~/.espressif/idf-env.json` (or elsewhere if you moved the Espressif directory) because at the moment we don't provide uninstall scripts.

higaski, 2024-09-26:
Oh I see. I'm sorry then. I must have have called the script with `all` as argument the very first time and then canceled the running installation process. That would explain why no matter what I throw at it now it keeps telling me all targets are already selected.

I've just cleared and re-installed the framework and only called the script with `esp32s3` as argument and now it works as expected as in only that target is selected and no others.

Thanks for clearing that up!

closed 2024-09-26 by higaski