espressif/esp-idf #14638: CONFIG_USJ_ENABLE_USB_SERIAL_JTAG can't be disabled

higaski, 2024-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should be possible to enable/disable CONFIG_USJ_ENABLE_USB_SERIAL_JTAG.

### What is the actual behavior?

Is it per design that currently the `CONFIG_USJ_ENABLE_USB_SERIAL_JTAG` option (btw, is this a typo or a wordplay on JTAG?) can't be disabled on ESP32-S3 devices?

[image]

### Steps to reproduce.

1. Set up any ESP32-S3 example
2. Try to change the CONFIG_USJ_ENABLE_USB_SERIAL_JTAG option using idf.py menuconfig

igrr, 2024-09-27:
It can't be disabled because it gets enabled by another option. You can see which one if you press '?', one of the things shown there will be:
```
Symbols currently y-selecting this symbol:
  - ESP_CONSOLE_USB_SERIAL_JTAG_ENABLED
```

and going to that option and pressing '?' again we get
```
Default:
  - y
    Condition (=y):
       ESP_CONSOLE_USB_SERIAL_JTAG(=n)
    || ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG(=y)
```

So basically, CONFIG_USJ_ENABLE_USB_SERIAL_JTAG gets selected if USB_SERIAL_JTAG is selected as either primary or secondary console output. If you change the console output setting to UART, then you should be able to disable `CONFIG_USJ_ENABLE_USB_SERIAL_JTAG` as well.

igrr, 2025-06-17:
@roma-jam since you have looked into the linked issue, could you please check this one as well?

roma-jam, 2025-07-28:
hi @higaski, 

> it seems that switching the device class "on-thy-fly" during initialization somehow confuses Windows 11 and you can end up with no device showing at all.

the USB OTG and USB JTAG share the PHY indeed, but could you please clarify the description: 

>  USB does no longer work for her on Windows 11 at all

Does the user uses the USB OTG functionality? If yes, then which class and Host or Device? If Device, then please share the version of the esp-tinyusb component.

I will check some things and return back ASAP.

roma-jam, 2025-07-29:
HI @higaski ,

> I'm sorry I'm not sure I understand.

sorry for not a clear question, but you got it: yes, the idea was to have the possibility to verify the scenario on Win11. So, technically, there is an option to ask the user to add several lines in the esp-idf code, build and flash it and then connect to the Win11. Am I right? 

> It's` a single USB C receptacle, the board is open source so you can see for yourself here:

I have the M5 Stamp module, which has pretty much the same idea where the USB port connector is shared between the USB OTG and USB JTAG. But unfortunately, I am not able to reproduce the problem. But if this related to the same problem as on Mac, the difference in the hardware could also be important.   

As far as I understand, we have two different not related problems so far : 
- Unable to completely disable the USB JTAG
- USB function doesn't changed from the USB JTAG to the USB OTG, when it suppose to do so.

Please, correct me if I am wrong.

roma-jam, 2025-07-29:
Hi @higaski , 

> I also wonder if either #15912 or #15932 is related to this?

Highly unlikely, as for this specific case, we need to change the PHY configuration from the USB OTG to JTAG with `usb_phy_new()` call. 

Ok, let's try to eliminate the obvious changes and is that possible to add pull-up override back for the device in here (line 172): 
https://github.com/espressif/esp-idf/blob/b66b5448e0dfac6be135e6f0ac21010aba231180/components/usb/usb_phy.c#L172

So, here is the diff of the changes that need to be added: 

```
diff --git a/components/usb/usb_phy.c b/components/usb/usb_phy.c
index 9432b446cb..41cba55864 100644
--- a/components/usb/usb_phy.c
+++ b/components/usb/usb_phy.c
@@ -170,6 +170,14 @@ esp_err_t usb_phy_otg_set_mode(usb_phy_handle_t handle, usb_otg_mode_t mode)
         esp_rom_gpio_connect_in_signal(GPIO_MATRIX_CONST_ONE_INPUT,  otg_sig->bvalid, false);     // HIGH to force USB device mode
         esp_rom_gpio_connect_in_signal(GPIO_MATRIX_CONST_ONE_INPUT,  otg_sig->vbusvalid, false);  // receiving a valid Vbus from device
         esp_rom_gpio_connect_in_signal(GPIO_MATRIX_CONST_ZERO_INPUT, otg_sig->avalid, false);
+        // Override pull resistors for FS device
+        usb_wrap_pull_override_vals_t vals = {
+            .dp_pu = true,
+            .dm_pu = false,
+            .dp_pd = false,
+            .dm_pd = false,
+        };
+        usb_wrap_hal_phy_enable_pull_override(&handle->wrap_hal, &vals);
     }
```

Please, add the lines, build and flash project. Then try to connect the device to the Win11 Host and share the result. 
Sorry for the inconvenience and thanks for the collaboration.

roma-jam, 2025-08-05:
Hi @higaski, 

just for info: are you using the light sleep mode in you project? 

Otherwise, there could be a potential requirement to be able to configure these pull resistors based on the hardware specifics.
We will investigate the better way how to make it available and do that asap. 

I'll rename the issue to something which reflects the nature of the problem and keep it open.
Thanks.

tore-espressif, 2025-08-26:
@higaski do you have any updates on this?
I reviewed the schematic you shared and everything looks correct. I also reproduced your setup locally, including `.self_powered = true`, and tested on Windows 11 with both USB-A and USB-C host ports — everything worked as expected.

For your hardware, you need to set:
``` c
.vbus_monitor_io = 7
```

tore-espressif, 2025-09-03:
@higaski I could not reproduce the problem on S3 with default settings. Could you share your full `sdkconfig` file so we could check if there is something else that forces USB-Serial-JTAG on?

For your reference, this section of code disables USB-Serial-JTAG on startup in case it was disabled https://github.com/espressif/esp-idf/blob/2044fba6e71422446986f9ae0909b1ab67e57815/components/esp_system/port/soc/esp32s3/clk.c#L278
you can check if this code is executed in your app eg. by placing warning pragma in your code
``` c
#warning "USB-Serial-JTAG clock and pad disabled"
```

tore-espressif, 2025-09-03:
If your device shows up as debug unit in your application code, make sure that your app does not call `usb_serial_jtag_ll_enable_bus_clock(true)` or `usb_serial_jtag_ll_phy_enable_pad(true)` anywhere.

> yet my device always initializes JTAG during boot

If it shows up as debug unit only during the 1st and 2nd stage bootloader (but does not once the your application code starts), you can burn
`EFUSE_DIS_USB_SERIAL_JTAG` eFuse to permanently disable USB-Serial-JTAG peripheral. Please note that this step is irreversible

tore-espressif, 2025-09-03:
Yes, the USB-Serial-JTAG is enabled by default in the bootloader, so the name is confusing.

In this case we won't rename the symbol (it could break some user's confiurgation) but we can provide better description

closed 2025-09-20 by espressif-bot