espressif/esp-idf #14639: MIPI DSI LCD Panel Example: how to use ILI9881C panel in landscape orientation?

dmartauz, 2024-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I attempted to run following example on ESP32-P4 with ILI9881C display:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/mipi_dsi

To make it work, I had to comment out part of esp_lcd_ili9881c driver which queries ID as it got stuck at `esp_lcd_panel_io_rx_param` :
```
    uint8_t ID1, ID2, ID3;
    esp_lcd_panel_io_tx_param(io, ILI9881C_CMD_CNDBKxSEL, (uint8_t[]) {
        ILI9881C_CMD_BKxSEL_BYTE0, ILI9881C_CMD_BKxSEL_BYTE1, ILI9881C_CMD_BKxSEL_BYTE2_PAGE1
    }, 3);
    esp_lcd_panel_io_rx_param(io, 0x00, &ID1, 1);
    esp_lcd_panel_io_rx_param(io, 0x01, &ID2, 1);
    esp_lcd_panel_io_rx_param(io, 0x02, &ID3, 1);
    ESP_LOGI(TAG, "ID1: 0x%x, ID2: 0x%x, ID3: 0x%x", ID1, ID2, ID3);
```
After that I managed to run the example which was displayed in portrait mode.
[image]

I then attempted to rotate the display using `esp_lcd_panel_swap_xy()` with following result:
```
E (1658) lcd_panel: esp_lcd_panel_swap_xy(50): swap_xy is not supported by this panel
```
I also tried to rotate the screen at LVGL level using `lv_display_set_rotation()` but this attempt failed as well.

IN LVGL documentation I found: In lack of hardware display rotation support `lv_draw_sw_rotate` can be used to rotate the buffer in the flush_cb.
However I am not sure how to implement it correctly. Could you please provide an example?

suda-morris, 2024-09-28:
duplicated with https://github.com/espressif/esp-bsp/issues/400

espzav, 2024-11-12:
Fixed in https://github.com/espressif/esp-bsp/issues/400

closed 2024-11-12 by espzav