espressif/esp-idf #14643: Error Uploading Code to ESP32-WROVER-E: Failed to execute script 'esptool' due to unhandled exception!

invictus-anic3tus, 2024-09-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi! Firstly, thank you for taking the time to comment and help me out here. The issue that I'm getting seems to be fairly common, but I have tried many of the more popular solutions, and obviously, nothing has worked.

I'm using an ESP32-WROVER-DEV (-WROVER-E chip) and Arduino IDE. I've updated my ESP board to v3.0.5, and Arduino is also at the latest.

My issue: Previously, while uploading sketches (blank or full) I'd get this error:

```
Sketch uses 298193 bytes (9%) of program storage space. Maximum is 3145728 bytes.
Global variables use 20440 bytes (6%) of dynamic memory, leaving 307240 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.6
Serial port COM5
Connecting.....Traceback (most recent call last):
  File "esptool.py", line 37, in <module>
  File "esptool\__init__.py", line 1064, in _main
  File "esptool\__init__.py", line 859, in main
  File "esptool\cmds.py", line 466, in write_flash
  File "esptool\util.py", line 37, in flash_size_bytes
TypeError: argument of type 'NoneType' is not iterable
[33448] 
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:b2:1c:51:fd:84
Uploading stub...
Running stub...
Stub running...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Configuring flash size...
Failed to execute script 'esptool' due to unhandled exception!
```

... in the output. Strangely enough, after unplugging and replugging the esp, I'm getting a successful upload, but

```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
csum err:0xa1!=0x6c
ets_main.c 384 
ets Jul 29 2019 12:21:46
```
is being spammed in the Serial Monitor. I've changed the Upload Speed, the partition speed is at Huge APP, I've verified the USB cord, COM port, drivers and board. I'm at a loss. Any help or pointer in the right direction is very much appreciated! If I missed out on any information, just ask.

Thanks!
~Anicetus

invictus-anic3tus, 2024-09-28:
Ah. Reuploaded sketch. First issue is back! This is my full output after uploading (again):

```
Sketch uses 289665 bytes (9%) of program storage space. Maximum is 3145728 bytes.
Global variables use 20296 bytes (6%) of dynamic memory, leaving 307384 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.6
Serial port COM5
Connecting.....Traceback (most recent call last):
  File "esptool.py", line 37, in <module>
  File "esptool\__init__.py", line 1064, in _main
  File "esptool\__init__.py", line 859, in main
  File "esptool\cmds.py", line 466, in write_flash
  File "esptool\util.py", line 37, in flash_size_bytes
TypeError: argument of type 'NoneType' is not iterable
[4072] Failed to execute script 'esptool' due to unhandled exception!

Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:b2:1c:51:fd:84
Uploading stub...
Running stub...
Stub running...
WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs.
Configuring flash size...
```
No GPIOs or any pins are connected. Flash Mode QIO, Frequency 80 MHz.

This is being spammed in Serial Monitor now:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46
```

invictus-anic3tus, 2024-09-29:
The issue persists across computers. I have tried on a Windows 10 and an old Mac 12.7.6 (Monterey). I'll make sure to update you soon on whether the problem is also across ESPs. I have an ESP32-WROOM that I will test.

Any help much appreciated!

invictus-anic3tus, 2024-09-29:
The issue is not across different ESP modules. Granted, the one I tested successfully with is a different chip. I tested yet another USB cord. No functioning. But for some reason, (sorry for being confusing) the upload is completely successful again, but with the 
```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
csum err:0x1f!=0x6c
ets_main.c 384 
ets Jul 29 2019 12:21:46
```
being spammed again.

radimkarnis, 2024-09-30:
Hello @invictus-anic3tus, 
I believe your issue is caused by hardware. The ESP on-board of your devkit seems to be working (it communicates with esptool and prints data to the serial console).  The issue is with the flash memory chip:

> WARNING: Failed to communicate with the flash chip, read/write operations will fail. Try checking the chip connections or removing any other hardware connected to IOs

This message from [esptool](https://github.com/espressif/esptool) is printed when the flash memory chip doesn't respond (it is interrogated during the connection process to verify if it's connected properly). 

> csum err:0x1f!=0x6c

This message from the first stage bootloader happens when the contents of the flash memory do not match with what's expected. Meaning the data is corrupted.

> invalid header: 0xffffffff

This is again a message from the first stage bootloader, but this time it means the flash memory is completely empty (all `0xff`).

> the upload is completely successful again

None of your logs show a successful upload. Do you see a confirmation from esptool like `Wrote 25536 bytes (15935 compressed) at 0x00001000 in 0.7 seconds (effective 275.5 kbit/s)... Hash of data verified.` ?

All of the above-mentioned cases point to a faulty flash chip. Either it is somehow damaged, or it is write-protected. If you have two units - one of them working and one of them not - you can try running the [`read_flash_status`](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/advanced-commands.html#read-flash-chip-registers-read-flash-status) esptool command and compare the output between the two. If the outputs don't match, this might be a sign of write protection bits being turned on.

invictus-anic3tus, 2024-09-30:
Thank you, @radimkarnis for helping me out!

As for a successful upload, I did previously get the `Wrote 25536 bytes...` line, but currently am getting this error:

```
Sketch uses 289645 bytes (22%) of program storage space. Maximum is 1310720 bytes.
Global variables use 20296 bytes (6%) of dynamic memory, leaving 307384 bytes for local variables. Maximum is 327680 bytes.
"/Users/morethermac2/Library/Arduino15/packages/esp32/tools/esptool_py/4.6/esptool" --chip esp32 --port "/dev/cu.usbserial-1410" --baud 921600  --before default_reset --after hard_reset write_flash  -z --flash_mode keep --flash_freq keep --flash_size keep 0x1000 "/private/var/folders/hx/5jddqb7x7zd4246777bw1fh80000gq/T/arduino/sketches/C6976A51AF9D62618F0A9A2077252A0E/TestUpload2.ino.bootloader.bin" 0x8000 "/private/var/folders/hx/5jddqb7x7zd4246777bw1fh80000gq/T/arduino/sketches/C6976A51AF9D62618F0A9A2077252A0E/TestUpload2.ino.partitions.bin" 0xe000 "/Users/morethermac2/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/tools/partitions/boot_app0.bin" 0x10000 "/private/var/folders/hx/5jddqb7x7zd4246777bw1fh80000gq/T/arduino/sketches/C6976A51AF9D62618F0A9A2077252A0E/TestUpload2.ino.bin" 
esptool.py v4.6
Serial port /dev/cu.usbserial-1410
Connecting......
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:b2:1c:51:fd:84
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.

A fatal error occurred: Unable to verify flash chip connection (Packet content transfer stopped (received 7 bytes)).
```
It's hard to exactly tell what is happening (at least for me) because it seems that every time I upload code, a different error is presented. Whenever I was getting the `invalid header: 0xfffffff` line spammed in the serial monitor, I did get a successful upload, including the `Wrote... Hash of data verified` line. It's pretty confusing.

As for the `read_flash_status` command, this is my output on the functioning ESP:

```
esptool.py v4.8.1
Found 2 serial ports
Serial port /dev/cu.usbserial-1410
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:b2:1c:42:18:fc
Uploading stub...
Running stub...
Stub running...
Status value: 0x0200
Hard resetting via RTS pin...
```

And this is the output on the malfunctioning one:

```
esptool.py v4.8.1
Found 2 serial ports
Serial port /dev/cu.usbserial-1410
Connecting.......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b0:b2:1c:51:fd:84
Uploading stub...
Running stub...
Stub running...
Status value: 0x0200
Hard resetting via RTS pin...
```

What do you think?

invictus-anic3tus, 2024-09-30:
P.S: If I do need to buy a new ESP32, what would be a good option that is:

1: Small and lightweight (external antenna fine, if it makes the board smaller and is included)
2: Has at least 20 available GPIOs
3: No camera
4: Wifi

Thank you!

invictus-anic3tus, 2024-10-01:
@radimkarnis, one last thing: if the ESP is permanently damaged, why did it happen, and how do I ensure it doesn't happen with another?

Thanks again.

radimkarnis, 2024-10-01:
> What do you think?

Both of the flash chips show `Status value: 0x0200`, so it's not write protection or any other internal settings difference between them. 

> If I do need to buy a new ESP32, what would be a good option

There are many devkits offered by us and also third-party companies. You can compare Espressif devkits [here](https://www.espressif.com/en/products/devkits). Adafruit Feather series is very popular, or you can search aliexpress for other cheap solutions. Your requirements are not very restricting, there will be many options.

> if the ESP is permanently damaged, why did it happen, and how do I ensure it doesn't happen with another?

The ESP is not damaged. The flash chip shows signs of damage. Maybe it's just a soldering error (one of the legs can have a cold solder joint) or it's a manufacturing error. If you are skilled with soldering iron, you can even replace the flash chip.

I don't think this is your mistake. The only thing I can think of happening is maybe accidental shorting of some pins or applying high voltage to them.

With the current prices of esp devkits (some can go for under 2 USD) I would advise you just getting a new one.

invictus-anic3tus, 2024-10-01:
@radimkarnis thanks for your help! I'm glad to know that it wasn't my mistake. By the way, what exactly is the flash chip?

Thanks yet again,
~Anicetus

radimkarnis, 2024-10-01:
The flash chip on an odule is non-volatile memory used to store the firmware and program code that runs on the ESP32. 
When you upload a program (or "sketch" in the Arduino IDE), it gets written to this flash memory. The ESP32 then reads from this memory to execute the code when it starts up.

In simpler terms, it's like a mini hard drive inside the ESP32 where your program is saved. If this chip is damaged or faulty, it can cause issues with uploading, running, or retaining the code. That seems to be your case.

The flash chip is typically soldered onto the development board alongside the ESP32 module. It’s a separate integrated circuit (IC) that communicates with the ESP32 through a serial interface, usually SPI (Serial Peripheral Interface).

On many ESP32 development boards, including the ESP32-WROVER, you can visibly see the flash memory chip as a small square or rectangular component on the board (under the metal shielding). 

[image]

In some compact or embedded modules, the flash might be stacked on top of the microcontroller in a system-in-package (SiP) format, but it's still technically an external component.

I am now going to close this ticket because this doesn't seem to be a software issue.

closed 2024-10-01 by radimkarnis