espressif/esp-idf #14645: Failed to resolve component 'esp_littlefs'

sova32, 2024-09-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

builded project

### What is the actual behavior?

`Executing task: /usr/bin/cmake -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2/build -S=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2 

-- Found Git: /usr/bin/git (found version "2.46.0")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/vadiki/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/vadiki/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/vadiki/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: не найден git репозиторий (или один из его каталогов вплоть до точки монтирования /run/media/vadiki)
Останавливаю поиск на границе файловой системы (так как GIT_DISCOVERY_ACROSS_FILESYSTEM не установлен).'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 2 dependencies:
[1/2] joltwallet/littlefs (1.14.8)
[2/2] idf (5.3.1)
CMake Error at /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:268 (message):
  Failed to resolve component 'esp_littlefs'.
Call Stack (most recent call first):
  /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:304 (__build_resolve_and_add_req)
  /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:607 (__build_expand_requirements)
  /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!

 *  The terminal process "/usr/bin/cmake '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2/build', '-S=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2'" terminated with exit code: 1. `

### Steps to reproduce.

1. idf.py add-dependency --component fileSystem joltwallet/littlefs==1.14.8
2. #include "esp_littlefs.h" in my fileSystem.c (at tha screenshot)
3. click "build project" in VSCode.
[image]

### Build or installation Logs.

```plain
`Executing task: /usr/bin/cmake -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2/build -S=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2 

-- Found Git: /usr/bin/git (found version "2.46.0")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/vadiki/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/vadiki/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/vadiki/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: не найден git репозиторий (или один из его каталогов вплоть до точки монтирования /run/media/vadiki)
Останавливаю поиск на границе файловой системы (так как GIT_DISCOVERY_ACROSS_FILESYSTEM не установлен).'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 2 dependencies:
[1/2] joltwallet/littlefs (1.14.8)
[2/2] idf (5.3.1)
CMake Error at /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:268 (message):
  Failed to resolve component 'esp_littlefs'.
Call Stack (most recent call first):
  /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:304 (__build_resolve_and_add_req)
  /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:607 (__build_expand_requirements)
  /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!

 *  The terminal process "/usr/bin/cmake '-G=Ninja', '-DPYTHON_DEPS_CHECKED=1', '-DESP_PLATFORM=1', '-B=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2/build', '-S=/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2'" terminated with exit code: 1. `
```

nopnop2002, 2024-09-29:
There is an official sample here.

https://github.com/espressif/esp-idf/tree/master/examples/storage/littlefs

igrr, 2024-09-30:
> CMake Error at /home/vadiki/esp/v5.3.1/esp-idf/tools/cmake/build.cmake:268 (message):
> Failed to resolve component 'esp_littlefs'.

Could you please check if any other CMakeLists file in your project references `esp_littlefs` component? If yes, please remove that reference and try to build again. Also, as @nopnop2002 pointed out, please try the example and see if that works.

closed 2024-10-30 by adokitkat