espressif/esp-idf #14646: 程序正常运行时出现mqtt和http无法连接，重连失败现象：esp-tls: Failed to open new connection；transport_base: Failed to open a new connection；HTTP_CLIENT: Connection failed, sock < 0

gl-LanYH, 2024-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[19:41:28.492] D (206933) wifi_init_default: Got IP wifi default handler entered[0m
[19:41:28.492] D (206933) esp_netif_handlers: esp_netif action got_ip with netif0x3fcba1dc from event_id=0[0m
[19:41:28.492] [0;32mI (206943) esp_netif_handlers: sta ip: 192.168.88.194, mask: 255.255.255.0, gw: 192.168.88.1[0m
[19:41:28.492] D (206953) HTTP_CLIENT: Begin connect to: https://gslb.lafaer.cn:443[0m
[19:41:28.492] D (206963) esp-tls: host:gslb.lafaer.cn: strlen 14[0m
[19:41:28.492] D (206963) esp-tls: [sock=54] Resolved IPv4 address: 120.24.55.11[0m
[19:41:28.492] D (206973) esp-tls: [sock=54] Connecting to server. HOST: gslb.lafaer.cn, Port: 443[0m
[19:41:30.210] D (208723) esp-tls: handshake in progress...[0m
[19:41:32.757] D (211233) mqtt_client: Reconnecting...[0m
[19:41:32.762] D (211233) event: running post MQTT_EVENTS:7 with handler 0x420114a6 and context 0x3fca6f30 on loop 0x3fca7a48[0m
[19:41:32.762] [1;32m./main/src/gl_mqtt.c[191]: Other event id:7[0m
[19:41:32.762] D (211233) esp-tls: host:120.24.55.11: strlen 12[0m
[19:41:32.762] D (211243) esp-tls: [sock=55] Resolved IPv4 address: 120.24.55.11[0m
[19:41:32.762] D (211243) esp-tls: [sock=55] Connecting to server. HOST: 120.24.55.11, Port: 8883[0m
[19:41:35.137] D (213653) esp-tls: handshake in progress...[0m
[19:41:37.113] [0;33mW (215293) esp-tls: Failed to open new connection in specified timeout[0m
[19:41:37.122] [0;31mE (215293) transport_base: Failed to open a new connection[0m
[19:41:37.122] [0;31mE (215293) HTTP_CLIENT: Connection failed, sock < 0[0m
[19:41:37.122] D (215293) HTTP_CLIENT: HTTP_EVENT_ERROR[0m
[19:41:37.122] D (215303) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:0 posted to loop 0x3fca9fb8[0m
[19:41:37.122] [0;31mE (215313) HTTP_CLIENT: HTTP POST request failed: ESP_ERR_HTTP_CONNECT[0m
[19:41:37.122] [0;32mI (215313) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED[0m
[19:41:37.122] [0;32mI (215323) HTTP_CLIENT: Last esp error code: 0x8006[0m
[19:41:37.122] [0;32mI (215333) HTTP_CLIENT: Last mbedtls failure: 0x0[0m
[19:41:37.122] D (215333) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:6 posted to loop 0x3fca9fb8[0m
[19:41:37.122] D (215343) event: running post BLE_NETWORK_EVENT:3 with handler 0x42014e30 and context 0x3fcba338 on loop 0x3fca9fb8[0m
[19:41:37.122] [1;32m./main/src/gl_network.c[329]: **********BLE_RPT_NETWORK_DISCONNECT**********[0m
[19:41:37.122] [1;32m./main/src/gl_network.c[345]: continue goto network[0m
[19:41:37.122] D (215363) wifi:Start wifi disconnect
[19:41:37.122] I (215373) wifi:state: run -> init (0)
[19:41:37.122] D (215373) wifi:connect status 5 -> 6
[19:41:37.122] D (215383) wifi:stop beacon/connect timer, send diassoc(8)
[19:41:37.122] D (215383) wifi:sta leave
[19:41:37.122] I (215383) wifi:pm stop, total sleep time: 2665863 us / 10177093 us
[19:41:37.122] 
[19:41:37.122] D (215393) wifi:stop CSA timer
[19:41:37.122] I (215393) wifi:<ba-del>idx:1, tid:0
[19:41:37.122] I (215393) wifi:<ba-del>idx:0, tid:7
[19:41:37.122] D (215403) wifi:remove 76:07:a6:f0:a8:4f from rc list
[19:41:37.122] I (215403) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
[19:41:37.122] D (215413) wifi:filter: set rx policy=8
[19:41:37.122] D (215413) wifi:Send disconnect event, reason=8, AP number=0
[19:41:37.122] D (215423) wifi:connect status 6 -> 0
[19:41:37.122] D (215423) wifi:filter: set rx policy=8
[19:41:37.122] D (215433) wifi:Start wifi connect
[19:41:37.122] D (215433) wifi:connect status 0 -> 0
[19:41:37.122] D (215433) wifi:connect chan=0
[19:41:37.122] D (215433) wifi:first chan=11
[19:41:37.122] D (215443) wifi:connect status 0 -> 1
[19:41:37.122] D (215443) wifi:filter: set rx policy=3
[19:41:37.122] D (215443) wifi:clear scan ap list
[19:41:37.122] D (215453) wifi:start scan: type=0x50f, priority=2, cb=0x420f36ce, arg=0x0, ss_state=0x1, time=215460798, index=0
[19:41:37.122] D (215463) wifi:perform scan: ss_state=0x9, chan<11,0>, dur<0,120>
[19:41:37.122] D (215463) event: running post WIFI_EVENT:5 with handler 0x42014e30 and context 0x3fcba2d4 on loop 0x3fca9fb8[0m
[19:41:37.122] [1;32m./main/src/gl_network.c[202]: Disconnect reason: 8[0m
[19:41:37.122] [1;32m./main/src/gl_network.c[232]: continue to connect wifi![0m
[19:41:37.122] D (215483) wifi:rsn valid: gcipher=3 ucipher=3 akm=5
[19:41:37.122] 
[19:41:37.122] D (215493) wifi:profile match: ss_state=0x7
[19:41:37.122] D (215493) wifi:Start wifi connect
[19:41:37.122] E (215503) wifi:sta is connecting, return error
[19:41:37.122] D (215503) event: running post WIFI_EVENT:5 with handler 0x420a9238 and context 0x3fcbb768 on loop 0x3fca9fb8[0m
[... 223 lines cut ...]
[19:41:43.102] D (221543) esp_netif_handlers: esp_netif action got_ip with netif0x3fcba1dc from event_id=0[0m
[19:41:43.102] [0;32mI (221553) esp_netif_handlers: sta ip: 192.168.88.194, mask: 255.255.255.0, gw: 192.168.88.1[0m
[19:41:43.102] D (221563) HTTP_CLIENT: Begin connect to: https://gslb.lafaer.cn:443[0m
[19:41:43.102] D (221573) esp-tls: host:gslb.lafaer.cn: strlen 14[0m
[19:41:43.102] D (221573) esp-tls: [sock=54] Resolved IPv4 address: 120.24.55.11[0m
[19:41:43.102] D (221583) esp-tls: [sock=54] Connecting to server. HOST: gslb.lafaer.cn, Port: 443[0m
[19:41:47.506] D (225983) mqtt_client: Reconnecting...[0m
[19:41:47.512] D (225983) event: running post MQTT_EVENTS:7 with handler 0x420114a6 and context 0x3fca6f30 on loop 0x3fca7a48[0m
[19:41:47.512] [1;32m./main/src/gl_mqtt.c[191]: Other event id:7[0m
[19:41:47.512] D (225983) esp-tls: host:120.24.55.11: strlen 12[0m
[19:41:47.512] D (225993) esp-tls: [sock=55] Resolved IPv4 address: 120.24.55.11[0m
[19:41:47.512] D (225993) esp-tls: [sock=55] Connecting to server. HOST: 120.24.55.11, Port: 8883[0m
[19:41:48.081] [0;31mE (226603) esp-tls: [s
ock=54] select() timeout[0m
[19:41:48.432] [0;31mE (226603) esp-tls: Failed to open new connection[0m
[19:41:48.432] [0;31mE (226603) transport_base: Failed to open a new connection[0m
[19:41:48.432] [0;31mE (226603) HTTP_CLIENT: Connection failed, sock < 0[0m
[19:41:48.432] D (226613) HTTP_CLIENT: HTTP_EVENT_ERROR[0m
[19:41:48.432] D (226613) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:0 posted to loop 0x3fca9fb8[0m
[19:41:48.432] [0;31mE (226623) HTTP_CLIENT: HTTP POST request failed: ESP_ERR_HTTP_CONNECT[0m
[19:41:48.432] [0;32mI (226633) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED[0m
[19:41:48.432] [0;32mI (226633) HTTP_CLIENT: Last esp error code: 0x8006[0m
[19:41:48.432] [0;32mI (226643) HTTP_CLIENT: Last mbedtls failure: 0x0[0m
[19:41:48.432] D (226643) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:6 posted to loop 0x3fca9fb8[0m
[19:41:48.432] D (226653) event: running post BLE_NETWORK_EVENT:3 with handler 0x42014e30 and context 0x3fcba338 on loop 0x3fca9fb8[0m

hansw123, 2024-10-09:
@gl-LanYH 
从log 看，在发生断开连接时，wifi事件的处理是重新连接，mqtt 在此期间好像还会通信，然后通信失败的事件处理是断开连接重新连接，这就打断了wifi的重新连接。
麻烦确认相应的事件处理函数，和上层协议通信的逻辑有没有判断wifi连接的状态。如果能提供更多的信息就更好定位问题，谢谢！

gl-LanYH, 2024-10-14:
> @gl-LanYH 从log 看，在发生断开连接时，wifi事件的处理是重新连接，mqtt 在此期间好像还会通信，然后通信失败的事件处理是断开连接重新连接，这就打断了wifi的重新连接。 麻烦确认相应的事件处理函数，和上层协议通信的逻辑有没有判断wifi连接的状态。如果能提供更多的信息就更好定位问题，谢谢！

@hansw123 感谢您的回复，我所使用的设备型号是esp32c3，idf版本为5.1.2；我说下我这边的大概逻辑：mqtt的登录信息是通过访问http服务获取的，且获取mqtt的登录信息存在有限的周期，在这期限内只要不使用就会失效，失效后要重新获取；所以当遇到mqtt连接失败或者错误后我会暂停mqtt的继续重连，并重启wifi能正确获取到IP（确保网络可用）后再去重新访问http服务器获取mqtt登录信息重新登录，这期间会将旧的mqtt_client destroy并初始化新的mqtt_client ，以及调用esp_http_client_cleanup将http释放

hansw123, 2024-10-14:
@gl-LanYH 
从您的描述来看，应该是当wifi disconnect 事件发生时，会进行wifi重新连接。然后这时mqtt 重启wifi导致正在连接的wifi 被打断，再然后有点陷入循环的意思，建议mqtt 这里就不要wifi重启了，因为已经有相应的事件处理进行重新连接了，只需要等待got ip事件再操作。或者重新连接的事件处理函数就删除，只在mqtt 这边处理。

hansw123, 2024-10-14:
@gl-LanYH 
同时可以使用api esp_wifi_sta_get_ap_info() 查询wifi 连接状态，重新调整应用逻辑，最好确保mqtt 操作前可以确定wifi的连接状态。

gl-LanYH, 2024-10-18:
> @gl-LanYH 同时可以使用api esp_wifi_sta_get_ap_info() 查询wifi 连接状态，重新调整应用逻辑，最好确保mqtt 操作前可以确定wifi的连接状态。

感谢您提供的帮助！我在更换了另一个网络环境后暂未发现无法重连的现象。不过我这还遇到一种现象就是我在模拟弱网的环境下，重复上述逻辑中的重连wifi，http，mqtt的动作中发现 `esp_get_minimum_free_heap_size` 在不断缩小，挂载两天后由`esp_get_minimum_free_heap_size : 39328 bytes` 缩小到后来的 `esp_get_minimum_free_heap_size : 200 bytes`，虽然在网络良好的情况下能够重连上，但重连的时间也会时长时短，所以不确定esp_get_minimum_free_heap_size 的不断缩小是否会带来其他风险，我将继续挂载设备，下面附上一段日志：
[1;32m./main/src/gl_network.c[202]: Disconnect reason: 201[0m
[1;32m./main/src/gl_network.c[232]: continue to connect wifi![0m
I (98287029) wifi:new:<3,1>, old:<3,0>, ap:<255,255>, sta:<3,1>, prof:1
I (98287029) wifi:state: init -> auth (b0)
I (98288029) wifi:state: auth -> init (200)
I (98288029) wifi:new:<3,0>, old:<3,1>, ap:<255,255>, sta:<3,1>, prof:1
[1;32m./main/src/gl_network.c[202]: Disconnect reason: 2[0m
[1;32m./main/src/gl_network.c[232]: continue to connect wifi![0m
[1;32m./main/src/gl_network.c[202]: Disconnect reason: 205[0m
[1;32m./main/src/gl_network.c[232]: continue to connect wifi![0m
I (98291429) wifi:new:<3,1>, old:<3,0>, ap:<255,255>, sta:<3,1>, prof:1
I (98291429) wifi:state: init -> auth (b0)
I (98291459) wifi:state: auth -> assoc (0)
I (98291469) wifi:state: assoc -> run (10)
I (98291549) wifi:connected with TEST-LYH-S1300-2.4G, aid = 2, channel 3, 40U, bssid = 94:83:c4:06:53:1b
I (98291549) wifi:security: WPA2-PSK, phy: bgn, rssi: -21
I (98291549) wifi:pm start, type: 1
I (98291559) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
[1;32m./main/src/gl_network.c[196]: WIFI_EVENT_STA_CONNECTED[0m
I (98291579) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (98291589) wifi:<ba-add>idx:0 (ifx:0, 94:83:c4:06:53:1b), tid:6, ssn:2, winSize:64
[1;32m./main/src/gl_network.c[139]: IP_EVENT_STA_GOT_IP[0m
[1;32m./components/gl_common/src/gl_common.c[405]: *************************************************************[0m
[1;32m./components/gl_common/src/gl_common.c[406]: esp_get_free_heap_size : 89704 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[407]: esp_get_free_internal_heap_size : 81932 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[408]: esp_get_minimum_free_heap_size : 39328 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[409]: heap_caps_get_largest_free_block : 34816 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[410]: *************************************************************[0m
[0;32mI (98292619) esp_netif_handlers: sta ip: 192.168.7.194, mask: 255.255.255.0, gw: 192.168.7.1[0m
I (98292739) wifi:<ba-add>idx:1 (ifx:0, 94:83:c4:06:53:1b), tid:0, ssn:0, winSize:64
[1;32m./main/src/gl_mqtt.c[191]: Other event id:7[0m
[0;33mW (98294389) mqtt_client: Connection refused, bad username or password[0m
[1;31m./main/src/gl_mqtt.c[181]: MQTT_EVENT_ERROR[0m
[0;32mI (98294389) mqtt_client: Client asked to disconnect[0m
I (98294399) wifi:state: run -> init (0)
I (98294399) wifi:pm stop, total sleep time: 1484887 us / 2843325 us
I (98294409) wifi:<ba-del>idx:1, tid:0
I (98294409) wifi:<ba-del>idx:0, tid:6
I (98294409) wifi:new:<3,0>, old:<3,1>, ap:<255,255>, sta:<3,1>, prof:1
[1;32m./main/src/gl_network.c[202]: Disconnect reason: 8[0m
[1;32m./main/src/gl_network.c[232]: continue to connect wifi![0m
[0;31mE (98294429) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 54[0m
[0;31mE (98294439) HTTP_CLIENT: HTTP POST request failed: ESP_ERR_HTTP_FETCH_HEADER[0m
[0;32mI (98294449) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED[0m
I (98294449) wifi:new:<3,1>, old:<3,0>, ap:<255,255>, sta:<3,1>, prof:1
I (98294459) wifi:state: init -> auth (b0)
[1;32m./main/src/gl_network.c[345]: continue goto network[0m
I (98294479) wifi:state: auth -> init (8a0)
I (98294479) wifi:new:<3,0>, old:<3,1>, ap:<255,255>, sta:<3,1>, prof:1
[1;32m./main/src/gl_network.c[202]: Disconnect reason: 202[0m
[1;32m./main/src/gl_network.c[232]: continue to connect wifi![0m
E (98294499) wifi:sta is connecting, return error
E (98294529) wifi:sta is connecting, return error
[0;31mE (98294529) mqtt_client: MQTT connect failed[0m
[1;32m./main/src/gl_mqtt.c[140]: MQTT_EVENT_DISCONNECTED[0m
[1;32m./components/gl_common/src/gl_common.c[405]: *************************************************************[0m
[1;32m./components/gl_common/src/gl_common.c[406]: esp_get_free_heap_size : 89972 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[407]: esp_get_free_internal_heap_size : 82200 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[408]: esp_get_minimum_free_heap_size : 2524 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[409]: heap_caps_get_largest_free_block : 34816 bytes[0m
[1;32m./components/gl_common/src/gl_common.c[410]: *************************************************************[0m
[1;32m./main/src/gl_network.c[202]: Disconnect reason: 205[0m
[1;32m./main/src/gl_network.c[232]: continue to connect wifi![0m
I (98297899) wifi:new:<3,1>, old:<3,0>, ap:<255,255>, sta:<3,1>, prof:1
I (98297899) wifi:state: init -> auth (b0)
I (98297909) wifi:state: auth -> assoc (0)
I (98297929) wifi:state: assoc -> run (10)
I (98297969) wifi:connected with TEST-LYH-S1300-2.4G, aid = 2, channel 3, 40U, bssid = 94:83:c4:06:53:1b
I (98297979) wifi:security: WPA2-PSK, phy: bgn, rssi: -22
I (98297979) wifi:pm start, type: 1
I (98297979) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
[1;32m./main/src/gl_network.c[196]: WIFI_EVENT_STA_CONNECTED[0m
I (98298029) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (98298049) wifi:<ba-add>idx:0 (ifx:0, 94:83:c4:06:53:1b), tid:6, ssn:2, winSize:64
I (98298069) wifi:<ba-add>idx:1 (ifx:0, 94:83:c4:06:53:1b), tid:0, ssn:1, winSize:64

REEASD, 2024-11-12:
我也有这个问题，设备在运行大概2条后就容易出现[20:01:23.451]收←◆[0;33mW (107933554) esp-tls: Failed to open new connection in specified timeout[0m
[0;31mE (107933554) transport_base: Failed to open a new connection[0m
[0;31mE (107933559) HTTP_CLIENT: Connection failed, sock < 0[0m的问题

hansw123, 2024-11-13:
@REEASD 
如果是server 出现问题导致tcp 的连接出现问题，根据协议是有固定的一些操作以及等待时间的，只有这些超时达成，才会回收连接的资源。具体的比如说tcp 四次挥手中的TIME_WAIT 阶段的时间

closed 2025-01-14 by Alvin1Zhang