espressif/esp-idf #14647: SPI related core panic

y-nk, 2024-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

latest

### Espressif SoC revision.

ESP32-D0WD-V3

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I'm streaming wav files from sdcard to a bluetooth sink using arduino-audio-tools. For this, I want to keep a dedicated bus for the sdcard reader (for performance reason) so it is wired on a custom set of gpio (not 5/18/19/21) but rather 25/26/27/33 (i had tried 1st 12/13/14/27 but using 12 crashes my board at startup). 

This is the sketch I'm using:

```ino
#include "AudioTools.h"
#include "AudioLibs/A2DPStream.h"
#include "AudioLibs/AudioSourceSD.h"
#include "AudioCodecs/CodecWAV.h"

#define HSPI_SCLK 26
#define HSPI_MISO 33
#define HSPI_MOSI 25
#define HSPI_SS   27

const char *startFilePath="/";
const char *ext="wav";

SPIClass hspi(HSPI);
AudioSourceSD source(startFilePath, ext, HSPI_SS, hspi);
A2DPStream out;
WAVDecoder decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  hspi.begin(HSPI_SCLK, HSPI_MISO, HSPI_MOSI, HSPI_SS);

  // setup output
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.name = "Bose Color II SoundLink";
  cfg.auto_reconnect = true;
  out.begin(cfg);

  // setup player
  player.begin();
}

void loop() {
  player.copy();
}
```

This will crash the board at some point, invoking an SPI related error. Decoded stack trace is:

```log
PC: 0x40083d80 EXCVADDR: 0x00000000  Decoding stack results 0x400e71fa: pthread_key_create at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/pthread/pthread_local_storage.c line 56 0x401a2e0d: mbedtls_ecdsa_read_signature_restartable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ecdsa.c line 762 0x400df265: SPIClass::endTransaction() at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/SPI/src/SPI.cpp line 196 0x400dec09: VFSImpl::mkdir(char const*) at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src/vfs_api.cpp line 215 0x400deb0d: VFSImpl::remove(char const*) at /$HOME/Library/Arduino15/packages/esp32/hardware/esp32/3.0.5/libraries/FS/src/vfs_api.cpp line 438 0x40099171: prvTimerTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/timers.c line 587 0x40099bae: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c line 162
```

I thought first to report to arduino-audio-tools but since the stack reveals no relationship with it, i thought this is much more related to an error here rather than there. I'm happy to move this report to the relevant place if found that it's not a core issue.

### What is the actual behavior?

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 

### Steps to reproduce.

1. Have my board
2. Have my components and wiring
4. Prepare an SD card
5. Build the sketch
6. Should crash after some time

cristianfunes79, 2024-10-08:
Based on ESP documentation:
`The address which has been written/read is found in the EXCVADDR register in the register dump. If this address is zero, it usually means that the application has attempted to dereference a NULL pointer.`

I assume there is something that needs to be initialized. When is you app crashing? during setup or loop?

https://github.com/pschatzmann/arduino-audio-tools/blob/aeda11b96e23a09f9013e3fe92e1c4c8b97d201a/src/AudioTools/AudioLibs/AudioSourceSD.h#L33

I also think that maybe you need to define SPI macro to properly initialize Audio class.

y-nk, 2024-10-15:
@cristianfunes79 during loop.

>This will crash the board **at some point**