espressif/esp-idf #14649: we are facing challenges while working with alert notification profile and service

Laxmikanth234, 2024-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

2.12.0

### Espressif SoC revision.

ESP32- S3

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 -S3

### Power Supply used.

USB

### What is the expected behavior?

we are working esp32 s3 to display notification of andriod mobile to display on terminal we are using iot solution git repository code added neccesary .yml files for dependency purposes we are facing issues while running the code  

### What is the actual behavior?

macors nor define and some dummy fuctions and connection manager (conn_mgr)

### Steps to reproduce.

1.during proojecr build

### Debug Logs.

```plain
macors not defines
```

### More Information.

working on  ( ESPRESSIF IDE )
i want only ANS or ANP code for andriod notification to display on terminal , and any help will be apreciated 
thank you

esp-zhp, 2024-10-08:
Apple devices support this feature, and we have an example demonstrating how to receive notifications from Apple devices.
see：https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_ancs
However, as I know, Android does not support this feature. You may need to develop an Android app to implement it.

Laxmikanth234, 2024-10-08:
> Apple devices support this feature, and we have an example demonstrating how to receive notifications from Apple devices. [...]
we can start the android  app , presently working on esp32 micro controller side so, could you please send me the any android code for(alert notification profile or service)

closed 2024-10-08 by Laxmikanth234