espressif/esp-idf #14651: cmake -D settings are not passed to components

KammutierSpule, 2024-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected that the defined option on main cmakefile or via command line, eg: cmake -DOPTION=ON to be passed to components and main component

### What is the actual behavior?

At this moment, only the main CMakeLists.txt file receive this options (I test) but they don't appear on components or main

### Steps to reproduce.

1. on main root CMakeLists.txt add `option(BUILD_OPTION "Set my option define" OFF)`
2. add something to verify that is is detected on main root, but not on components
3. run the build of your project such way that it adds -DBUILD_OPTION=ON
...

igrr, 2024-09-30:
Could you please show the part of your component CMakeLists file which uses the `BUILD_OPTION`?

For example, modifying hello_world example with `option(...)` and the following works as expected:
```
idf_component_register(SRCS "hello_world_main.c"
                    PRIV_REQUIRES spi_flash
                    INCLUDE_DIRS "")

message(STATUS "BUILD_OPTION from main/CMakelists.txt: ${BUILD_OPTION}")
```

KammutierSpule, 2024-09-30:
```
option(BUILD_OPTION "Set DEBUG define" OFF)
message(STATUS "BUILD_OPTION from root CMakelists.txt: ${BUILD_OPTION}")
```

```
message(STATUS "BUILD_OPTION from main/CMakelists.txt: ${BUILD_OPTION}")
```

I got:
```
-- BUILD_OPTION from root CMakelists.txt: ON
...
Processing 2 dependencies:
[1/2] espressif/esp_modem (1.1.0)
[2/2] idf (5.3.0)
-- BUILD_OPTION from main/CMakelists.txt: 
```

igrr, 2024-09-30:
> Am I able to create a workspace project with cmake build options without create a new folder/repository with a new project for every option?

I'm sorry, I don't understand what you mean. Could you show an example of how you actually use this `BUILD_OPTION` in main/CMakeLists.txt? Then I might be able to suggest a solution.

KammutierSpule, 2024-09-30:
> I'm sorry, I don't understand what you mean. Could you show an example of how you actually use this `BUILD_OPTION` in main/CMakeLists.txt? Then I might be able to suggest a solution.

I have different uses cases: Debug/Release profiles, UnitTests profiles, different options that require different files to be included.. (different components/modules too)

igrr, 2024-09-30:
Regarding different components, you can use the following in project CMakeLists.txt:

```cmake
cmake_minimum_required(VERSION 3.16)

option(COMPONENT_X_INCLUDED "Include feature X" OFF)
set(COMPONENTS main)
if(COMPONENT_X_INCLUDED)
  list(APPEND COMPONENTS component_x)
endif()

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(your-project)
```

For adding source files or compilation options, you can use the following in your component CMakeLists file:
```
idf_component_register(SRCS "hello_world_main.c"
                    PRIV_REQUIRES spi_flash
                    INCLUDE_DIRS "")
                    
if (ENABLE_DEBUG)   # assuming you have defined ENABLE_DEBUG option in project CMakeLists file
  target_compile_definitions(${COMPONENT_LIB} PRIVATE ENABLE_DEBUG=1)
  target_sources(${COMPONENT_LIB} PRIVATE debug_helpers.c)
endif()
```

and so on.

KammutierSpule, 2024-09-30:
Do you have more resources to read on how to fine control selected COMPONENTS to the build?

```
set(COMPONENTS main)
if(COMPONENT_X_INCLUDED)
  list(APPEND COMPONENTS component_x)
endif()
```
Does this mean if I use it, it will stop from autodetect main and components on the subfolders?

igrr, 2024-09-30:
No, this way `main` component is still going to be found (it is listed in COMPONENTS list). Also any components that `main` depends on will be included into the build.

You can read about this in the docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#including-components-in-the-build and the nearby sections.

KammutierSpule, 2024-10-03:
Sorry I'm not understanding your example:
```
option(COMPONENT_X_INCLUDED "Include feature X" OFF)
set(COMPONENTS main)
if(COMPONENT_X_INCLUDED)
  list(APPEND COMPONENTS component_x)
endif()
```
If they are going to be found "automatically", what does this set(COMPONENTS main) and why append a component that will automatically be discovered?

I read it was possible to assign different paths to main (what about components?) and that way I could have a manual control of what components and main is added ? (not that I required but... )

Thanks for the slides (I didn't check the talk yet).

igrr, 2024-10-03:
> I read it was possible to assign different paths to main (what about components?) and that way I could have a manual control of what components and main is added ?

Yes, you can use the same method for this. If OPTION_A is set, you can set COMPONENTS to main_a, if OPTION_B is set, then to main_b.

dobairoland, 2025-03-07:
I think this can be closed for inactivity and no further action was identified.

closed 2025-03-07 by dobairoland