espressif/esp-idf #14654: Nimble BLufi Host Stack with some problem after 5.2.3 as stack size start to overflow telling BTC_STACK

filzek, 2024-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Nimble BLufi Host Stack with some problem after 5.2.3 as stack size start to overflow

***ERROR*** A stack overflow in task BTC_TASK has been detected.

So, something is weirdo, as this doenst happen before

rahult-github, 2024-10-01:
Hi @filzek , 

Request you to always share complete log and sdkconfig whenever you raise any issue. 

As for this issue, please try increasing the STACK_SIZE in menuconfig to see if it helps.

filzek, 2024-10-01:
@rahult-github 

***ERROR*** A stack overflow in task BTC_TASK has been detected.

Backtrace: 0x40081931:0x3ffed8e0 0x40095c61:0x3ffed900 0x4009615d:0x3ffed920 0x400968c9:0x3ffed9a0 0x40095d5c:0x3ffed9c0 0x40095d0e:0x40091fc3 |<-CORRUPTED
--- 0x40081931: panic_abort at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/panic.c:466
0x40095c61: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/port/esp_system_chip.c:93
0x4009615d: vApplicationStackOverflowHook at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:558
0x400968c9: vTaskSwitchContext at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/tasks.c:3664 (discriminator 7)
0x40095d5c: _frxt_dispatch at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x40095d0e: _frxt_int_exit at C:/Espressif/frameworks/esp-idf-v5.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246
0x40091fc3: strcmp at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strcmp.S:47

We are using NIMBLE and attached are the SDKCONFIG, it was supose that the BTC_TASK does nothing as the NIMBLE TASK should be in charge of all running code in the NIMBLE/BLUFI callback, and somehow it is trigerring the problem.

The SDK 5.2 to 5.2.2 did not bring this problem and this only happen in the 5.2.3, so, something related to the internal affairs of BTC_TASK to NIMBLE task has creating this issue

rahult-github, 2024-10-01:
Hi @filzek ,  

There is no sdkconfig attached in previous comment, please share it. Also, initial comment didn't mention regarding any crash / backtrace information. Can you please share the elf file also in this case ?

filzek, 2024-10-01:
> Hi @filzek ,
> 
> There is no sdkconfig attached in previous comment, please share it. Also, initial comment didn't mention regarding any crash / backtrace information. Can you please share the elf file also in this case ?

please send me an private email address to send it private to you as well, i will send the elf file.
[sdkconfig.txt](https://github.com/user-attachments/files/17211062/sdkconfig.txt)

rahult-github, 2024-10-03:
Hi @filzek ,

Your sdkconfig has below settings. So bluedroid is used and not nimble.   I will re-direct issue to correct team in this case, since the heading mentioned NimBLE stack . 

#
# Bluetooth
#
CONFIG_BT_ENABLED=y
CONFIG_BT_BLUEDROID_ENABLED=y
# CONFIG_BT_NIMBLE_ENABLED is not set
# CONFIG_BT_CONTROLLER_ONLY is not set
CONFIG_BT_CONTROLLER_ENABLED=y
# CONFIG_BT_CONTROLLER_DISABLED is not set

filzek, 2024-10-03:
> Hi @filzek ,
>
> Your sdkconfig has below settings. So bluedroid is used and not nimble. I will re-direct issue to correct team in this case, since the heading mentioned NimBLE stack .

Hi, something is wrong, as the SDKCONFIG attached show the below info:

#
# Bluetooth
#
CONFIG_BT_ENABLED=y
# CONFIG_BT_BLUEDROID_ENABLED is not set
CONFIG_BT_NIMBLE_ENABLED=y
# CONFIG_BT_CONTROLLER_ONLY is not set
CONFIG_BT_CONTROLLER_ENABLED=y
# CONFIG_BT_CONTROLLER_DISABLED is not set

we are using NMBLE, as per show in the SDKCONFIG file, maybe you have open a wrong SDKCONFIG.TXT file.

rahult-github, 2024-10-23:
Hi @filzek  ,  Thanks. Indeed your comment is correct. However, by just enabling nimble, there is no such issue observed. And the sdkconfig file has way too many differences, and taking all sdkconfig as is causing image to not boot. Tried picking ONLY bluetooth related configs and no such failure is observed.

filzek, 2024-11-08:
Thats is one of the major problem with the SDK right now, if differents changes in SDKConfig is set the behavior on the code is not 100% secure.

gogogoghost, 2025-09-05:
I have a similar issue. I am using NimBLE with blufi. And there has some work in `esp_blufi_callbacks_t::event_cb` callback when I got a blufi message.

It works fine on IDF 5.2.5 but crash on IDF 5.5.1. I found a stack over flow issue in BTC_TASK. I want to increase the stack size of BTC_TASK but I can't configure `CONFIG_BT_BTC_TASK_STACK_SIZE` when bluedroid not enabled.

I tried to edit the souce code of IDF from
```
btc_thread = osi_thread_create(BTC_TASK_NAME, BTC_TASK_STACK_SIZE, BTC_TASK_PRIO, BTC_TASK_PINNED_TO_CORE,
                                   BTC_TASK_WORKQUEUE_NUM, workqueue_len);
```
to
```
btc_thread = osi_thread_create(BTC_TASK_NAME, 8192, BTC_TASK_PRIO, BTC_TASK_PINNED_TO_CORE,
                                   BTC_TASK_WORKQUEUE_NUM, workqueue_len);
```
The issue can be solved.

So I think there should be a way to configure `CONFIG_BT_BTC_TASK_STACK_SIZE` without bluedroid enabled.

rahult-github, 2025-09-05:
Hi @gogogoghost , 

Please find attached patch to be applied in $IDF_PATH .  When selecting nimble, you should be able to set BT_NIMBLE_HOST_TASK_STACK_SIZE from menuconfig to handle this. 

[set_blufi_nimble_size.txt](https://github.com/user-attachments/files/22170866/set_blufi_nimble_size.txt)

rahult-github, 2026-10-19:
original issue is reported on 5.2 branch which is now EOL and the later issue has already been [fixed](https://github.com/espressif/esp-idf/commit/26f429844efc5dbbc888790e139e888683304923) . Closing this issue. Please free to reopen if any issue is observed on existing active branches.

closed 2026-10-19 by rahult-github