espressif/esp-idf #14655: Bluetooth Controller on ESP32-C6 does not seem to call notify_host_send_available

mringwal, 2024-09-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1-395-g707d097b01

### Espressif SoC revision.

ESP32-C6

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C6-N8

### Power Supply used.

USB

### What is the expected behavior?

Our BTstack (https://github.com/bluekitchen/btstack/) uses the VHCI API on all ESP32 versions that support Bluetooth. On ESP32-C6 with esp-idf v5.3,  

### What is the actual behavior?

I expect the .notify_host_send_available to be called by the Controller from a thread. I also expect that this happens on a regular thread and not from an ISR.

### Steps to reproduce.

- Follow instructions here:
https://github.com/bluekitchen/btstack/blob/master/port/esp32/README.md
- go to an example, e.g. example/gatt_counter
- $ idf.py set-target esp32-c6
- $ idf.py menuconfig
   - Component -> Bluetooth: enable Bluetooth, but disable Host
   - Component -> BTstack: disable I2S Audio driver
- in src/main, enable logging by uncommenting
    `hci_dump_init(hci_dump_embedded_stdout_get_instance());`
- in esp-idf/component/btstack/btstack_port_esp32.c, in host_send_pkt_available_cb(), add
    `log_info("host_send_pkt_available_cb!");
- $ idf.py build flash monitor

You'll get a startup log that ends with:

[00:00:00.181] LOG -- btstack_run_loop_freertos.c.236: run loop init, task 0x4081c738, queue item size 8
[00:00:00.181] LOG -- main.c.285: transport_register_packet_handler
[00:00:00.181] LOG -- btstack_tlv.c.48: TLV Instance 0x42075df4
[00:00:00.181] LOG -- le_device_db_tlv.c.159: num valid le device entries 0
[00:00:00.181] LOG -- att_db.c.250: att_set_db 0x42075380
[00:00:00.181] LOG -- hci.c.1681: Num LE Peripheral roles: 0 of 1
BTstack counter 0001
[00:00:00.181] LOG -- hci.c.5531: hci_power_control: 1, current mode 0
[00:00:00.181] LOG -- main.c.205: transport_init
[00:00:00.181] LOG -- main.c.216: transport_open
I (181) BLE_INIT: Using main XTAL as clock source
I (191) BLE_INIT: ble controller commit:[5bc682c]
I (191) BLE_INIT: Bluetooth MAC: 40:4c:ca:56:1a:92
I (191) phy_init: phy_version 310,dde1ba9,Jun  4 2024,16:38:11
I (241) phy: libbtbb version: 04952fd, Jun  4 2024, 16:38:26
[00:00:00.241] LOG -- hci.c.8136: BTSTACK_EVENT_STATE 1
[00:00:00.241] CMD => 03 0C 00
[00:00:00.241] EVT <= 0E 04 01 03 0C 00

The stack stops, as the . notify_host_send_available callback is not called and BTstack is waiting for this.

### Debug Logs.

```plain
See 'steps to reproduce'
```

### More Information.

I had a look at the NimBLE port for comparison, but got confused as there are two port files.
- esp-idf/components/bt/host/nimble/esp-hci/src/esp_nimble_hci.c
- esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c

esp_vhci_host_register_callback is only called in esp_nimble_hci.c.

Anyway. Is there a plan to deprecate the .notify_host_send_available? If yes, please let us know how we can find out when the stack can send the next packet.

zhaoweiliang2021, 2024-10-10:
Hi, @mringwal Thanks for raising this question.
The notify_host_send_available api won't be called by controller on ESP32-C6.  For how to use VHCI on ESP32-C6, please refer to the hci_driver_standard.c file. 

The api  esp_err_t esp_vhci_host_register_callback(const esp_vhci_host_callback_t *callback)  shall be called in Host.

The  notify_host_send_available function pointer in  esp_vhci_host_callback  structure will not be used temporarily based on the existing functionality on ESP32-C6 .

 For issue : how we can find out when the stack can send the next packet？

According to the Num_Completed_Packets of "Number Of Completed Packets event" and the return
parameters of the HCI_Read_Buffer_Size and HCI_LE_Read_Buffer_Size commands, the Host can determine for which Connection_Handles the following HCI packets should be sent to the Controller.

According to the  Num_HCI_Command_Packets of  "Command Complete event" ,  the Host can send HCI command packets to the Controller.

mringwal, 2024-10-10:
Hi @zhaoweiliang2021 Thanks for the explanation. 

To recap: for the ESP32-C6, we don't need to wait for notify_host_send_available to get called. It is sufficient to follow the standard Bluetooth HCI rules for sending ACL packets (read buffer size + num completed) as well as Command packets (num completed packets), correct?

Is this specific to the ESP32-C6, or does it apply to all other SoCs, e.g. ESP32 or ESP32-C3? i.e. can we ignore notify_host_send_available on all platform, or just on the ESP32-C6 (and probably ESP32-H2)?

zhaoweiliang2021, 2024-10-10:
@mringwal  Hi,
Question 1: for the ESP32-C6, we don't need to wait for notify_host_send_available to get called. It is sufficient to follow the standard Bluetooth HCI rules for sending ACL packets (read buffer size + num completed) as well as Command packets (num completed packets), correct?
Yes.
Question 2: Is this specific to the ESP32-C6, or does it apply to all other SoCs, e.g. ESP32 or ESP32-C3? i.e. can we ignore notify_host_send_available on all platform, or just on the ESP32-C6 (and probably ESP32-H2)?
It apply to ESP32-C6,ESP32-C2,ESP32-H2 currently.

mringwal, 2024-10-10:
Thanks for the clarification. 

We'll update our code to consider VHCI as synchronous on ESP32-C2/C6/H2 and keep the asynchronous mode for the others (ESP32, ESP32-S3. ESP32-C3). 

Any idea what the ESP32-C5 might use? My blind guess would be that it will also be synchronous.

closed 2024-10-10 by mringwal

zhaoweiliang2021, 2024-10-10:
@mringwal 
Any idea what the ESP32-C5 might use? My blind guess would be that it will also be synchronous.
It will be synchronous.

mringwal, 2024-10-10:
Thanks. I'll then make a list for device that use the (older) asynchronous mode and use synchronous otherwise.