espressif/esp-idf #14656: Failed to resolve component 'esp_tls'

sova32, 2024-10-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

build

### What is the actual behavior?

 *  Executing task: ninja  

[0/1] Re-running CMake...
-- git rev-parse returned 'fatal: не найден git репозиторий (или один из его каталогов вплоть до точки монтирования /run/media/vadiki)
Останавливаю поиск на границе файловой системы (так как GIT_DISCOVERY_ACROSS_FILESYSTEM не установлен).'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 2 dependencies:
[1/2] joltwallet/littlefs (1.14.8)
[2/2] idf (5.4.0)
CMake Error at /home/vadiki/esp/esp-idf/tools/cmake/build.cmake:298 (message):
  Failed to resolve component 'esp_tls' required by component 'my_WiFi':
  unknown name.
Call Stack (most recent call first):
  /home/vadiki/esp/esp-idf/tools/cmake/build.cmake:341 (__build_resolve_and_add_req)
  /home/vadiki/esp/esp-idf/tools/cmake/build.cmake:644 (__build_expand_requirements)
  /home/vadiki/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
FAILED: build.ninja 
/usr/bin/cmake --regenerate-during-build -S/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2 -B/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2/build
ninja: error: rebuilding 'build.ninja': subcommand failed

 *  The terminal process "ninja" terminated with exit code: 1. 

### Steps to reproduce.

1. #include "esp_tls"
2. build

### Build or installation Logs.

```plain
*  Executing task: ninja  

[0/1] Re-running CMake...
-- git rev-parse returned 'fatal: не найден git репозиторий (или один из его каталогов вплоть до точки монтирования /run/media/vadiki)
Останавливаю поиск на границе файловой системы (так как GIT_DISCOVERY_ACROSS_FILESYSTEM не установлен).'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 2 dependencies:
[1/2] joltwallet/littlefs (1.14.8)
[2/2] idf (5.4.0)
CMake Error at /home/vadiki/esp/esp-idf/tools/cmake/build.cmake:298 (message):
  Failed to resolve component 'esp_tls' required by component 'my_WiFi':
  unknown name.
Call Stack (most recent call first):
  /home/vadiki/esp/esp-idf/tools/cmake/build.cmake:341 (__build_resolve_and_add_req)
  /home/vadiki/esp/esp-idf/tools/cmake/build.cmake:644 (__build_expand_requirements)
  /home/vadiki/esp/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
FAILED: build.ninja 
/usr/bin/cmake --regenerate-during-build -S/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2 -B/run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2/build
ninja: error: rebuilding 'build.ninja': subcommand failed

 *  The terminal process "ninja" terminated with exit code: 1.
```

fhrbata, 2024-10-01:
Hello @sova32 , shouldn't the component name be **esp-tls**, with the dash instead of underscore? Could you please check that `my_WiFi` has `esp_tls` in requirements?

```
$ git ls-tree v5.3.1 components/ | grep tls
040000 tree 4d77415b832f8365d739e9f5232b241c4d420031	components/esp-tls
040000 tree c3f1dfdad57a4ed4ba9db9f3c9d2a0c7293e907b	components/mbedtls

```

cristianfunes79, 2024-10-08:
> Hello @sova32 , shouldn't the component name be **esp-tls**, with the dash instead of underscore? Could you please check that `my_WiFi` has `esp-tls` in requirements?

Yes if you see build output it shows that error:
`  Failed to resolve component 'esp_tls' required by component 'my_WiFi':
  unknown name.`

It should be named as `esp-tls`

closed 2024-10-14 by mahavirj