espressif/esp-idf #14658: USB vendor specific device

maaamcube, 2024-10-01:
### Is your feature request related to a problem?

Hello everyone,
I checked the Esp idf USB Host Library and i didn't find that Vendor Specific USB is supported, can it be added ? (i need to connect my Stm32 device Vendor Specific USB to send it through wifi)
Thanks.

### Describe the solution you'd like.

Ideally, the Esp idf USB Host Library support Vendor Specific USB so i can send multiple hubs and vendor specific usb devices through wifi.

tore-espressif, 2024-10-02:
Hello @maaamcube , thank you for the request.

We don't plan to implement Vendor specific class driver because, each vendor specific device is different and requires custom driver that can handle all its features, so it must be custom made for a specific device.

If your device uses standard BULK IN/OUT endpoints, you can use our [CDC-ACM driver](https://components.espressif.com/components/espressif/usb_host_cdc_acm/versions/2.0.4)

Could you also please clarify what do you mean by 'send multiple hubs and vendor specific usb devices through wifi'?

maaamcube, 2024-10-02:
Hello @tore-espressif,
Thanks for your reply.
On my PCB, I have 3 hubs, each hub has an Stm32 usb device vendor specific
[image]

tore-espressif, 2024-10-02:
Duplicate of https://github.com/espressif/esp-idf/issues/13307

The HUB support will be added, but we can't help with this very custom design

closed 2024-10-02 by tore-espressif

chegewara, 2024-11-07:
https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge

maaamcube, 2024-11-07:
> https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-usb-bridge/examples/usb_wireless_bridge

there is no option with usb between the Esp32 and the MCU