espressif/esp-idf #14659: BLE advertisement channel 39 is not working on Bluedroid on ESP32S3

aircable, 2024-10-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.4

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed ESP32S3 module

### Power Supply used.

USB

### What is the expected behavior?

I configured the advertisement like this:
```
static esp_ble_adv_params_t adv_params = {
    .adv_int_min        = 0x20, // fastest, very 20ms
    .adv_int_max        = 0x20,
    .adv_type           = ADV_TYPE_NONCONN_IND, 
    .own_addr_type      = BLE_ADDR_TYPE_RANDOM,  
    .channel_map        = ADV_CHNL_39,         // bitmap 0x07, ADV_CHNL_ALL
    .adv_filter_policy  = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
};
```

After setting the advertisement data esp_ble_gap_config_adv_data_raw() it starts advertising.
Confirming with a spectrum analyzer as well as a BLE scanner app on the phone (LightBlue) the chip is not advertising.
There are no error reports, all events are occurring correctly.

Chainging the channel_map to either ADV_CHNL_38 or ADV_CHNL_ALL works ok. I can see the advertisments.

Unfortunately I have no way to control the receiving channels to listen only on channel 39 for example. There is no API, unless you know how to do that. I would appreciate it. So I use external instruments and receivers to check for any channel 39 only advertisements and there are none.

### What is the actual behavior?

no error, just does not do what it suppose to do

### Steps to reproduce.

I only tested Bluedroid, not Nimble.
To reproduce, use the iBeacon demo app in the IDF examples and change the advertisement config to channel 39 only.

### Debug Logs.

```plain
No logs, no errors, just not doing what it should.
```

aircable, 2024-10-06:
This hardware bug seems to be on a ESP32S3 chip with revision v0.1.  I was successfully working with only channel 39 on revision v0.2 on a ESP32-S3-WROOM-1 module, not on a XIAO-ESP32-S3 module.
If you can verify, it's probably ok to close this issue.

esp-zhp, 2024-10-08:
@aircable 
Could you provide me with a full log?

aircable, 2024-10-08:
Log of what? There is no error. All commands are executing correctly. You just look on a Bluetooth scanner or write your own scanner code for another ESP32 or use LightBlue on your Android. On the ESP32S3 v0.1 there is just no microwaves generated.

esp-zhp, 2024-10-09:
I would like to obtain the startup log information to identify various details about the chip（v0.1）.

Alvin1Zhang, 2024-12-17:
Thanks for reporting, sorry that we failed to reproduce the issue locally, will close due to inactivity, feel free to reopen with more updates per the suggestion https://github.com/espressif/esp-idf/issues/14659#issuecomment-2401169715, thanks.

closed 2024-12-17 by Alvin1Zhang