espressif/esp-idf #14661: unexpected crash when enabling flash encryption with virtual eFuses

aygh4266, 2024-10-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

the firmware should be booted normally. No crash. Simulation should be shown and the value of eFuses should be copied intor RAM and not be burned permanently  

### What is the actual behavior?

unexpected crash. Pleas refer to Debug Logs

### Steps to reproduce.

1. in kConfig: set simulate eFuse operations in RAM
2. in Security Features: Enable flash encryption on boot has been set
3. Leave UART Bootloader encryption enabled
4. UART ROM download mode has been set to UART ROM dowload mode (Enabled)
...

### Build or installation Logs.

```plain
Debug-Logs:

SPIWP:0xee
mode:DIO, clock div:1
Valid secure boot key blocks: 0
secure boot verification succeeded
load:0x3fce3980,len:0x322c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x4c60
entry 0x403c9914
I (68) boot: ESP-IDF v5.2.2 2nd stage bootloader
I (68) boot: compile time Sep  2 2024 09:42:52
I (68) boot: Multicore bootloader
I (71) boot: chip revision: v0.1
I (75) boot.esp32s3: Boot SPI Speed : 80MHz
I (80) boot.esp32s3: SPI Mode       : DIO
I (85) boot.esp32s3: SPI Flash Size : 8MB
I (89) boot: Enabling RNG early entropy source...
I (95) boot: Partition Table:
I (98) boot: ## Label            Usage          Type ST Offset   Length
I (106) boot:  0 nvs              WiFi data        01 02 00011000 00004000
I (113) boot:  1 otadata          OTA data         01 00 00015000 00002000
I (121) boot:  2 phy_init         RF data          01 01 00017000 00000400
I (128) boot:  3 ota_0            OTA app          00 10 00020000 00200000
I (136) boot:  4 ota_1            OTA app          00 11 00220000 00200000
I (143) boot:  5 storage          Unknown data     01 82 00420000 00100000
I (151) boot: End of partition table
I (155) esp_image: segment 0: paddr=00020020 vaddr=3c0e0020 size=32e48h (208456) map
I (201) esp_image: segment 1: paddr=00052e70 vaddr=3fc9dc00 size=05208h ( 21000) load
I (206) esp_image: segment 2: paddr=00058080 vaddr=40374000 size=07f98h ( 32664) load
I (214) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=d6d2ch (879916) map
I (373) esp_image: segment 4: paddr=00136d54 vaddr=4037bf98 size=11c0ch ( 72716) load
I (389) esp_image: segment 5: paddr=00148968 vaddr=00000000 size=07668h ( 30312) 
I (395) esp_image: Verifying image signature...
I (396) secure_boot_v2: Verifying with RSA-PSS...
I (400) secure_boot_v2: Signature verified successfully!
I (413) boot: Loaded app from partition at offset 0x20000
I (413) secure_boot_v2: enabling secure boot v2...
I (415) secure_boot_v2: secure boot v2 is already enabled, continuing..
I (422) boot: Disabling RNG early entropy source...
I (439) cpu_start: Multicore app
I (448) cpu_start: Pro cpu start user code
I (448) cpu_start: cpu freq: 160000000 Hz
I (448) cpu_start: Application information:
I (451) cpu_start: Project name:     app-template
I (457) cpu_start: App version:      9737206-dirty
I (462) cpu_start: Compile time:     Oct  2 2024 15:54:16
I (468) cpu_start: ELF file SHA256:  d41fe71de...
I (474) cpu_start: ESP-IDF:          v5.2.2
I (478) cpu_start: Min chip rev:     v0.0
I (483) cpu_start: Max chip rev:     v0.99 
I (488) cpu_start: Chip rev:         v0.1
I (493) heap_init: Initializing. RAM available for dynamic allocation:
I (500) heap_init: At 3FCAB6E0 len 0003E030 (248 KiB): RAM
I (506) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (512) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (518) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (526) spi_flash: detected chip: gd
I (529) spi_flash: flash io: dio
W (533) cpu_start: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
E (547) flash_encrypt: Flash encryption eFuse bit was not enabled in bootloader but CONFIG_SECURE_FLASH_ENC_ENABLED is on

abort() was called at PC 0x42040ec2 on core 0
0x42040ec2: esp_flash_encryption_init_checks at C:/Espressif/frameworks/esp-idf-v5.2.2/components/bootloader_support/src/flash_encrypt.c:34
```

### More Information.

previously I have activated SBV2 by burning the real eFuse permanently. Now I want to test the flash encryption but only in virtual mode

AdityaHPatwardhan, 2024-10-03:
Hi @aygh4266 The security features such as flash encryption cannot be tested with the Virtual eFuse mode.

If you see the `help` section of the eFuse virtual mode config then you can see the following line

> If it is "y", then SECURE_FLASH_ENCRYPTION_MODE_RELEASE cannot be used. Because the EFUSE VIRT mode is for testing only.

([ref](https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/components/efuse/Kconfig#L27)

mahavirj, 2024-10-22:
Additional pointers about trying out security features under qemu: 

- https://developer.espressif.com/blog/trying-out-esp32-c3s-security-features-using-qemu/
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/tools/qemu.html

closed 2024-10-22 by mahavirj