espressif/esp-idf #14662: Log interceptor support

denravonska, 2024-10-02:
### Is your feature request related to a problem?

We want to be able to send some of the log messages from the logging framework to BLE to be able diagnose problems. However, we don't want to send _everything_ to BLE, just log statements from a selection of our own components.

### Describe the solution you'd like.

Similar to how you are able to register a custom `vprintf` function it would be nice to be able to register a pre-format hook that includes the tag pointer. That way we we'd be able to register tags-of-interest whose logs would be sent to BLE in addition to the regular IDF logger.

### Describe alternatives you've considered.

This is technically possible to achieve today by parsing the format string and compare any pointer that points to a `char*`, but it's tedious and slow compare to one tag comparison.

jrahlf, 2024-10-04:
You can use the wrap feature of GCC to overwrite `esp_log_write`, like the following (also need to specifiy linker flags). Then you can react to certain tags or certain log levels.

```c
// overrides the default `esp_log_write` function for custom handling
extern "C" void __wrap_esp_log_write(esp_log_level_t level,
                   const char *tag,
                   const char *format, ...)
{

    va_list list;
    va_start(list, format);

    esp_log_writev(level, tag, format, list);

	if (level == esp_log_level_t::ESP_LOG_ERROR) {
		// handle errors
	}
	
    va_end(list);
}
```

denravonska, 2024-10-04:
Oh, that's true! Yup, that would certainly be a good alternative.

SoucheSouche, 2024-10-09:
Hi @denravonska, I will close thee ticket since it seems that your query was answered by @jrahlf. Feel free to reopen if you need any further support with this topic.

closed 2024-10-09 by SoucheSouche