espressif/esp-idf #14664: Lillygo t-Camera esp32-s3 is giving the gdma: gdma_disconnect(234): no peripheral is connected to the channel error in esp-idf version 5.1.2

nitinGangurde, 2024-10-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I (335) s3 ll_cam: DMA Channel=0
I (335) cam_hal: cam init ok
I (335) sccb: pin_sda 5 pin_scl 4
I (345) sccb: sccb_i2c_port=1
I (345) gpio: GPIO[39]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (385) camera: Detected camera at address=0x30
I (385) camera: Detected OV2640 camera
I (385) camera: Camera PID=0x26 VER=0x42 MIDL=0x7f MIDH=0xa2
I (465) s3 ll_cam: node_size: 7680, nodes_per_line: 1, lines_per_node: 16
I (465) s3 ll_cam: dma_half_buffer_min:  7680, dma_half_buffer:  7680, lines_per_half_buffer: 16, dma_buffer_size: 30720
I (475) cam_hal: buffer_size: 30720, half_buffer_size: 7680, node_buffer_size: 7680, node_cnt: 4, total_cnt: 15
I (485) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
E (485) cam_hal: cam_dma_config(301): frame buffer malloc failed
E (495) cam_hal: cam_config(385): cam_dma_config failed
E (505) gdma: gdma_disconnect(234): no peripheral is connected to the channel
E (505) camera: Camera config failed with error 0xffffffff
E (515) example:take_picture: Camera Initialization Unsuccessful
E (525) example:take_picture: Camera Init Failed
I (525) main_task: Returned from app_main()

nitinGangurde, 2024-10-03:
Below is the code 
#include <esp_log.h>
#include <esp_system.h>
#include <nvs_flash.h>
#include <sys/param.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

// support IDF 5.x
#ifndef portTICK_RATE_MS
#define portTICK_RATE_MS portTICK_PERIOD_MS
#endif

#include "esp_camera.h"

#define BOARD_WROVER_KIT 1

// WROVER-KIT PIN Map
#ifdef BOARD_WROVER_KIT

#define CAM_PIN_PWDN -1  //power down is not used
#define CAM_PIN_RESET 39 //software reset will be performed
#define CAM_PIN_XCLK 38
#define CAM_PIN_SIOD 5
#define CAM_PIN_SIOC 4

#define CAM_PIN_D7 9
#define CAM_PIN_D6 10
#define CAM_PIN_D5 11
#define CAM_PIN_D4 13
#define CAM_PIN_D3 21
#define CAM_PIN_D2 48
#define CAM_PIN_D1 47
#define CAM_PIN_D0 14
#define CAM_PIN_VSYNC 8
#define CAM_PIN_HREF 18
#define CAM_PIN_PCLK 12

#endif

// ESP32Cam (AiThinker) PIN Map
#ifdef BOARD_ESP32CAM_AITHINKER

#define CAM_PIN_PWDN 32
#define CAM_PIN_RESET -1 //software reset will be performed
#define CAM_PIN_XCLK 0
#define CAM_PIN_SIOD 26
#define CAM_PIN_SIOC 27

#define CAM_PIN_D7 35
#define CAM_PIN_D6 34
#define CAM_PIN_D5 39
#define CAM_PIN_D4 36
#define CAM_PIN_D3 21
#define CAM_PIN_D2 19
#define CAM_PIN_D1 18
#define CAM_PIN_D0 5
#define CAM_PIN_VSYNC 25
#define CAM_PIN_HREF 23
#define CAM_PIN_PCLK 22

#endif

static const char *TAG = "example:take_picture";

#if ESP_CAMERA_SUPPORTED
static camera_config_t camera_config = {
    .pin_pwdn = CAM_PIN_PWDN,
    .pin_reset = CAM_PIN_RESET,
    .pin_xclk = CAM_PIN_XCLK,
    .pin_sccb_sda = CAM_PIN_SIOD,
    .pin_sccb_scl = CAM_PIN_SIOC,

    .pin_d7 = CAM_PIN_D7,
    .pin_d6 = CAM_PIN_D6,
    .pin_d5 = CAM_PIN_D5,
    .pin_d4 = CAM_PIN_D4,
    .pin_d3 = CAM_PIN_D3,
    .pin_d2 = CAM_PIN_D2,
    .pin_d1 = CAM_PIN_D1,
    .pin_d0 = CAM_PIN_D0,
    .pin_vsync = CAM_PIN_VSYNC,
    .pin_href = CAM_PIN_HREF,
    .pin_pclk = CAM_PIN_PCLK,

    //XCLK 20MHz or 10MHz for OV2640 double FPS (Experimental)
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,

    .pixel_format = PIXFORMAT_RGB565, //YUV422,GRAYSCALE,RGB565,JPEG
    .frame_size = FRAMESIZE_240X240,    //QQVGA-UXGA, For ESP32, do not use sizes above QVGA when not JPEG. The performance of the ESP32-S series has improved a lot, but JPEG mode always gives better frame rates.

    .jpeg_quality = 12, //0-63, for OV series camera sensors, lower number means higher quality
    .fb_count = 2,       //When jpeg mode is used, if fb_count more than one, the driver will work in continuous mode.
    .fb_location = CAMERA_FB_IN_PSRAM,
    .grab_mode = CAMERA_GRAB_WHEN_EMPTY,
};

static esp_err_t init_camera(void)
{
    //initialize the camera
    esp_err_t err = esp_camera_init(&camera_config);
    if (err != ESP_OK)
    {
    	ESP_LOGE(TAG, "Camera Initialization Unsuccessful");
        ESP_LOGE(TAG, "Camera Init Failed");
        return err;
    }

    ESP_LOGE(TAG, "Camera Initialization Successful");

    return ESP_OK;
}
#endif

void app_main(void)
{

#if ESP_CAMERA_SUPPORTED
    if(ESP_OK != init_camera()) {
        return;
    }

    while (1)
    {
        ESP_LOGI(TAG, "Taking picture...");
        camera_fb_t *pic = esp_camera_fb_get();

        // use pic->buf to access the image
        ESP_LOGI(TAG, "Picture taken! Its size was: %zu bytes", pic->len);
        esp_camera_fb_return(pic);

        vTaskDelay(5000 / portTICK_RATE_MS);
    }
#else
    ESP_LOGE(TAG, "Camera support is not available for this chip");
    return;
#endif
}

igrr, 2024-10-03:
The actual error is here:

> I (485) cam_hal: Allocating 115200 Byte frame buffer in PSRAM
> E (485) cam_hal: cam_dma_config(301): frame buffer malloc failed

Please check that you have sufficient PSRAM available for the frame buffer.