espressif/esp-idf #14666: The esp_local_ctrl example doesn't work

pavel808, 2024-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have build the `esp_local_ctrl `example and flashed it to an ESP32-C3 device (on an ESP Thread Border Router). It connects to my local Wi-Fi. The HTTP server starts to listen on port 443. 

Now this is where the problem is. The instructions provided in the `esp_local_ctrl/README.md`. 

It states to first `ping my_esp_ctrl_device.local`. 
The DNS cannot be resolved. 

But if I do `ping my_esp_ctrl_device`, this pings ok. It makes sense from looking at the service name in the `app_main.cpp ` it is just `my_esp_ctrl_device`

Connection fails on trying to run the following command : 

`python scripts/esp_local_ctrl.py --sec_ver 2 --sec2_username localctrl --sec2_pwd abcd1234`

However, if I remove the .local extension from line 38 of the `esp_local_ctrl.py,` so that the service name is just `my_esp_ctrl_device`, on connecting, I then get the following errors from the script :

```
++++ Connecting to my_esp_ctrl_device++++
Connection Failure : [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Hostname mismatch, certificate is not valid for 'my_esp_ctrl_device'. (_ssl.c:1007)
Traceback (most recent call last):
  File "/home/user/esp-idf-new/examples/protocols/esp_local_ctrl/scripts/esp_local_ctrl.py", line 437, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/user/esp-idf-new/examples/protocols/esp_local_ctrl/scripts/esp_local_ctrl.py", line 342, in main
    raise RuntimeError('Failed to establish connection')
RuntimeError: Failed to establish connection
```

And from the device : 

```
(9453) control: Development mode: using hard coded salt
I (9463) wifi:<ba-add>idx:1 (ifx:0, 2c:3a:fd:af:ca:0e), tid:0, ssn:0, winSize:64
I (9473) control: Development mode: using hard coded verifier
I (9483) esp_https_server: Starting server
I (9483) esp_https_server: Server listening on port 443
I (9483) control: esp_local_ctrl service started with name : my_esp_ctrl_device
I (156043) esp_https_server: performing session handshake
E (156713) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (156713) esp_https_server: esp_tls_create_server_session failed
E (156723) httpd: httpd_accept_conn: session creation failed
W (156723) httpd: httpd_server: error accepting new connection
```

Please help:)

chegewara, 2024-10-04:
Did you try to change esp32 example to match certificate name '.local'?

pavel808, 2024-10-04:
> Did you try to change esp32 example to match certificate name '.local'?

I did also that. It still can't resolve the DNS for some reason.

chegewara, 2024-10-04:
I think it's probably not the issue with example, thus using ".local" domain is not best choice.
It's most likely this problem 

https://askubuntu.com/questions/1068131/ubuntu-18-04-local-domain-dns-lookup-not-working

Please try this solution 
```
Essentially put two lines in /etc/mdns.allow:

.local. 
.local
```

pavel808, 2024-10-04:
> I think it's probably not the issue with example, thus using ".local" domain is not best choice. It's most likely this problem [...]

That doesn't work for me unfortunately.

pavel808, 2024-10-04:
@chegewara I will try various changes in Ubuntu. 

For a start, should this line be change in the application ?: 

`#define SERVICE_NAME "my_esp_ctrl_device"`

to 

`#define SERVICE_NAME "my_esp_ctrl_device.local"`

chegewara, 2024-10-04:
In theory yes.
I have no time right now, but I will try to test it too.
Maybe tomorrow.

Harshal5, 2024-10-07:
Hello @pavel808 thanks for the report.

I tested the example on a MacOS device and it seems to resolve the `my_esp_ctrl_device.local` service name and the example works as expected. Whereas, when I tried running the `ping` command on a Linux device, it failed to connect.
The fix that helped me to resolve the service name: [https://askubuntu.com/a/1114887](https://askubuntu.com/a/1114887).

Please help try out the fix discussed here and let me know if this addresses the issue on your side.

Thanks!

Harshal5, 2024-10-29:
Closing this issue as of now, please feel free to re-open if the issue persists.

closed 2024-10-29 by Harshal5