espressif/esp-idf #14667: Cannot set ESP32 S3 I2C clock stretching - fixed at 20ms

HarryFairhead, 2024-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board clone of devkitC

### Power Supply used.

USB

### What is the expected behavior?

Read the HTU21D temperature measurement

### What is the actual behavior?

Times out

### Steps to reproduce.

1) Connect a HTU21 to ESP32 S3 using any GPIO  pins used 15 and 16 plus 3.3V and GND.

2) run program:
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "driver/i2c_master.h"
#include "hal/i2c_ll.h"
#include "driver/gpio.h"
#include "soc/i2c_reg.h"
// #include "soc/i2c_struct.h"

void app_main(void)
{

    gpio_reset_pin(2);
    gpio_set_direction(2, GPIO_MODE_INPUT_OUTPUT);
    i2c_master_bus_config_t i2cBus = {
        .i2c_port = I2C_NUM_0,
        .scl_io_num = 15,
        .sda_io_num = 16,
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .glitch_ignore_cnt = 7,
        .flags.enable_internal_pullup = true};
    i2c_master_bus_handle_t bus_handle;
    i2c_new_master_bus(&i2cBus, &bus_handle);

    i2c_device_config_t i2cdev = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = 0x40,
        .scl_speed_hz = 100000,
        .scl_wait_us = 10
    };
    esp_err_t ret;
        i2c_master_dev_handle_t dev_handle;
    i2c_master_bus_add_device(bus_handle, &i2cdev, &dev_handle);
    uint8_t buf[10] = { 0xE3 };
    ret = i2c_master_transmit(dev_handle, buf, 1, 100);
    ret = i2c_master_receive(dev_handle, buf, 3, 200);
    uint8_t msb = buf[0];
    uint8_t lsb = buf[1];
    uint8_t check = buf[2];
    printf("msb %d \n lsb %d \n checksum %d \n", msb, lsb, check);
    fflush(stdout);
}

Result is a timeout.  Logic analyzer shows that the slave pulls the clock low  
for 40ms. Adding a marker for when the  i2c_master_receive returns indicates that it gives up waiting after 20ms.

### Debug Logs.

```plain
I (301) main_task: Started on CPU0
I (321) main_task: Calling app_main()
I (321) gpio: GPIO[2]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (321) gpio: GPIO[16]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
I (331) gpio: GPIO[15]| InputEn: 1| OutputEn: 1| OpenDrain: 1| Pullup: 1| Pulldown: 0| Intr:0 
E (341) i2c.master: I2C transaction timeout detected
E (351) i2c.master: s_i2c_synchronous_transaction(872): I2C 
transaction failed
E (361) i2c.master: i2c_master_receive(1117): I2C transaction failed
msb 227
 lsb 0
 checksum 0 
I (371) main_task: Returned from app_main()
```

### More Information.

I have traced the code and spent a lot of time changing things at the register level but the 20ms timeout never changes no matter what I do.

Alvin1Zhang, 2024-10-04:
Thanks for reporting.

mythbuster5, 2024-10-09:
https://github.com/espressif/esp-idf/commit/deb91c7abe8b1fafc5e9410fd2fdfd2aeebcea57 already fixed here. on release/5.3 branch but not on 5.3.1, I'm going to close this as this has been solved

closed 2024-10-09 by mythbuster5