espressif/esp-idf #14668: Memory allocation problem occurs in AWS IoT OTA process

Gyunamchoi, 2024-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3 2nd stage bootloader

### Espressif SoC revision.

ESP32-WROOM-32UE chip revision: v3.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board (ESP32-WROOM-32UE)

### Power Supply used.

USB

### What is the expected behavior?

The ultimate goal is to proceed with OTA updates through aws-iot.

### What is the actual behavior?

An error related to heap_caps occurs when allocating memory such as malloc.

### Steps to reproduce.

1. Power on 
2. AWS IoT connection and MQTT connection via SIM7600EK modem with lwip(ppp) interface
3. Transmitting shadow and measurement information - works well
4. Create ota job in aws iot
5. AWS -> device : topic : $aws/things/(device id)/jobs/notify-next (OK)
6. device -> AWS : topic : $aws/things/(device id)/jobs/start-next (OK)
7.  AWS -> device : topic : $aws/things/testdev21/jobs/start-next/accepted (OK)
8. device -> AWS : $aws/things/(device id)/streams/(stream_id)/get/json ( message Publish ERROR ) 

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d1c6  PS      : 0x00060b33  A0      : 0x8008cb64  A1      : 0x3ffc40e0  
0x4008d1c6: remove_free_block at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:332
 (inlined by) block_locate_free at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:1015

A2      : 0x3ffd039c  A3      : 0x00000058  A4      : 0x00000002  A5      : 0x00000036  
A6      : 0x3ffc4c1c  A7      : 0x3ffb55f4  A8      : 0x00000006  A9      : 0xffffffff  
A10     : 0x00000003  A11     : 0xffffffff  A12     : 0x00000468  A13     : 0x3ffb5618  
A14     : 0x3ffc4240  A15     : 0x0000cdcd  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000b  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
0x4000c349: memcpy in ROM
[... 135 lines cut ...]
//                        char* msg_mqtt_start = ota_request_start_stream(json_otadocument);
                        ESP_LOGI(TAG, "topic_mqtt_start =%s  msg_mqtt_start = %s ", topic_mqtt_start, msg_mqtt_start);                    
                        **mqtt_publish(topic_mqtt_start, msg_mqtt_start);****
                        ESP_LOGI(TAG, "Start ota [%s] from factory mode to stream image file request complete.",stream_id);
                        free(topic_mqtt_start);
                        free(msg_mqtt_start);
                    } else {
`

1. When an MQTT event occurs, the event handler processes the event and passes the topic and data as char* arguments.
2. Parse the topic
3. If it is jobs, pass it to the jobs process
4. Receive the $aws/things/(device id)/jobs/notify-next topic and check if the status is queued.
5. Once the check is complete, publish the $aws/things/(device id)/jobs/start-next message
7. Then, AWS checks that the message status of the $aws/things/testdev21/jobs/start-next/accepted topic is IN PROGRESS as the device. 8. Once confirmed, publish the $aws/things/(device id)/streams/(stream_id)/get/json message to AWS
(In this process, a kernel panic occurs during memory allocation)

When creating cJSON, cJSON_CreateObject() uses malloc, so a kernel panic occurs, and the same problem occurs in esp_mqtt_publish() when publishing. A heap_caps_calloc error occurs when performing memory-related operations.
Whether creating with malloc, creating with fixed generation, or using cJSON, the same problem occurs in the same part.

There is about 170KB of free heap memory, and there are no problems at all in periodic message publishing, except for the jobs-related part.

I have no idea whether it is a problem with the mqtt module itself when publishing, a problem with another part, or a problem with json. Please help me.

Thank you.

nileshkale123, 2024-10-07:
Hello @Gyunamchoi,

Thank you for bringing up the issue.

Could you please print the free heap memory before the function fails? Alternatively, if possible, could you create a small working example to reproduce the issue or integrate it into one of the existing IDF examples?

Gyunamchoi, 2024-10-08:
Hello @nileshkale123 

Could you wait for 2~3days?
I will summarize and deliver the following responses.
1. I will deliver all heap dump data from heap_caps_dump_all()
2. I will extract the part related to processing the aws iot jobs reserved command among the mqtt handlers or integrate it into the existing sample code and deliver it.

2-1) However, since it is a problem that occurs when connecting to the aws iot core service and creating an ota job, I wonder if the environment is suitable for testing the aws iot core
2-2) I will deliver the job creation process in the aws iot service later when uploading the code or heap data.

Thank you for your response.

Gyunamchoi, 2024-10-11:
Hello @nileshkale123 

This is all heap data just before the problem occurred.
`
Showing data for heap: 0x3ffae6e0 
Block 0x3ffae860 data, size: 184 bytes, Free: No 
Block 0x3ffae91c data, size: 12 bytes, Free: No 
Block 0x3ffae92c data, size: 12 bytes, Free: No 
Block 0x3ffae93c data, size: 16 bytes, Free: No 
Block 0x3ffae950 data, size: 196 bytes, Free: No 
Block 0x3ffaea18 data, size: 196 bytes, Free: No 
Block 0x3ffaeae0 data, size: 432 bytes, Free: No 
Block 0x3ffaec94 data, size: 88 bytes, Free: No 
Block 0x3ffaecf0 data, size: 88 bytes, Free: No 
Block 0x3ffaed4c data, size: 88 bytes, Free: No 
Block 0x3ffaeda8 data, size: 132 bytes, Free: No 
Block 0x3ffaee30 data, size: 132 bytes, Free: No 
Block 0x3ffaeeb8 data, size: 16 bytes, Free: No 
Block 0x3ffaeecc data, size: 88 bytes, Free: No 
Block 0x3ffaef28 data, size: 88 bytes, Free: No 
Block 0x3ffaef84 data, size: 16 bytes, Free: No 
Block 0x3ffaef98 data, size: 12 bytes, Free: No 
Block 0x3ffaefa8 data, size: 88 bytes, Free: No 
Block 0x3ffaf004 data, size: 1028 bytes, Free: No 
Block 0x3ffaf40c data, size: 344 bytes, Free: No 
Block 0x3ffaf568 data, size: 1028 bytes, Free: No 
Block 0x3ffaf970 data, size: 344 bytes, Free: No 
Block 0x3ffafacc data, size: 344 bytes, Free: No 
Block 0x3ffafc28 data, size: 12 bytes, Free: No 
Block 0x3ffafc38 data, size: 16 bytes, Free: No 
Block 0x3ffafc4c data, size: 12 bytes, Free: No 
Block 0x3ffafc5c data, size: 16 bytes, Free: No 
Block 0x3ffafc70 data, size: 344 bytes, Free: No 
Block 0x3ffafdcc data, size: 344 bytes, Free: No 
Block 0x3ffaff28 data, size: 88 bytes, Free: No 
Block 0x3ffaff84 data, size: 16 bytes, Free: No 
Block 0x3ffaff98 data, size: 100 bytes, Free: No 
Showing data for heap: 0x3ffb55e0 
Block 0x3ffb5990 data, size: 4100 bytes, Free: No 
Block 0x3ffb6998 data, size: 4100 bytes, Free: No 
Block 0x3ffb79a0 data, size: 1540 bytes, Free: No 
Block 0x3ffb7fa8 data, size: 1540 bytes, Free: No 
Block 0x3ffb85b0 data, size: 344 bytes, Free: No 
Block 0x3ffb870c data, size: 20 bytes, Free: No 
Block 0x3ffb8724 data, size: 16 bytes, Free: No 
Block 0x3ffb8738 data, size: 20 bytes, Free: No 
Block 0x3ffb8750 data, size: 20 bytes, Free: No 
Block 0x3ffb8768 data, size: 28 bytes, Free: No 
Block 0x3ffb8788 data, size: 600 bytes, Free: No 
Block 0x3ffb89e4 data, size: 88 bytes, Free: No 
Block 0x3ffb8a40 data, size: 2820 bytes, Free: No 
Block 0x3ffb9548 data, size: 344 bytes, Free: No 
Block 0x3ffb96a4 data, size: 88 bytes, Free: No 
Block 0x3ffb9700 data, size: 44 bytes, Free: No 
Block 0x3ffb9730 data, size: 88 bytes, Free: No 
Block 0x3ffb978c data, size: 48 bytes, Free: No 
Block 0x3ffb97c0 data, size: 48 bytes, Free: No 
Block 0x3ffb97f4 data, size: 20 bytes, Free: No 
Block 0x3ffb980c data, size: 24 bytes, Free: No 
Block 0x3ffb9828 data, size: 56 bytes, Free: No 
Block 0x3ffb9864 data, size: 56 bytes, Free: No 
Block 0x3ffb98a0 data, size: 56 bytes, Free: No 
Block 0x3ffb98dc data, size: 56 bytes, Free: No 
Block 0x3ffb9918 data, size: 56 bytes, Free: No 
Block 0x3ffb9954 data, size: 56 bytes, Free: No 
Block 0x3ffb9990 data, size: 108 bytes, Free: No 
Block 0x3ffb9a00 data, size: 344 bytes, Free: No 
Block 0x3ffb9b5c data, size: 132 bytes, Free: No 
Block 0x3ffb9be4 data, size: 132 bytes, Free: No 
Block 0x3ffb9c6c data, size: 132 bytes, Free: No 
[... 106 lines cut ...]

To explain the log,
" Event dispatched from event loop event_id=6 " => 데이터 이벤트 발생 
`
01-02 : aws -> device :  $aws/things/(id)/jobs/notify-next 수신 ( status QUEUED ) 
03 : device -> aws : $aws/things/(id)/jobs/start-next 로 응답 ( status QUEUED ) 
04-05 : aws -> device :  $aws/things/(id)/jobs/start-next 수신 ( status QUEUED ) 
06-08 : aws -> device :  $aws/things/(id)/jobs/start-next/accepted 수신 ( status IN_PROGRESS ) 
09 after : device -> aws : $aws/things/(id)/streams/(stream-id)/get/json (Cannot proceed due to error)

`

The data that the device needs to pass to AWS is as follows.
`char msg_mqtt_start[1000] ;
sprintf(msg_mqtt_start, "{\"c\":%s, \"s\":%d, \"f\":%d,\"l\":%d,\"o\":%d,\"n\":%d}",
    CONFIG_OTA_CLIENT_TOKEN, CONFIG_OTA_STREAM_VERSION, (int)ota_stream.total.file_size, 
    CONFIG_OTA_STREAM_BLOCK_SIZE, ota_stream.block_offset, ota_stream.request_block_num);
"{ 
    "c": "device-dev",
    "s": 1,
    "f": 730304,
    "l": 4096,
    "o": 0, 
    "n": 0
}"     `

Gyunamchoi, 2024-10-11:
> Hello @nileshkale123
>
> This is all heap data just before the problem occurred. `Showing data for heap: 0x3ffae6e0 Block 0x3ffae860 data, size: 184 bytes, Free: No Block 0x3ffae91c data, size: 12 [...]

It looks like the data from ota should be sent to the stream immediately after this, but an error occurs before that happens.

SoucheSouche, 2024-10-15:
@Gyunamchoi, it seems that the heap memory got corrupted along the line. you have to find out where exactly. You can use `heap_caps_check_integrity_all(true);` (`esp_heap_caps.h`) at different places in the code to find which part of your code induced the corruption. Make sure that the comprehensive poisoning is enabled in the menuconfig ( `Component config` → `Heap memory debugging` → `heap corruption detection`).

Gyunamchoi, 2024-10-21:
hello @SoucheSouche.

As you advised, I enabled comprehensive poisoning, added heap_caps_check_integrity_all(true); in several places, and checked for memory corruption.
This is the code of the corrupt heap right before the problem occurred.
Please check.

`I (50769) [mqtt]: Execute new OTA update task. (from factory partition) 
CORRUPT HEAP: Bad tail at 0x3ffcf320. Expected 0xbaad5678 got 0xffffffff
CORRUPT HEAP: Invalid data at 0x3ffcf334. Expected 0xfefefefe got 0x3ffc52b4
CORRUPT HEAP: Invalid data at 0x3ffcf338. Expected 0xfefefefe got 0x3767029e
CORRUPT HEAP: Invalid data at 0x3ffcf33c. Expected 0xfefefefe got 0x78141a98
CORRUPT HEAP: Invalid data at 0x3ffcf340. Expected 0xfefefefe got 0x1d7fce4e
CORRUPT HEAP: Invalid data at 0x3ffcf344. Expected 0xfefefefe got 0xadf51b98
CORRUPT HEAP: Invalid data at 0x3ffcf348. Expected 0xfefefefe got 0x8d482231
CORRUPT HEAP: Invalid data at 0x3ffcf34c. Expected 0xfefefefe got 0xd2d47c66
CORRUPT HEAP: Invalid data at 0x3ffcf350. Expected 0xfefefefe got 0x48d1ad6b
CORRUPT HEAP: Invalid data at 0x3ffcf354. Expected 0xfefefefe got 0xf45bfd9c
CORRUPT HEAP: Invalid data at 0x3ffcf358. Expected 0xfefefefe got 0xdbcbf209
CORRUPT HEAP: Invalid data at 0x3ffcf35c. Expected 0xfefefefe got 0xf4ed7d9c
CORRUPT HEAP: Invalid data at 0x3ffcf360. Expected 0xfefefefe got 0xe3302886
CORRUPT HEAP: Invalid data at 0x3ffcf364. Expected 0xfefefefe got 0x2c285da1
CORRUPT HEAP: Invalid data at 0x3ffcf368. Expected 0xfefefefe got 0x457f49b1
CORRUPT HEAP: Invalid data at 0x3ffcf36c. Expected 0xfefefefe got 0xd925344d
CORRUPT HEAP: Invalid data at 0x3ffcf370. Expected 0xfefefefe got 0x8e55f80a
CORRUPT HEAP: Invalid data at 0x3ffcf374. Expected 0xfefefefe got 0x5f5837f3
CORRUPT HEAP: Invalid data at 0x3ffcf378. Expected 0xfefefefe got 0xb8ff7f04
CORRUPT HEAP: Invalid data at 0x3ffcf37c. Expected 0xfefefefe got 0x3702ca47
CORRUPT HEAP: Invalid data at 0x3ffcf380. Expected 0xfefefefe got 0xb8c93118
CORRUPT HEAP: Invalid data at 0x3ffcf384. Expected 0xfefefefe got 0x0e9f0a15
CORRUPT HEAP: Invalid data at 0x3ffcf388. Expected 0xfefefefe got 0xba692d56
CORRUPT HEAP: Invalid data at 0x3ffcf38c. Expected 0xfefefefe got 0xed367bec
CORRUPT HEAP: Invalid data at 0x3ffcf390. Expected 0xfefefefe got 0xe090a69a
CORRUPT HEAP: Invalid data at 0x3ffcf394. Expected 0xfefefefe got 0xd6097dbf
CORRUPT HEAP: Invalid data at 0x3ffcf398. Expected 0xfefefefe got 0x2ba89c60
CORRUPT HEAP: Invalid data at 0x3ffcf39c. Expected 0xfefefefe got 0xe9e7d1c2
CORRUPT HEAP: Invalid data at 0x3ffcf3a0. Expected 0xfefefefe got 0xaa32d2de
CORRUPT HEAP: Invalid data at 0x3ffcf3a4. Expected 0xfefefefe got 0x68a640d7
CORRUPT HEAP: Invalid data at 0x3ffcf3a8. Expected 0xfefefefe got 0x62264983
CORRUPT HEAP: Invalid data at 0x3ffcf3ac. Expected 0xfefefefe got 0x4355f993
CORRUPT HEAP: Invalid data at 0x3ffcf3b0. Expected 0xfefefefe got 0x56fbca0d
CORRUPT HEAP: Invalid data at 0x3ffcf3b4. Expected 0xfefefefe got 0x5baf764c
CORRUPT HEAP: Invalid data at 0x3ffcf3b8. Expected 0xfefefefe got 0xbaad5678
CORRUPT HEAP: Invalid data at 0x3ffcf3bc. Expected 0xfefefefe got 0x00000649
CORRUPT HEAP: Invalid data at 0x3ffcf3c0. Expected 0xfefefefe got 0x3ffb5664
CORRUPT HEAP: Invalid data at 0x3ffcf3c4. Expected 0xfefefefe got 0x3ffb5664
CORRUPT HEAP: Invalid data at 0x3ffcfa04. Expected 0xfefefefe got 0x3ffcf3b8
CORRUPT HEAP: Invalid data at 0x3ffcfa08. Expected 0xfefefefe got 0x00000042
CORRUPT HEAP: Invalid data at 0x3ffcfa0c. Expected 0xfefefefe got 0xabba1234
CORRUPT HEAP: Invalid data at 0x3ffcfa10. Expected 0xfefefefe got 0x0000002c
CORRUPT HEAP: Invalid data at 0x3ffcfa14. Expected 0xfefefefe got 0x3ffc52b4
CORRUPT HEAP: Invalid data at 0x3ffcfa18. Expected 0xfefefefe got 0x3ffcebc8
CORRUPT HEAP: Invalid data at 0x3ffcfa1c. Expected 0xfefefefe got 0x00000046
CORRUPT HEAP: Invalid data at 0x3ffcfa20. Expected 0xfefefefe got 0x0000Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x40009222  PS      : 0x00060734  A0      : 0x80007d16  A1      : 0x3ffc4da0  
0x40009222: uart_tx_one_char in ROM

A2      : 0x00800000  A3      : 0x38000000  A4      : 0x00000000  A5      : 0x0000000d  
A6      : 0x3ffcf32c  A7      : 0x3ffcf324  A8      : 0x3ff40000  A9      : 0x00000037  
A10     : 0x00800000  A11     : 0x3ff4001c  A12     : 0x00000000  A13     : 0x3ffc4f30  
A14     : 0xffffffff  A15     : 0x00000001  SAR     : 0x00000005  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
0x4000c349: memcpy in ROM
0x4000c36b: memcpy in ROM

Backtrace: 0x4000921f:0x3ffc4da0 0x40007d13:0x3ffc4dc0 0x40007c69:0x3ffc4de0 0x40008106:0x3ffc4e00 0x4008cea1:0x3ffc4e90 0x4008ccb1:0x3ffc4eb0 0x40148e9b:0x3ffc4ed0 0x40149201:0x3ffc4ef0 0x40149343:0x3ffc4f10 0x4014938f:0x3ffc4f30 0x40148ee7:0x3ffc4f60 0x400d42f5:0x3ffc4f80 0x400d4319:0x3ffc4fa0 0x400dce47:0x3ffc4fc0 0x400dbd82:0x3ffc4fe0 0x400dc509:0x3ffc5030 0x400dc882:0x3ffc5050 0x401481bd:0x3ffc50f0 0x40147c52:0x3ffc5120 0x400e4dfe:0x3ffc5160 0x400e5860:0x3ffc5180 0x400e59c7:0x3ffc51b0 0x400e5ec4:0x3ffc51e0 0x40088425:0x3ffc5210
0x4000921f: uart_tx_one_char in ROM
0x40007d13: ets_write_char_uart in ROM
0x40007c69: ets_write_char in ROM
[... 15 lines cut ...]
0x40147c52: esp_event_loop_run at /Users/odn/esp/esp-idf/components/esp_event/esp_event.c:584
0x400e4dfe: esp_mqtt_dispatch_event at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1032
[... cut ...]
0x40085917: xt_utils_wait_for_intr at /Users/odn/esp/esp-idf/components/xtensa/include/xt_utils.h:82
 (inlined by) esp_cpu_wait_for_intr at /Users/odn/esp/esp-idf/components/esp_hw_support/cpu.c:55
0x400d3dd3: esp_vApplicationIdleHook at /Users/odn/esp/esp-idf/components/esp_system/freertos_hooks.c:58
0x400894e9: prvIdleTask at /Users/odn/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4344 (discriminator 1)
0x40088425: vPortTaskWrapper at /Users/odn/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134`