espressif/esp-idf #14669: restful_server crashing

pavel808, 2024-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to get the `protocols/http_server/restful_server` example running on an ESP32-S3 on the Thread Border Router device. It keeps crashing as below. Is it an issue of not having enough flash memory on that device, or something else? 

```
[0;32mI (26) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader[0m
[0;32mI (26) boot: compile time Oct  4 2024 13:39:56[0m
[0;32mI (26) boot: Multicore bootloader[0m
[0;32mI (30) boot: chip revision: v0.2[0m
[0;32mI (34) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (38) boot.esp32s3: SPI Mode       : DIO[0m
[0;32mI (43) boot.esp32s3: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Enabling RNG early entropy source...[0m
[0;32mI (53) boot: Partition Table:[0m
[0;32mI (57) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (79) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (86) boot:  3 www              Unknown data     01 82 00110000 00200000[0m
[0;32mI (94) boot: End of partition table[0m
[0;32mI (98) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=2628ch (156300) map[0m
[0;32mI (135) esp_image: segment 1: paddr=000362b4 vaddr=3fc97e00 size=04650h ( 18000) load[0m
[0;32mI (139) esp_image: segment 2: paddr=0003a90c vaddr=40374000 size=0570ch ( 22284) load[0m
[0;32mI (145) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=99560h (628064) map[0m
[0;32mI (261) esp_image: segment 4: paddr=000d9588 vaddr=4037970c size=0e5ech ( 58860) load[0m
[0;32mI (283) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (283) boot: Disabling RNG early entropy source...[0m
[0;32mI (295) cpu_start: Multicore app[0m
[0;32mI (305) cpu_start: Pro cpu start user code[0m
[0;32mI (305) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (305) cpu_start: Application information:[0m
[0;32mI (308) cpu_start: Project name:     restful_server[0m
[0;32mI (313) cpu_start: App version:      v5.4-dev-3201-g46acfdce96-dirty[0m
[0;32mI (320) cpu_start: Compile time:     Oct  4 2024 13:39:54[0m
[0;32mI (326) cpu_start: ELF file SHA256:  fee019d60...[0m
[0;32mI (332) cpu_start: ESP-IDF:          v5.2.1-dirty[0m
[0;32mI (337) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (342) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (347) cpu_start: Chip rev:         v0.2[0m
[0;32mI (351) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (359) heap_init: At 3FCA0678 len 00049098 (292 KiB): RAM[0m
[0;32mI (365) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM[0m
[0;32mI (371) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM[0m
[0;32mI (377) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM[0m
[0;32mI (384) spi_flash: detected chip: gd[0m
[0;32mI (387) spi_flash: flash io: dio[0m
[0;33mW (391) spi_flash: Detected size(8192k) larger than the size in the binary image header(4096k). Using the size in the binary image header.[0m
[0;32mI (405) sleep: Configure to isolate all GPIO pins in sleep state[0m
[0;32mI (411) sleep: Enable automatic switching of GPIO sleep configuration[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42009dfc
file: "/IDF-new/examples/protocols/http_server/restful_server/main/esp_rest_main.c" line 136
func: app_main
expression: init_fs()
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375984
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914

```

chegewara, 2024-10-04:
Did you check if sd card is connected in 4 line mode and the pins are correct?

nileshkale123, 2024-10-29:
Hello @pavel808,

Does the issue still persist after trying the recommended suggestions by @chegewara?

nileshkale123, 2024-11-07:
Hello @pavel808,

I tried running this example at my end, and I didn't face any issue. Please go through the step-by-step guide as per the README and try it again carefully. 

I am closing this issue for now. If you face this again, please feel free to reopen it with more details information and debug logs.

closed 2024-11-07 by nileshkale123