espressif/esp-idf #14670: Send a command to a device using 4 characteristics

yanmartins, 2024-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello!

I'm using ESPIDF version 5.3.1 with a ESP32-S3.
I flash the ESP as a BLE client with the example [gatt_client](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_client) and adapted it to try make it compatible with the device.

This device has 4 characteristics:

> 1. UART Data TX Characteristics
> UUID: 00000002-0000-1000-8000-008025000000
> Type: uint8 array (20 bytes)
> Properties: Notify
> The Terminal I/O client uses the UART Data TX characteristic to receive UART data from the server (peripheral) via notifications.
> Terminal I/O Profile Client Implementation Guide
> 
> 2. UART Data RX Characteristics
> UUID: 00000001-0000-1000-8000-008025000000
> Type: uint8 array (20 bytes)
> Properties: Write without response
> The Terminal I/O client uses the UART Data RX characteristic to write UART data to the server (peripheral).
> 
> 3. UART Credits TX Characteristics
> UUID: 00000004-0000-1000-8000-008025000000
> Type: uint8 (1 byte)
> Properties: Indicate
> The Terminal I/O client uses the UART Credits TX characteristic to receive UART credits (0 - 255) from the server (peripheral) via indications.
> 
> 4. UART Credits RX Characteristics
> UUID: 00000003-0000-1000-8000-008025000000
> Type: uint8 (1 byte)
> Properties: Write
> The Terminal I/O client uses the UART Credits RX characteristic to grant UART credits (0 - 255) to the server (peripheral).

Here is the file with my changes:
[gattc_demo.txt](https://github.com/user-attachments/files/17258420/gattc_demo.txt) (Yes, there are redundant parts, it is a test xD)

The flow is:
1. Active `Indicate`
2. Activate `Notify`
3. Use `Write` to set the UART credits size
4. Use `Write without response` to send a command

The log:
```
I (1164) GATTC_DEMO:  
I (1174) GATTC_DEMO:  
I (1174) GATTC_DEMO:  
W (1184) GATTC_DEMO: searched device perma
W (1184) GATTC_DEMO: connect to the remote device.
I (1184) GATTC_DEMO: stop scan successfully
I (1184) GATTC_DEMO:  
I (1194) GATTC_DEMO:  
I (1194) GATTC_DEMO:  
I (1194) GATTC_DEMO:  
W (5264) GATTC_DEMO: EVENT: 40
I (5264) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (5264) GATTC_DEMO: REMOTE BDA:
I (5274) GATTC_DEMO: 00 21 7e 70 fa e3 
W (5274) GATTC_DEMO: EVENT: 2
I (5274) GATTC_DEMO: open success
I (5364) GATTC_DEMO: packet length updated: rx = 162, tx = 162, status = 0
W (6294) GATTC_DEMO: EVENT: 46
I (6294) GATTC_DEMO: discover service complete conn_id 0
W (6294) GATTC_DEMO: EVENT: 7
I (6294) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (6304) GATTC_DEMO: start handle 17 end handle 65535 current handle value 17
I (6304) GATTC_DEMO: service found
I (6314) GATTC_DEMO: UUID16: fefb
W (6314) GATTC_DEMO: EVENT: 6
I (6324) GATTC_DEMO: Get service information from remote device
I (6324) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (6334) GATTC_DEMO: count: 6
E (6334) GATTC_DEMO: ASK FOR REG WRITE
E (6344) GATTC_DEMO: ASK FOR REG WRITE_NR
E (6344) GATTC_DEMO: ASK FOR REG NOTIFY
E (6344) GATTC_DEMO: ASK FOR REG INDICATE
W (6354) GATTC_DEMO: EVENT: 18
I (6354) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 158, conn_id 0
W (6364) GATTC_DEMO: EVENT: 38
I (6364) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
W (6374) GATTC_DEMO: EVENT: 38
I (6374) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
W (6444) GATTC_DEMO: EVENT: 9
I (6444) GATTC_DEMO: write descr success 
W (6494) GATTC_DEMO: EVENT: 9
I (6494) GATTC_DEMO: write descr success 
W (6494) ***: WRITE
W (6494) ***: WRITE NR
E (6494) ---: msg_len: 18
W (6544) GATTC_DEMO: EVENT: 4
I (6544) GATTC_DEMO: write char success 
W (6544) GATTC_DEMO: EVENT: 4
I (6544) GATTC_DEMO: write char success 
```
I tested sending commands using nRF Connect app and the device performs the actions correctly.

What could I be doing wrong?

esp-zhp, 2024-10-08:
What is your expected behavior?

yanmartins, 2024-10-08:
> What is your expected behavior?

@esp-zhp When sending a message in the "Write without response" characteristic, the device should react to this command

yanmartins, 2024-10-08:
@esp-zhp  The flow is:

1. Active Indicate
2. Activate Notify
3. Use Write to set the UART credits size
4. Use Write without response to send a command

I don't have access to the device. But I have visual feedback when a command is successful.

Using the nRF Connect for Mobile app on an Android phone, I ran this sequence of commands and it worked

yanmartins, 2024-10-08:
I activate some verbose logs:

```
W (43768) ***: WRITE
D (43768) BT_GATT: gatt_get_ch_state: ch_state=4
D (43768) BT_BTC: btc_transfer_context msg 0 3 12 0x3fca45e4

D (43768) BT_BTC: btc_thread_handler msg 0 3 12 0x3fcb63a0

D (43768) BT_GATT: gatt_get_ch_state: ch_state=4
I (43778) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 00217e70fae3
D (43788) BT_L2CAP: l2c_link_check_send_pkts
D (43788) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (43798) BT_L2CAP: l2c_link_send_to_lower
D (43798) BT_L2CAP: TotalWin=11,Hndl=0x1,Quota=12,Unack=1,RRQuota=0,RRUnack=0
D (43808) BT_OSI: alarm_cbs_lookfor_available 4 0x3fc9ef38

D (43808) BT_OSI: osi_alarm_cancel failed to stop timer, err 0x103

D (43818) BT_HCI: Receive packet event_code=0x13

D (43828) BT_L2CAP: l2c_link_check_send_pkts
D (43828) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (43838) BT_L2CAP: TotalWin=12,LinkUnack(0x1)=0,RRCheck=0,RRUnack=0

D (43848) BT_HCI: HCI Enqueue Command opcode=0xc35

D (43848) BT_HCI: reassemble_and_dispatch

D (43848) BT_L2CAP: L2CAP - rcv_cid CID: 0x0004

D (43858) BT_GATT: gatt_get_ch_state: ch_state=4
D (43858) BT_GATT: gatt_end_operation status=0 op=3 subtype=2
D (43868) BT_OSI: osi_alarm_cancel failed to stop timer, err 0x103

D (43868) BT_BTC: btc_transfer_context msg 1 3 4 0x3fcb3730

D (43878) BT_BTC: btc_thread_handler msg 1 3 4 0x3fcb5dc8

W (43888) GATTC_DEMO: EVENT: 4
I (43888) GATTC_DEMO: write char success 
W (45818) ***: WRITE NR
E (45818) ---: msg_len: 18
D (45818) BT_GATT: gatt_get_ch_state: ch_state=4
D (45818) BT_BTC: btc_transfer_context msg 0 3 12 0x3fca45d4

D (45818) BT_BTC: btc_thread_handler msg 0 3 12 0x3fcb63a0

D (45828) BT_GATT: gatt_get_ch_state: ch_state=4
I (45828) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 00217e70fae3
D (45838) BT_L2CAP: l2c_link_check_send_pkts
D (45838) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (45848) BT_L2CAP: l2c_link_send_to_lower
D (45858) BT_L2CAP: TotalWin=11,Hndl=0x1,Quota=12,Unack=1,RRQuota=0,RRUnack=0
D (45858) BT_GATT: gatt_end_operation status=0 op=3 subtype=1
D (45868) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (45868) BT_HCI: Receive packet event_code=0x13

D (45878) BT_L2CAP: l2c_link_check_send_pkts
D (45878) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (45888) BT_L2CAP: TotalWin=12,LinkUnack(0x1)=0,RRCheck=0,RRUnack=0

D (45898) BT_BTC: btc_transfer_context msg 1 3 4 0x3fcb3730

D (45898) BT_BTC: btc_thread_handler msg 1 3 4 0x3fcb5dc8

W (45908) GATTC_DEMO: EVENT: 4
I (45908) GATTC_DEMO: write char success 
```

esp-zhp, 2024-10-09:
So far, I haven't encountered any issues.

You need to verify that the **handle** values used when calling **esp_ble_gattc_register_for_notify**, **esp_ble_gattc_write_char_descr**, and **esp_ble_gattc_write_char** are correct.

If you are certain that all the handles you are passing in are accurate, you can proceed with the next debugging steps:

Capture over-the-air packets (if you have a packet-sniffing device), or
Print the HCI log information.
If you don't have a packet-sniffing device, you can use the following code to print the HCI log information. First, enable the configuration option: (Top) → Component config → Bluetooth → [*] Enable Bluetooth HCI debug mode.
```
extern void bt_hci_log_hci_data_show(void);
extern void bt_hci_log_hci_adv_show(void);

while(1) {
    bt_hci_log_hci_data_show();
    bt_hci_log_hci_adv_show();
    vTaskDelay(10 / portTICK_PERIOD_MS);
}
```
This code continuously displays HCI data and advertising information, which can help you debug the issue further.

yanmartins, 2024-10-11:
@esp-zhp I added logs to see if handle values are correct. And they are. I also add the HCI debug log, and I think everything looks ok.

I do a test: I created a python code to execute in windows to see if my flow is rigth. And it succeed.

```python
import asyncio
from bleak import BleakClient, BleakScanner

# Endereço MAC do dispositivo Bluetooth
device_address = "00:21:7E:70:FA:E3"  # Substitua pelo endereço do seu dispositivo

# Características UUIDs
UART_TX_UUID = "00000002-0000-1000-8000-008025000000"
UART_RX_UUID = "00000001-0000-1000-8000-008025000000"
CREDITS_TX_UUID = "00000004-0000-1000-8000-008025000000"
CREDITS_RX_UUID = "00000003-0000-1000-8000-008025000000"

async def main():

    async with BleakClient(device_address, timeout=60.0) as client:
        # Inscrever nas características Indicate e Notify
        print("Conectado")
        await client.start_notify(UART_TX_UUID, notification_handler)
        await client.start_notify(CREDITS_TX_UUID, indication_handler)

        # Enviar no Write o tamanho dos UART credits
        uart_credits_size = bytearray([0x50])  # Exemplo de tamanho dos UART credits
        await client.write_gatt_char(CREDITS_RX_UUID, uart_credits_size, response=True)

        # Enviar no Write no response os comandos
        commands = bytearray([0x48, 0x65, 0x6C, 0x6C, 0x6F, 0x70, 0x65, 0x72, 0x6D, 0x61, 0x30, 0x45, 0x30, 0x30, 0x30, 0x30, 0x30, 0x30])  # Exemplo de comandos
        await client.write_gatt_char(UART_RX_UUID, commands, response=False)

        print("Aguardando notificações e indicações...")
        await asyncio.sleep(30)  # Aguarda por 30 segundos para receber notificações

def notification_handler(sender, data):
    print(f"Notification from {sender}: {data}")

def indication_handler(sender, data):
    print(f"Indication from {sender}: {data}")

asyncio.run(main())
```

Could ESP have any limitations when dealing with so many features?

yanmartins, 2024-10-11:
@esp-zhp This is the HCI log:

```
W (16814) GATTC_DEMO: connect to the remote device.
I (16824) GATTC_DEMO: stop scan successfully
c3 C:0a fd 02 0a 00
c4 C:0c 20 02 00 01
c5 E:0e 04 05 0c 20 00
c6 C:0d 20 19 60 00 30 00 00 00 e3 fa 70 7e 21 00 00 0a 00 14 00 00 00 58 02 05 00 05 00 
c7 E:0f 04 00 05 0d 20
d3 27 02 01 00 00 e3 fa 70 7e 21 00 1b 02 01 06 06 09 70 65 72 6d 61 02 0a 04 0d ff 00 21 7e 70 fa e3 e1 00 18 c6 2e 01 d9
d4 2b 02 01 03 01 85 7f 20 7f e2 0f 1f 1e ff 06 00 01 09 20 02 a3 29 98 8e 8d 8a 21 94 5a 5e a2 72 80 a4 07 5c fe 2d f7 13 a5 f9 fe c9 
d5 21 02 01 00 00 58 02 e9 9b bd c3 15 02 01 06 11 07 00 00 45 53 49 57 50 4c 41 00 00 00 01 53 00 00 bd 
d6 21 02 01 00 00 19 92 69 2c 6d d3 15 02 01 06 11 07 00 00 45 53 49 57 50 4c 41 00 00 00 01 53 00 00 a3
d7 2b 02 01 03 01 de 96 b8 db 3c 15 1f 1e ff 06 00 01 09 20 02 05 b6 2b 9f db 5c 2b 95 79 63 23 20 9f 02 86 3b 14 19 66 79 74 0f 03 b2 
d8 2b 02 01 03 01 c1 35 91 e0 69 31 1f 1e ff 06 00 01 09 20 02 39 72 d4 2e 90 7a e1 d9 ee 67 26 91 3d ba 8e a3 16 6f 34 2b d6 34 0e d3 
d9 2b 02 01 03 01 56 d0 5d c6 4d 2c 1f 1e ff 06 00 01 09 20 22 92 c9 14 b2 2e f2 38 07 0c 98 65 77 81 31 7f 08 02 1a 11 f9 32 55 b7 be
da 1e 02 01 00 00 72 67 dd b3 5b 4c 12 02 01 06 0e 09 44 79 4c 30 2e 30 33 2e 41 32 31 38 31 ae 
db 1d 02 01 03 01 a8 9e 5a 6d 73 6f 11 02 01 1a 0d ff 4c 00 09 08 13 a4 0a 1b 00 5c 1b 58 ae
c8 E:3e 1f 0a 00 01 00 00 00 e3 fa 70 7e 21 00 00 00 00 00 00 00 00 00 00 00 00 00 14 00 00 00 58 02 00 
c9 E:3e 04 14 01 00 01 
ca C:16 20 02 01 00
cb E:0f 04 00 05 16 20
cc E:3e 0c 04 00 01 00 25 41 00 00 00 00 00 00 
cd C:1d 04 02 01 00
ce E:0f 04 00 05 1d 04
W (18774) GATTC_DEMO: EVENT: 40
I (18774) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (18774) GATTC_DEMO: REMOTE BDA:
I (18774) GATTC_DEMO: 00 21 7e 70 fa e3 
W (18784) GATTC_DEMO: EVENT: 2
I (18784) GATTC_DEMO: open success
cf D:01 20 07 00 03 00 04 00 02 9e 00 
d0 C:35 0c 05 01 01 00 01 00
d1 H:01 00 07 00 03 00 04 00 03 9e 00
d2 H:01 00 0b 00 07 00 04 00 10 01 00 ff ff 00 28
d3 E:0c 08 00 01 00 09 59 00 a5 00 
d4 E:13 05 01 01 00 01 00 
d5 C:22 20 06 01 00 fb 00 90 42
d6 E:0e 06 05 22 20 00 01 00
d7 E:13 05 01 01 00 01 00 
I (18864) GATTC_DEMO: packet length updated: rx = 162, tx = 162, status = 0
d8 E:3e 0b 07 01 00 a2 00 80 05 a2 00 80 05
d9 D:01 20 1b 00 1a 00 04 00 11 06 01 00 09 00 00 18 0a 00 0d 00 01 18 0e 00 10 00 0a 18 11 00 ff 
da C:35 0c 05 01 01 00 01 00 
db D:01 10 03 00 ff fb fe 
dc C:35 0c 05 01 01 00 01 00 
dd H:01 00 0b 00 07 00 04 00 08 01 00 09 00 02 28
de E:13 05 01 01 00 01 00
df D:01 20 09 00 05 00 04 00 01 08 01 00 0a 
e0 C:35 0c 05 01 01 00 01 00
e1 H:01 00 0b 00 07 00 04 00 08 01 00 09 00 03 28
e2 E:13 05 01 01 00 01 00 
e3 D:01 20 22 00 1e 00 04 00 09 07 02 00 02 03 00 00 2a 04 00 02 05 00 01 2a 06 00 02 07 00 04 2a 08 00 02 09 00 a6 2a 
e4 C:35 0c 05 01 01 00 01 00 
e5 H:01 00 0b 00 07 00 04 00 08 09 00 09 00 03 28
e6 E:13 05 01 01 00 01 00 
e7 D:01 20 09 00 05 00 04 00 01 08 09 00 0a 
e8 C:35 0c 05 01 01 00 01 00
e9 H:01 00 0b 00 07 00 04 00 08 0a 00 0d 00 02 28 
ea E:13 05 01 01 00 01 00
eb D:01 20 09 00 05 00 04 00 01 08 0a 00 0a 
ec C:35 0c 05 01 01 00 01 00 
ed H:01 00 0b 00 07 00 04 00 08 0a 00 0d 00 03 28
ee E:13 05 01 01 00 01 00
ef D:01 20 0d 00 09 00 04 00 09 07 0b 00 20 0c 00 05 2a 
f0 C:35 0c 05 01 01 00 01 00 
f1 H:01 00 0b 00 07 00 04 00 08 0c 00 0d 00 03 28
f2 E:3e 06 0c 00 01 00 02 02
f3 E:13 05 01 01 00 01 00 
[... 88 lines cut ...]
I (19994) GATTC_DEMO: write descr success 
2c D:01 20 05 00 01 00 04 00 13
2d C:35 0c 05 01 01 00 01 00
2e H:01 00 09 00 05 00 04 00 12 20 00 01 00 
2f E:13 05 01 01 00 01 00 
W (20044) GATTC_DEMO: EVENT: 9
I (20044) GATTC_DEMO: write descr success 
W (20044) ***: WRITE
E (20044) !!!: gl_profile_tab[PROFILE_A_APP_ID].char_handle_write: 29
W (20054) ***: WRITE NR
E (20054) !!!: gl_profile_tab[PROFILE_A_APP_ID].char_handle_write_nr: 24
E (20064) ---: msg_len: 18
30 D:01 20 05 00 01 00 04 00 13
31 C:35 0c 05 01 01 00 01 00 
32 H:01 00 08 00 04 00 04 00 12 1d 00 50
33 E:13 05 01 01 00 01 00
W (20094) GATTC_DEMO: EVENT: 4
I (20094) GATTC_DEMO: write char success 
W (20094) GATTC_DEMO: EVENT: 4
I (20094) GATTC_DEMO: write char success 
34 D:01 20 05 00 01 00 04 00 13 
35 C:35 0c 05 01 01 00 01 00
36 H:01 00 19 00 15 00 04 00 52 18 00 48 65 6c 6c 6f 70 65 72 6d 61 30 45 30 30 30 30 30 30
37 E:13 05 01 01 00 01 00
```

esp-xzk, 2024-10-12:
Your HCI log seems to indicate that we have dispatched the last write command to the Controller, and to the correct attribute handle. 
If you have the ability to sniff and capture mid-air packets, could you please kindly do so to check if the packet has been successfully sent from  our device? Or, if it is in the correct format.
Otherwise, I would suggest switching the target device to a device that you can control and monitor if it has successfully received the write command and in the correct format.

yanmartins, 2024-10-14:
@esp-xzk @esp-zhp Unfortunately I don't have a sniffer. Could you tell me if it might be necessary to add some delays between sending the commands? I've tested this before, but I don't know if it would be necessary to invest time in it again.

yanmartins, 2024-10-14:
I ran the python script and captured the packets using wireshark. Here is the result:
[ble_esp.zip](https://github.com/user-attachments/files/17365085/ble_esp.zip)

I used the filter `"bluetooth.addr == 00:21:7E:70:FA:E3"`

## Differences:
I noticed the following differences in the writing characteristics packages

Python implementation:
`02 01 0e 08 00 04 00 04 00 12 1d 00 50`
`02 01 0e 19 00 15 00 04 00 52 18 00 48 65 6c 6c 6f 70 65 72 6d 61 30 45 30 30 30 30 30 30`

ESP implementation:
`01 00 08 00 04 00 04 00 12 1d 00 50`
`01 00 19 00 15 00 04 00 52 18 00 48 65 6c 6c 6f 70 65 72 6d 61 30 45 30 30 30 30 30 30`

yanmartins, 2024-10-14:
I also checked the bluetooth log of my android phone to see the behavior of the app.

- `02 40 00 19 00 15 00 04 00 52 18 00 48 65 6C 6C 6F 70 65 72 6D 61 30 45 30 30`

In both scenarios that worked, the command 0x02 was used. Now I'm trying to get ESP to use this same command. Any tips?

esp-xzk, 2024-10-15:
Hello,
The capture is very helpful. Thank you.
First of all, I am not quite sure what your `wait_response()` function is waiting on, for write without response packets, it is expected that there won't be any response for you to listen to on your host. 

The "missing" 0x02 at the beginning of each ESP implementation is that the debug code did not print out the type of the packet. In this case, 0x02 simply stands for 'ACL_Data' . In our HCI log, it is substituted with `H:`.
[image]

I believe the rest of the packet seems identical. 

Furthermore, if you could kindly check on the last line of the previous HCI log:
```
37 E:13 05 01 01 00 01 00
```
This is a message that the Controller tells the Host that the number of completed packets is 1. This means that the Controller has successfully delivered the packet to the receiver's Controller. 

My suggestion remains that you may try switching the target device to a device that you can control and monitor if it has successfully received the write command and in the correct format.

yanmartins, 2024-10-15:
Hello, @esp-xzk 

> First of all, I am not quite sure what your `wait_response()` function is waiting on, for write without response packets, it is expected that there won't be any response for you to listen to on your host.

I created `wait_response` and `notify_calling_task` to wait for one event to complete before requesting another.

```c
static void wait_response(void)
{
    task_to_notify_mesh = xTaskGetCurrentTaskHandle();

    xTaskNotifyWait(0, 0, 0, pdMS_TO_TICKS(30000));
    vTaskDelay(1000 / portTICK_PERIOD_MS);
}

static void notify_calling_task(void)
{
    if(task_to_notify_mesh != NULL) xTaskNotify(task_to_notify_mesh, 0, eSetValueWithOverwrite);

    task_to_notify_mesh = NULL;
}
```
I put it in some events:

[image]

Now I'm sure the problem wasn't competition xD

```
W (17759) **********: REGISTER FOR NOTIFY 1
W (17759) GATTC_DEMO: EVENT: 38
I (17759) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
E (17759) ///: p_data->reg_for_notify.handle: 26
E (17759) ///: descr_elem_result[0].handle: 27
e7 H:01 00 09 00 05 00 04 00 12 1b 00 01 00
W (17799) GATTC_DEMO: EVENT: 9
I (17799) GATTC_DEMO: write descr success 
e8 E:13 05 01 01 00 01 00 
e9 D:01 20 05 00 01 00 04 00 13
ea C:35 0c 05 01 01 00 01 00
W (18799) **********: REGISTER FOR NOTIFY 2
W (18799) GATTC_DEMO: EVENT: 38
I (18799) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
E (18799) ///: p_data->reg_for_notify.handle: 31
E (18799) ///: descr_elem_result[0].handle: 32
eb H:01 00 09 00 05 00 04 00 12 20 00 01 00
ec E:13 05 01 01 00 01 00 
W (18849) GATTC_DEMO: EVENT: 9
I (18849) GATTC_DEMO: write descr success 
ed D:01 20 05 00 01 00 04 00 13 
ee C:35 0c 05 01 01 00 01 00
W (19849) **********: WRITE UART CREDITS
W (19849) ***: WRITE
E (19849) !!!: gl_profile_tab[PROFILE_A_APP_ID].char_handle_write: 29
ef H:01 00 08 00 04 00 04 00 12 1d 00 50
f0 E:13 05 01 01 00 01 00 
W (19899) GATTC_DEMO: EVENT: 4
I (19899) GATTC_DEMO: write char success 
f1 D:01 20 05 00 01 00 04 00 13 
f2 C:35 0c 05 01 01 00 01 00
W (20899) **********: WRITE COMMAND
W (20899) ***: WRITE NR
E (20899) !!!: gl_profile_tab[PROFILE_A_APP_ID].char_handle_write_nr: 24
E (20899) ---: msg_len: 18
W (20899) GATTC_DEMO: EVENT: 4
I (20909) GATTC_DEMO: write char success 
f3 H:01 00 19 00 15 00 04 00 52 18 00 48 65 6c 6c 6f 70 65 72 6d 61 30 45 30 30 30 30 30 30
f4 E:13 05 01 01 00 01 00 
```

> My suggestion remains that you may try switching the target device to a device that you can control and monitor if it has successfully received the write command and in the correct format.

I'll see if I can get another device to perform the tests.

> This is a message that the Controller tells the Host that the number of completed packets is 1. This means that the Controller has successfully delivered the packet to the receiver's Controller.

About this: If the Notification and Indication characteristics are not subscribed, the Write command will not work. Based on this, does this same confirmation message guarantee me that subscriptions to these characteristics have also been made?

Thanks for your help

yanmartins, 2024-10-15:
Is there a possibility that when I sign up for Notify and then sign up for Indicate, this will overwrite the data on the device and cancel Notify and vice versa?

yanmartins, 2024-10-15:
Analyzing the logs better, I noticed this:

## Windows python application log
- `02 01 0e 09 00 05 00 04 00 12 1b 00 01 00`
- `02 01 0e 09 00 05 00 04 00 12 20 00 02 00`

[image]

## ESP application log
- `01 00 09 00 05 00 04 00 12 1b 00 01 00`
- `01 00 09 00 05 00 04 00 12 20 00 01 00`

the penultimate byte is different. It is not setting the indication

yanmartins, 2024-10-15:
Solved!

```diff
case ESP_GATTC_REG_FOR_NOTIFY_EVT: {
                    ...
                    /* Every char has only one descriptor in our 'ESP_GATTS_DEMO' demo, so we used first 'descr_elem_result' */
                    ESP_LOGE("///", "descr_elem_result[0].handle: %d", descr_elem_result[0].handle);
+                    if (count > 0 && descr_elem_result[0].uuid.len == ESP_UUID_LEN_16 && descr_elem_result[0].uuid.uuid.uuid16 == ESP_GATT_UUID_CHAR_CLIENT_CONFIG && descr_elem_result[0].handle == 27){
                        ESP_LOGW("---------", "WRITE NOTIFY");
                        ret_status = esp_ble_gattc_write_char_descr( gattc_if,
                                                                     gl_profile_tab[PROFILE_A_APP_ID].conn_id,
                                                                     descr_elem_result[0].handle,
                                                                     sizeof(notify_en),
                                                                     (uint8_t *)&notify_en,
                                                                     ESP_GATT_WRITE_TYPE_RSP,
                                                                     ESP_GATT_AUTH_REQ_NONE);
                    }
+                    else {
+                        ESP_LOGW("---------", "WRITE INDICATE");
+                        uint16_t indicate_en = 2;
+                        ret_status = esp_ble_gattc_write_char_descr( gattc_if,
+                                                                     gl_profile_tab[PROFILE_A_APP_ID].conn_id,
+                                                                     descr_elem_result[0].handle,
+                                                                     sizeof(indicate_en),
+                                                                     (uint8_t *)&indicate_en,
+                                                                     ESP_GATT_WRITE_TYPE_RSP,
+                                                                     ESP_GATT_AUTH_REQ_NONE);
        
                    }

                    if (ret_status != ESP_GATT_OK){
                        ESP_LOGE(GATTC_TAG, "esp_ble_gattc_write_char_descr error");
                    }

                    /* free descr_elem_result */
                    free(descr_elem_result);
                }
            }
            else{
                ESP_LOGE(GATTC_TAG, "decsr not found");
            }
            // notify_calling_task();

        }
        break;
```

My big mistake was not writing correctly in the indication

Thank you very much for the support @esp-xzk @esp-zhp

yanmartins, 2024-10-16:
@esp-xzk @esp-zhp I started optimizing the SDK for this example, and when I disabled the option `CONFIG_BT_GATTS_ENABLE` the example stopped working.

Is this correct? The example is for GATTC should not have dependencies with GATTS or yes?

esp-zhp, 2024-10-16:
@yanmartins 
GATTC should not have dependencies with GATTS.
please create new issue to trace your problem.thanks.and share me your log.

closed 2024-10-16 by yanmartins