espressif/esp-idf #14673: esp_http_client fails due to base_poll_read() select() usage

gonzzor, 2024-10-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Reliably download files with esp_http_client() even when the last IP packets arrive quickly.

### What is the actual behavior?

esp_http_client_read() return an error instead of the last data.

### Steps to reproduce.

1. Use a slow network uplink. I'm using PPP over an uart. 
2. Try to use esp_http_client_open() and esp_http_client_read() to download files.
3. Sometimes, depending on the file size and buffer size to esp_http_client_read() the call fails.

### Debug Logs.

```plain
D (15:15:01.618) HTTP_CLIENT: Begin connect to: http://10.0.0.1:41351
D (15:15:01.633) HTTP_CLIENT: Write header[3]: GET /foo.bin HTTP/1.1
User-Agent: ESP32 HTTP Client/1.0
Host: 10.0.0.1:41351
Content-Length: 0

D (15:15:01.735) HTTP_CLIENT: on_message_begin
D (15:15:01.736) HTTP_CLIENT: HEADER=Server:SimpleHTTP/0.6 Python/3.12.6
D (15:15:01.737) HTTP_CLIENT: HEADER=Date:Fri, 27 Sep 2024 15:15:01 GMT
D (15:15:01.737) HTTP_CLIENT: HEADER=Content-type:application/octet-stream
D (15:15:01.738) HTTP_CLIENT: HEADER=Content-Length:129792
D (15:15:01.739) HTTP_CLIENT: HEADER=Last-Modified:Fri, 27 Sep 2024 14:56:50 GMT
D (15:15:01.739) HTTP_CLIENT: http_on_headers_complete, status=200, offset=204, nread=204
D (15:15:01.740) HTTP_CLIENT: http_on_body 308
D (15:15:01.740) HTTP_CLIENT: Body received in fetch header state, 0x3fcebb9c, 308
D (15:15:01.741) HTTP_CLIENT: content_length = 129792
D (15:15:01.742) HTTP_CLIENT: Data processed 308 != Data specified in content length 129792
D (15:15:01.742) HTTP_CLIENT: Data processed 308 != Data specified in content length 129792
D (15:15:01.743) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
D (15:15:01.744) HTTP_CLIENT: need_read=76, byte_to_read=76, rlen=76, ridx=52
D (15:15:01.745) HTTP_CLIENT: http_on_body 76
D (15:15:01.894) HTTP_CLIENT: Data processed 384 != Data specified in content length 129792
D (15:15:01.895) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
D (15:15:01.896) HTTP_CLIENT: need_read=128, byte_to_read=128, rlen=128, ridx=0
D (15:15:01.897) HTTP_CLIENT: http_on_body 128
D (15:15:01.898) HTTP_CLIENT: Data processed 512 != Data specified in content length 129792
D (15:15:01.898) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
D (15:15:01.900) HTTP_CLIENT: need_read=128, byte_to_read=128, rlen=128, ridx=0
D (15:15:01.900) HTTP_CLIENT: http_on_body 128
D (15:15:01.904) HTTP_CLIENT: Data processed 640 != Data specified in content length 129792
D (15:15:01.905) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
D (15:15:01.906) HTTP_CLIENT: need_read=128, byte_to_read=128, rlen=128, ridx=0
D (15:15:01.907) HTTP_CLIENT: http_on_body 128

.... snip ...

D (15:15:20.333) HTTP_CLIENT: Data processed 124416 != Data specified in content length 129792
D (15:15:20.333) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
D (15:15:20.335) HTTP_CLIENT: need_read=128, byte_to_read=128, rlen=128, ridx=0
D (15:15:20.335) HTTP_CLIENT: http_on_body 128
D (15:15:20.340) HTTP_CLIENT: Data processed 124544 != Data specified in content length 129792
D (15:15:20.341) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
D (15:15:20.342) HTTP_CLIENT: need_read=128, byte_to_read=128, rlen=128, ridx=0
D (15:15:20.343) HTTP_CLIENT: http_on_body 128
D (15:15:20.344) HTTP_CLIENT: Data processed 124672 != Data specified in content length 129792
D (15:15:20.344) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
D (15:15:20.346) HTTP_CLIENT: need_read=128, byte_to_read=128, rlen=128, ridx=0
D (15:15:20.346) HTTP_CLIENT: http_on_body 128
D (15:15:20.446) HTTP_CLIENT: Data processed 124800 != Data specified in content length 129792
D (15:15:20.447) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=129792
E (15:15:20.449) transport_base: poll_read select error 128, errno = Socket is not connected, fd = 57
D (15:15:20.450) HTTP_CLIENT: need_read=128, byte_to_read=128, rlen=-2, ridx=0
E (15:15:20.451) HTTP_CLIENT: transport_read: error - 57347 | ERROR
D (15:15:20.451) HTTP_CLIENT: Data processed 124800 != Data specified in content length 129792
D (15:15:20.452) HTTP_CLIENT: Data processed 124800 != Data specified in content length 129792
```

### More Information.

The select() call might return 2. That means two bits are set in total
in the sets used. As there is only one fd it means that the fd both has
data and an error.

This happens when the last data is received and the peer has notified
that it won't send any more data.

If this happens one should use the information in the readset first
before handling any exceptions.

This caused the http_client to fail at the last packet depending on the
timing.

Change this so that it only considers the errorset if it's only one bit
set in the returned sets.

```patch
--- a/components/tcp_transport/transport_ssl.c
+++ b/components/tcp_transport/transport_ssl.c
@@ -169,7 +169,7 @@ static int base_poll_read(esp_transport_handle_t t, int timeout_ms)
         return remain;
     }
     ret = select(ssl->sockfd + 1, &readset, NULL, &errset, esp_transport_utils_ms_to_timeval(timeout_ms, &timeout));
-    if (ret > 0 && FD_ISSET(ssl->sockfd, &errset)) {
+    if (ret == 1 && FD_ISSET(ssl->sockfd, &errset)) {
         int sock_errno = 0;
         uint32_t optlen = sizeof(sock_errno);
         getsockopt(ssl->sockfd, SOL_SOCKET, SO_ERROR, &sock_errno, &optlen);
```

AxelLin, 2024-10-13:
@david-cermak
Any chance to take a look? Thanks.

euripedesrocha, 2024-10-16:
@AxelLin the fix is already under review. Should be on master soon.

AxelLin, 2024-11-05:
> @AxelLin the fix is already under review. Should be on master soon.

@euripedesrocha 
So you confirm the bug is real and you have a fix for it?
Could you share the patch and test code?
(The fix is not yet in master, it's so slow)

nileshkale123, 2024-11-05:
Hello @AxelLin ,

We've already submitted a merge request for this fix, and we're currently testing it across additional scenarios to ensure stability. It should be merged soon.

AxelLin, 2024-11-14:
@nileshkale123 @euripedesrocha 

I'm testing on v5.2 branch.
I just realize this is a big issue because now my device cannot update
firmware via PPP due to this issue.
Can you fix this ASAP?

AxelLin, 2024-11-25:
> Hello @AxelLin ,
> 
> We've already submitted a merge request for this fix, and we're currently testing it across additional scenarios to ensure stability. It should be merged soon.

Your reply gives people the expect that this will be fix **soon**.
However, having wait for 3 weeks, still cannot get the fix.

AxelLin, 2024-11-26:
@nileshkale123  This issue was reported by @gonzzor, not me.
@gonzzor Could you verify if the patch from https://github.com/espressif/esp-idf/issues/14673#issuecomment-2499797262 works?

gonzzor, 2024-11-26:
@AxelLin I'm using my suggested change in transport_ssl.c in production and it for sure improved things, haven't seen any more problems with it.

The second part mentioned in https://github.com/espressif/esp-idf/issues/14673#issuecomment-2499797262 seem to have been merged already in https://github.com/espressif/esp-idf/issues/14724 so I don't see any problems with that change.

In addition, when testing this I'm using the python http.server module to simply serve files locally to my board. I didn't have problems all the time and it seem to be timing related as the size of file impacted things. I could see the issue when connected via uart and pppd running on Linux.

AxelLin, 2024-12-02:
@nileshkale123 
So you said "already submitted a merge request for this fix, and we're currently testing it" on Nov 5 https://github.com/espressif/esp-idf/issues/14673#issuecomment-2456330842.
But in the follow up you said "I have not been able to reproduce the issue" on Nov 26 https://github.com/espressif/esp-idf/issues/14673#issuecomment-2499797262.
And if look back the ealier response on Oct 16 https://github.com/espressif/esp-idf/issues/14673#issuecomment-2415839814, it does not make sense at all.
This is a very confusing situation.

nileshkale123, 2024-12-02:
Hello @AxelLin ,

Apologies for any confusion caused.

We have raised a merge request based on the user’s suggestion, as it seems logically correct and was agreed upon by the team. However, we suspect that this change might introduce another scenario that could potentially fail. To address this, we’ve been actively investigating further.

Additionally, the specific issue reported by the user is not being directly reproduced on our end. We are exploring more robust methods to accurately replicate the scenario. We kindly request some additional time to thoroughly analyze and resolve this. Thank you for your patience.

closed 2024-12-05 by espressif-bot

AxelLin, 2024-12-19:
@nileshkale123 

I thoutht you will add a test case for this, but the commit does not include a test case.
The fix is committed in master now, so could you confirm you do reproduce this issue?

BTW, the stable branches still need fix.

AxelLin, 2025-01-13:
@nileshkale123 @david-cermak 
The fix is not yet available in any release branches.
IMHO, a tcp_transport fix should be applied to release branches ASAP.

nileshkale123, 2025-01-13:
Hello @AxelLin .

We have tested the changes in all the scenarios discussed and have incorporated them accordingly. However, the changes have not yet been merged into the release branches due to some internal planning reasons. These changes will be reflected in the other release branches very soon.

AxelLin, 2025-01-28:
> Hello [@AxelLin](https://github.com/AxelLin) .
> 
> These changes will be reflected in the other release branches very soon.

**"very soon"**? This is not true. too bad.

AxelLin, 2025-01-28:
> ### Steps to reproduce.
> 1. Use a slow network uplink. I'm using PPP over an uart.
> 2. Try to use esp_http_client_open() and esp_http_client_read() to download files.
> 3. Sometimes, depending on the file size and buffer size to esp_http_client_read() the call fails.

This is exactly my use case.
And somehow, the fix is not available in release branches. Why?

@david-cermak

nileshkale123, 2025-02-14:
Hello @AxelLin,
We have initiated the backports, and they will be available in the release branches soon. The delay was due to internal reasons. Apologies for the inconvenience caused.

Thanks.

AxelLin, 2025-02-25:
@nileshkale123 The v5.2.4 does not include this fix.

AxelLin, 2025-03-06:
@nileshkale123

AxelLin, 2025-03-06:
@nileshkale123 

It has been "**3 Months**" since the fix was available on master (Dec 6, 2024), but the fix is **not yet available in 5.2 branch**.
How much time is required to get a backport fix? I don't think this is a normal situation.

mmackh, 2025-08-06:
Is this issue fixed in 5.4? 

```
E (129706) transport_base: poll_read select error 128, errno = Socket is not connected, fd = 36
E (129707) HTTP_CLIENT: transport_read: error - 57347 | ERROR
E (129707) esp_https_ota: data read -1, errno 0
[OTA]: download error.
```

AxelLin, 2025-08-06:
> Is this issue fixed in 5.4?

v5.4 branch fix (v5.4-495-g5ad73e97e8) : 5ad73e97e8fa6ec209b601d465a8126b36ede34c

> 
> ```
> E (129706) transport_base: poll_read select error 128, errno = Socket is not connected, fd = 36
> E (129707) HTTP_CLIENT: transport_read: error - 57347 | ERROR
> E (129707) esp_https_ota: data read -1, errno 0
> [OTA]: download error.
> ```

What is your 'git describe --tags' output?