espressif/esp-idf #14675: official esp_https_server dont work

sova32, 2024-10-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Waveshare esp32-S3-Pico

### Power Supply used.

USB

### What is the expected behavior?

worked example !!!
https://github.com/espressif/esp-idf/tree/v5.3.1/examples/protocols/https_server/simple

### What is the actual behavior?

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1874
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ea4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dev-3201-g46acfdce96 2nd stage bootloader
I (27) boot: compile time Sep 29 2024 19:09:38
I (27) boot: Multicore bootloader
I (32) boot: chip revision: v0.2
I (35) boot: efuse block revision: v1.3
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 16MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (93) boot:  3 factory          factory app      00 00 00010000 00200000
I (100) boot:  4 ota_0            OTA app          00 10 00210000 00200000
I (108) boot:  5 ota_1            OTA app          00 10 00410000 00200000
I (115) boot:  6 nvs_key          NVS keys         01 04 00610000 00001000
I (123) boot:  7 littleFS_ROOT    Unknown data     01 82 00611000 00100000
[... 110 lines cut ...]
0x4200b1cd: app_main at /run/media/vadiki/250_GB_ST_1GB/projects/filamentor_v2/program/filamentor_v2/main/main.c:231
0x42098a0f: main_task at /home/vadiki/esp/esp-idf/components/freertos/app_startup.c:208
0x4037d029: vPortTaskWrapper at /home/vadiki/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 66d4994f4

Rebooting...
���ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b18
--- 0x40375b18: esp_restart_noos at /home/vadiki/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:160

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x1874
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ea4
entry 0x403c8928

### Steps to reproduce.

1. clone your official example
2. bulid and flash

nileshkale123, 2024-10-07:
Hello @sova32,

I attempted to reproduce the issue but did not encounter any errors during execution.

I recommend reviewing the relevant README file and ensuring that the Wi-Fi credentials match the ones configured on the system you're using to connect to the server.

Please feel free to reach out if the issue persists or if you need further assistance.

nileshkale123, 2024-10-15:
Hello @sova32 

Since we couldn't reproduce this issue on our end, I'm going to go ahead and close it.  If anything changes or new details pop up, feel free to reopen it!

closed 2024-10-15 by nileshkale123

freddy916, 2025-05-13:
I faced the same issue with the same error code. I faced this when my router went offline and ESP32 tried to reconnect 6 times. When it reconnects fails, it keeps restarting, and the same happens until it connects successfully.