espressif/esp-idf #14677: operating hours counter

diplfranzhoepfinger, 2024-10-06:
### Is your feature request related to a problem?

operating hours counter

KaeLL, 2024-10-06:
By operating hours you mean uptime?

diplfranzhoepfinger, 2024-10-07:
Yes. Uptime. in hours. With a say 1 minute precision.

SoucheSouche, 2024-10-07:
Hi @diplfranzhoepfinger,

You can use the `esp_timer_get_time` (https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32/api-reference/system/esp_timer.html#obtaining-current-time) returning the uptime in microseconds.

Could you be more specific regarding your "how to handle unexpected shutdowns" question?

kriegste, 2024-10-07:
I think he means total running time, i.e. cumulated minutes the device was powered up. You would have to store this into NVS. But at one minute intervals, the flash would wear out maybe too soon.
You could:
- increase the NVS partition to help wear leveling
- use bigger intervals like 5 minutes

SoucheSouche, 2024-10-08:
@diplfranzhoepfinger, 
- using NVS to periodically store the uptime is the obvious approach but as @kriegste mentioned, using a 1 minute precision will wear out the flash.

Alternatively, you could also:
- Create a variable storing the uptime with the attribute `RTC_NOINIT_ATTR` (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/memory-types.html#rtc-slow-memory). It should retain the value unless an RTC related panic induces the crash or the board is powered off completely.
- You can wrap the panic handler function and store the uptime in the NVS in the wrapper function before calling the panic handler function (https://github.com/espressif/esp-idf/issues/7681#issuecomment-941957783). This way, you minimize the interaction with the NVS.

KaeLL, 2024-10-08:
@SoucheSouche wouldn't the last option conflict with what igrr said last in that thread?

kriegste, 2024-10-08:
If you can devote a whole flash sector to this feature, you could flip a new bit (1->0) each minute. At minute 0 all bits are 1. At minute 32768 (on day 22) all bits are 0. Only then you would have to erase the sector and increase an overflow counter somewhere else (NVS).

diplfranzhoepfinger, 2024-10-09:
good ideas Folks. i will check,

SoucheSouche, 2024-10-09:
> @SoucheSouche wouldn't the last option conflict with what igrr said last in that thread?

I think it should be possible since there is already an option for the core dump to be stored in flash.

@diplfranzhoepfinger, I will close the ticket now since you got as few leads on how to proceed with your uptime counter. Feel free to reopen if you need more support.

closed 2024-10-09 by SoucheSouche

chipweinberger, 2024-10-14:
you should use battery backed RAM, like the DS1302

you can have very high precision.

f-hoepfinger-hr-agrartechnik, 2025-05-01:
> you should use battery backed RAM, like the DS1302
> 
> you can have very high precision.

you are right, but the Hardware is already done.

f-hoepfinger-hr-agrartechnik, 2025-05-01:
> If you can devote a whole flash sector to this feature, you could flip a new bit (1->0) each minute. At minute 0 all bits are 1. At minute 32768 (on day 22) all bits are 0. Only then you would have to erase the sector and increase an overflow counter somewhere else (NVS).

this sounds very reasonable.

f-hoepfinger-hr-agrartechnik, 2025-05-01:
> If you can devote a whole flash sector to this feature, you could flip a new bit (1->0) each minute. At minute 0 all bits are 1. At minute 32768 (on day 22) all bits are 0. Only then you would have to erase the sector and increase an overflow counter somewhere else (NVS).

https://docs.google.com/document/d/1k7Hugy9E87zSfl7LuwlkfQDMVaaveGYZNEK8msIz5v0/edit?usp=sharing

will try.

diplfranzhoepfinger, 2025-05-01:
https://gist.github.com/diplfranzhoepfinger/f6a95179ae5cb632a2b6caa07b68d0cc

seems working. 
credits to @kriegste 

and to Gemini of course.