espressif/esp-idf #14681: Nimble host linking errors

Daphne-HU, 2024-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.3-416-gd23b7a0361

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

building my application where i can use nimble

### What is the actual behavior?

When building the application where i try to initialize nimBLE i get linking errors

### Steps to reproduce.

1. I'm working in a devcontainer under WSL 
2. setup a project, and configure the nimBLE host in the menuconfig
3. include "nimble/nimble_port.h"
4. init NVS
5. run nimble_port_init();

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running ninja in directory /workspaces/BCE-Monitor/build
Executing "ninja all"...
[1/14] Generating version_infoThe software was not tagged correctly (like 0.45.1), add tag to the GIT-repo
******* Test build: 0.0.0-feature.setupBluetoothLE+dirty.a57118c *******
[1/1] cd /workspaces/BCE-Monitor/build/bootloader/esp-idf/esptool_py && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/...components/partition_table/check_sizes.py --offset 0x10000 bootloader 0x0 /workspaces/BCE-Monitor/build/bootloader/bootloader.binBootloader binary size 0x5060 bytes. 0xafa0 bytes (69%) free.
[9/12] Linking CXX executable BCE_Monitor.elfFAILED: BCE_Monitor.elf 
: && /opt/esp/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imc_zicsr_zifencei -nostartfiles -march=rv32imc_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C3=0 -Wl,--Map=/workspaces/BCE-Monitor/build/BCE_Monitor.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32c3.peripherals.ld -T esp32c3.rom.ld -T esp32c3.rom.api.ld -T esp32c3.rom.libgcc.ld -T esp32c3.rom.newlib.ld -T esp32c3.rom.version.ld -T memory.ld -T sections.ld CMakeFiles/BCE_Monitor.elf.dir/project_elf_src_esp32c3.c.obj -o BCE_Monitor.elf -L/opt/esp/idf/components/soc/esp32c3/ld   -L/opt/esp/idf/components/esp_rom/esp32c3/ld   -L/workspaces/BCE-Monitor/build/esp-idf/esp_system/ld   -L/opt/esp/idf/components/esp_phy/lib/esp32c3   -L/opt/esp/idf/components/esp_coex/lib/esp32c3   -L/opt/esp/idf/components/esp_wifi/lib/esp32c3   -L/opt/esp/idf/components/bt/controller/lib_esp32c3_family/esp32c3 esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/bt/libbt.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/BlueCurrent/libBlueCurrent.a  esp-idf/DSMR/libDSMR.a  esp-idf/coreMQTT/libcoreMQTT.a  esp-idf/azure-sdk-for-c/libazure-sdk-for-c.a  esp-idf/coreHTTP/libcoreHTTP.a  esp-idf/azure-iot-middleware-freertos/libazure-iot-middleware-freertos.a  esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/unity/libunity.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/protocomm/libprotocomm.a  esp-idf/console/libconsole.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/BlueCurrent/libBlueCurrent.a  esp-idf/DSMR/libDSMR.a  esp-idf/ISL/isl-csdd-framework/libisl-csdd-framework.a  esp-idf/ISL/isl-assert/libisl-assert.a  esp-idf/ISL/isp-jsmn/libisp-jsmn.a  esp-idf/ISL/isp-spiffs/libisp-spiffs.a  esp-idf/ISL/isp-jwrite/libisp-jwrite.a  esp-idf/ISL/isl-operating-system-wrapper-freertos/libisl-operating-system-wrapper-freertos.a  esp-idf/azure-iot-middleware-freertos/libazure-iot-middleware-freertos.a  esp-idf/coreMQTT/libcoreMQTT.a  esp-idf/coreHTTP/libcoreHTTP.a  esp-idf/azure-sdk-for-c/libazure-sdk-for-c.a  esp-idf/esp-nimble-cpp/libesp-nimble-cpp.a  esp-idf/bt/libbt.a  esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  esp-idf/mbedtls/mbedtls/3rdparty/everest/libeverest.a  esp-idf/mbedtls/mbedtls/3rdparty/p256-m/libp256m.a  /opt/esp/idf/components/esp_coex/lib/esp32c3/libcoexist.a  /opt/esp/idf/components/esp_wifi/lib/esp32c3/libcore.a  /opt/esp/idf/components/esp_wifi/lib/esp32c3/libespnow.a  /opt/esp/idf/components/esp_wifi/lib/esp32c3/libmesh.a  /opt/esp/idf/components/esp_wifi/lib/esp32c3/libnet80211.a  /opt/esp/idf/components/esp_wifi/lib/esp32c3/libpp.a  /opt/esp/idf/components/esp_wifi/lib/esp32c3/libsmartconfig.a  /opt/esp/idf/components/esp_wifi/lib/esp32c3/libwapi.a  esp-idf/riscv/libriscv.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthre [... cut ...]
/opt/esp/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /opt/esp/idf/components/bt/controller/lib_esp32c3_family/esp32c3/libbtdm_app.a(ip_funcs.o):(.rodata.r_ip_funcs_ro+0x7c4): undefined reference to `r_lld_init_evt_end_type_check_state_get'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /workspaces/BCE-Monitor/build/log/idf_py_stderr_output_8461 and /workspaces/BCE-Monitor/build/log/idf_py_stdout_output_8461f
```

rahult-github, 2024-10-07:
Hi @Daphne-HU ,

These symbols are not from host but from controller.  Please check your compilation environment Cmakelists or Makefile ( or similar file in your setup) that links the needed library.  For reference, you can check IDF examples which link the contorller library [here](https://github.com/espressif/esp-idf/blob/master/components/bt/CMakeLists.txt#L850).

Daphne-HU, 2024-10-10:
Thank you for the response, after setting the target to esp32c3 again and re-configuring it worked

closed 2024-10-11 by rahult-github