espressif/esp-idf #14682: Multi hub and multi device not working

maaamcube, 2024-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello @roma-jam and everyone,
I have connected my board : USB vendor specific devices (see image below) to the Esp32 USB Host example from this repo : https://github.com/espressif/esp-idf/commit/70f222e5d29df8ffe5da25057601708c8097bcd1

[image]

Only the first hub is recognized and i got this:

E (5579) USBH: Dev 1 EP 0 STALL
E (5579) ENUM: Bad transfer status 4: CHECK_SHORT_LANGID_TABLE
E (5579) ENUM: [0:0] CHECK_SHORT_LANGID_TABLE FAILED
I (5579) CLASS: Opening device at address 1
I (5579) CLASS: Getting device information
I (5589) CLASS:         Full speed
I (5589) CLASS:         bConfigurationValue 1
I (5599) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x9
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x424
idProduct 0x2533
bcdDevice 1.80
iManufacturer 0
iProduct 0
iSerialNumber 0
bNumConfigurations 1
I (5619) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 25
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xe0
bMaxPower 2mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x9
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 1
                bInterval 255

Thanks.

roma-jam, 2024-10-07:
Hi @maaamcube ,

thanks for the feedback.

The commit you have referred to contains the adding of downstream ports logic only. 
Basically, these changes is crucial for external Hubs to work. 

The enabling Hubs support feature in the USB Host examples itself though has been merged internally and will be available in public soon.

roma-jam, 2024-10-15:
Hi @maaamcube, 

the enabling of Hubs support feature in the USB Host examples has been published. 

Now any external Hub will be initialized when attached. 

Feel free to try it out and do not hesitate to report issues in case there will be any.

closed 2024-10-15 by roma-jam

chegewara, 2024-10-15:
Actually there is some issue with HUB, but im not sure it is a real issue.
esp32 cant enumerate few devices if they are not the same speed, like fast and low speed, connected to the same hub.

roma-jam, 2024-10-15:
Hi @chegewara,

Thanks for the note. 

Yes, that is right. We are aware about that and such devices (when upstream port speed mismatches the downstream port device speed) should be just disabled without any influences on other devices, attached to the same hub. 
This feature was added as experimental one, as it is not stable yet.

We are working on making it stable.

chegewara, 2024-10-15:
I am not complaining at all. Actually it's very nice we have a hub feature implemented.
It's not that I am planning to use it in the near future, but I thought it is good to test it.

Thanks

PS I am more interested to see option to use both, HS and FS, ports on P4

roma-jam, 2024-10-15:
@chegewara 
Thanks.
No worries, feel free to report any problem you will find/run into. 

For HS and P4 it is a bit trickier, as to support the Full-speed devices, there should be implemented a layer, which is called Transaction Translator. 

ATM, the hub driver doesn‘t implement this feature as it is not supported in the current DMA mode (Scatter-Gather). 
So, to support the TT-logic, additional changes in the already implemented part of USB Host Library should be done (move to Buffer DMA).

But it is not even planned so far.

chegewara, 2024-10-15:
What I mean is that P4 has 2x USB controllers, or maybe I am wrong.
High speed and full speed, and JTAG.

I am wondering if we can or it will be possible to use both USB ports at the same time.

Thanks

maaamcube, 2024-10-15:
> Hi @maaamcube,
> 
> the enabling of Hubs support feature in the USB Host examples has been published.
> 
> Now any external Hub will be initialized when attached.
> 
> Feel free to try it out and do not hesitate to report issues in case there will be any.

Hi,

I connected the same board and i got this :

"E (20571) USBH: Dev 1 EP 0 STALL
E (20571) ENUM: Bad transfer status 4: CHECK_SHORT_LANGID_TABLE
E (20571) ENUM: [0:0] CHECK_SHORT_LANGID_TABLE FAILED
I (20571) CLASS: Opening device at address 1
I (20571) CLASS: Getting device information
I (20581) CLASS:        Full speed
I (20581) CLASS:        Parent info:
I (20581) CLASS:                Port: ROOT
I (20581) CLASS:        bConfigurationValue 1
I (20591) CLASS: Getting device descriptor
*** Device descriptor ***
bLength 18
bDescriptorType 1
bcdUSB 2.00
bDeviceClass 0x9
bDeviceSubClass 0x0
bDeviceProtocol 0x0
bMaxPacketSize0 64
idVendor 0x424
idProduct 0x2533
bcdDevice 1.80
iManufacturer 0
iProduct 0
iSerialNumber 0
bNumConfigurations 1
I (20611) CLASS: Getting config descriptor
*** Configuration descriptor ***
bLength 9
bDescriptorType 2
wTotalLength 25
bNumInterfaces 1
bConfigurationValue 1
iConfiguration 0
bmAttributes 0xe0
bMaxPower 2mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x9
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 1
                bInterval 255
I (20901) CLASS: Opening device at address 2
I (20901) CLASS: Getting device information
I (20901) CLASS:        Full speed
I (20901) CLASS:        Parent info:
I (20901) CLASS:                Bus addr: 1
I (20901) CLASS:                Port: 1
[... 221 lines cut ...]
bmAttributes 0xe0
bMaxPower 2mA
        *** Interface descriptor ***
        bLength 9
        bDescriptorType 4
        bInterfaceNumber 0
        bAlternateSetting 0
        bNumEndpoints 1
        bInterfaceClass 0x9
        bInterfaceSubClass 0x0
        bInterfaceProtocol 0x0
        iInterface 0
                *** Endpoint descriptor ***
                bLength 7
                bDescriptorType 5
                bEndpointAddress 0x81   EP 1 IN
                bmAttributes 0x3        INT
                wMaxPacketSize 1
                bInterval 255
E (22141) USBH: HCD Pipe alloc error: ESP_ERR_NOT_SUPPORTED
E (22141) HUB: Failed to add new downstream device
E (22141) EXT_PORT: [5:1] Port disabled, reset attempts=1
W (22141) HUB: Device tree node (parent_port=1): not found"

I see only two devices if i'm not wrong, but i am supposed to see three as i have three Stm32 USBs.

roma-jam, 2024-10-15:
Hi @maaamcube, 

the device address value represents the amount of devices. In the last opened device you have the address 5.

In the end you run out of the host channels, that is why the HCD pipe couldn't create any more pipes, so this is a hardware limitation.

maaamcube, 2024-10-15:
@roma-jam i got this at every device i don't know if it's ok : 
"E (7051) USBH: Dev 1 EP 0 STALL
E (7051) ENUM: Bad transfer status 4: CHECK_SHORT_LANGID_TABLE
E (7051) ENUM: [0:0] CHECK_SHORT_LANGID_TABLE FAILED"

What can i do to change and support the hole USBs ?

roma-jam, 2024-10-15:
@maaamcube ,

This is the error which shows that device STALLed the String Descriptor request. Which is a normal behavior for the device it doesn't have the requested descriptor and doesn't break the enumeration process. 

In general, each external Hub needs 2 Host channel to work (one for Control EP and one for Interrupt EP). The S2/3 has only 8 channels, so that means, with 3x Hubs is 6 channels occupied by the driver and every other device without claiming interface occupies 1 channel. 
So 2x Vendor specific devices in addition and you are run out of channels on S2/3.

maaamcube, 2024-10-18:
Hi @roma-jam,
i saw that the USB high speed Esp32 P4 support until 16 channels and full speed, so it may work no ?
Thanks.

roma-jam, 2024-10-18:
Hi @maaamcube, 

P4 does have 16 channels, but at the moment the Transaction Translator has not been implemented.
Only High-speed devices could be attached to the P4 through an external Hub.

AndyDevLat, 2025-02-23:
So what is the timeline of  Transaction Translator feature on ESP32-P4?  It is absolutely needed for our project and its absence defeats the use of external hubs.  If it is not even planned then we need to search for other solutions...