espressif/esp-idf #14683: ESP-IDF VS Code extension setup fails at import _ssl

IanLewis42, 2024-10-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected the setup to complete successfully.

### What is the actual behavior?

Setup halts with the error message:
non zero exit code 1

Traceback (most recent call last):
  File "c:\Installs\esp-idf-v5.3.1\esp-idf-v5.3.1\tools\idf_tools.py", line 43, in <module>
    import ssl
  File "C:\Users\ilewis\.platformio\python3\Lib\ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.

### Steps to reproduce.

Follow the instructions here:
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md
Choose 'Express Setup', and 'Find ESP-IDF in your system' point to previously downloaded esp-idf-v5.3.1 folder.

### Build or installation Logs.

```plain
Using existing C:\Users\ilewis\.espressif\dist\idf-git-2.39.2-win64.zip
Installing C:\Users\ilewis\.espressif\tools\idf-git\2.39.2 ...
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\cmd\git.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\cmd\scalar.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\awk.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\b2sum.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\basename.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\basenc.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\cat.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\chattr.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\cmp.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\bin\comm.exe
                                        already exists and was not updated.
[... 290 lines cut ...]
Traceback (most recent call last):
  File "c:\Installs\esp-idf-v5.3.1\esp-idf-v5.3.1\tools\idf_tools.py", line 43, in <module>

    import ssl

  File "C:\Users\ilewis\.platformio\python3\Lib\ssl.py", line 100, in <module>

    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.

non zero exit code 1

Traceback (most recent call last):
  File "c:\Installs\esp-idf-v5.3.1\esp-idf-v5.3.1\tools\idf_tools.py", line 43, in <module>
    import ssl
  File "C:\Users\ilewis\.platformio\python3\Lib\ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ImportError: DLL load failed while importing _ssl: The specified procedure could not be found.
```

### More Information.

I ran the doctor command, but the output seems to be too big to include in the issue.

jakub-kocka, 2024-10-08:
Hi @IanLewis42,
I was not able to reproduce your issue.

But it looks like the PATH (`"C:\Users\ilewis\.platformio\python3\Lib\ssl.py"`) is not from the tools directory and should not be here during the installation, can you try to disable the PlatformIO extension during the installation?

IanLewis42, 2024-10-09:
Hi, @jakub-kocka 

I have tried what you suggested, disable platformio extension, but it made no difference, the error was the same, I guess because PYTHONHOME and PYTHONPATH are both pointing to C:\Users\ilewis\.platformio\python3.

I tried changing both PYTHONHOME and PYTHONPATH  to C:\Users\ilewis\.espressif\tools\idf-python\3.11.2

and I got a different error, see below.

Downloading from https://dl.espressif.com/dl/esp-idf/idf_versions.txt
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Using existing C:\Users\ilewis\.espressif\dist\idf-git-2.39.2-win64.zip
Installing C:\Users\ilewis\.espressif\tools\idf-git\2.39.2 ...
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\cmd\git.exe
                                        already exists and was not updated.
Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\cmd\scalar.exe
                                        already exists and was not updated.

<snip>
---------
<snip>

Warning File C:\Users\ilewis\.espressif\tools\idf-git\2.39.2\usr\ssl\openssl.cnf.dist
                                        already exists and was not updated.
Extracted C:\Users\ilewis\.espressif\tools\idf-git\2.39.2 ...
Using existing C:\Users\ilewis\.espressif\tools\idf-python\3.11.2
"C:\Users\ilewis\.espressif\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)

jakub-kocka, 2024-10-09:
Hi, @IanLewis42,

Okay, this seems to be an issue with the environment.
You can try to run `python -m ensurepip` in the PowerShell console for example ... or VS Code terminal and repeat the installation.

I am not sure how you got the existing `ESP-IDF` environment but in case this does not help, you can try to reinstall the ESP-IDF directory and environment with the IDF installer (https://dl.espressif.com/dl/esp-idf/) - I recommend using the "Universal Online Installer", it will also sets the register values and environmental variables for you. Then proceed with the configuration of the VS Code extension with the installed directory.

jakub-kocka, 2024-10-15:
Hi @IanLewis42,
I am glad that you get it working 👍

However, the `PYTHONPATH` variable determines the global Python interpreter used.

> In resulting CMD prompt, run 'python -m ensurepip'

This should not be needed because when the installation is completed, for the "resulting" terminals its environment with the interpreter is used (e.g. `C:\Espressif\tools\idf-python\3.11.2;C:\Espressif\python_env\idf5.3_py3.11_env`) and there it should be included.

Be aware that this might be a little issue when using Python for other use cases not just with ESP-IDF tools that the interpreter will be the one from the PATH you have specified.

Thanks for providing more information, the resolution should be complete, so we will close this issue.

closed 2024-10-18 by dobairoland