espressif/esp-idf #14685: Usage example for External HUB support on ESP32 in USB Host mode

rtek1000, 2024-10-08:
### Is your feature request related to a problem?

Hello, I noticed that yesterday the issue 12554 was completed, which is about adding support for HUB,

I would like to suggest that a usage example be added to implement this support.

Ref.:

Support for 1-level USB hub in USB host implementation (IDFGH-11416)
(closed this as completed):
https://github.com/espressif/esp-idf/issues/12554

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/

Implementation & Usage
Host Stack Interaction

The Ext Hub Driver takes place between USB Host layer and USBH layer, next to the Hub Driver. The Hub Driver and the Ext Hub Driver were split into two Drivers to achieve the goal of logic distinguishing between root Hub and external Hub.
Device handling

The Ext Hub Driver can be installed via ext_hub_install() call and uninstalled via ext_hub_uninstall() call. After installation the Ext Hub provides the following APIs for external Hub addition and removal:

ext_hub_new_dev() which will verify the device class (HUB_CLASSCODE (09H)) and, if the device has the Hub class, the Ext Hub Driver:

- allocates a new device object

- adds it to the external device pool

- starts the process of Hub configuration (retrieving Hub Descriptor, Device status and Hub status)

- ext_hub_dev_gone() which will verify the device in the Ext Hub Driver list and start the process of external Hub device removing.

Events & Processing

The Ext Hub Driver is completely event driven and all event handling is done via the ext_hub_process() function. The ext_hub_config_t.proc_req_cb callback provided on the Ext Hub Driver installation will be called when processing is required. Typically, ext_hub_process() will be called from the Hub Driver hub_process() processing function.

The Ext Hub Driver does not expose any event callback.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/usb_host/usb_host_notes_ext_hub.html

### Describe the solution you'd like.

I would like to suggest that a usage example be added to implement this support.

### Describe alternatives you've considered.

If there are already examples, please add shortcuts (links) to find them.

roma-jam, 2024-10-08:
Hi @rtek1000,

the external hubs will be transparent in usage and no additional example is necessary: all current examples will be working as before. 

Only one thing that should be done is enabling this feature via menuconfig. 

The examples update with enabling this featiee by default was already done internally and will be availble in public shortly. 

Thanks for the feedback.

roma-jam, 2024-10-15:
Hi @rtek1000 , 

USB Host examples have been updated recently.
 
Feel free to try it out.

closed 2024-10-15 by roma-jam