espressif/esp-idf #14686: esp32-c3 anti-interference performance issues

ShengjieCh, 2024-10-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I used esp32-c3 and other 3 wifi chips to conduct wifi anti-interference test comparison. The test topology is shown in the figure:
[image]

The test method is to run iperf3 with tcp parameter -b 10M for 30 seconds to count the results. During this period, a vector signal generate is used to generate interference of different strengths, and a programmable attenuator is used to achieve signal attenuation. The test results are shown in the figure.
[image]

Here are a few questions:
1. When rssi=-40dBm, esp32 has a flow attenuation at -57dBm interference, which is earlier than other chips;
2. When rssi=-60dBm, the interference intensity has a slow attenuation between -65dBm and -60dBm. Why is this?
3. When rssi=-70dBm, even when the interference is very low (or even no interference), the flow of esp32 cannot reach 10Mbps;

The test results are confusing. Please help analyze it. Thank you very much!

MaxwellAlan, 2024-10-12:
Hi @ShengjieCh 

Thank you for the feedback on these issues. Since this is the first time we've encountered them, we still have a lot of questions. There may be more tests in the future that will require your assistance. Information on GitHub may not always be updated promptly, so if possible, please get in touch with our company's technical support. Our development team will prioritize addressing these issues.

ShengjieCh, 2024-10-14:
> Hi @ShengjieCh
>
> Thank you for the feedback on these issues. Since this is the first time we've encountered them, we still have a lot of questions. There may be more tests in the future that will [...]

How can I contact with technical support? Is there any contact information?

MaxwellAlan, 2024-10-14:
Hi @ShengjieCh 

You can get in touch with us from this website: https://www.espressif.com/en/contact-us/technical-inquiries
or send email to our sales support email: sales@espressif.com

MaxwellAlan, 2025-03-07:
Thanks for reporting this issue. Feel free to reopen if still have same issues.

closed 2025-03-07 by MaxwellAlan